openrazer/openrazer #200: Different Keyboard Layouts not supported in macros

Jujuedv, 2017-02-15:
I am using the de-neo layout on a de layouted BlackWidow Chroma (set by `setxkbmap de-neo`) 

If i now record a macro on the fly and try to replay this macro it uses the wrong layout. It now seems to ignore my keyboard layout and always defaults to the English layout (which is neither the layout of the keyboard nor my xserver).

lah7, 2017-02-15:
Which distro and desktop environment are you using?

I experienced a similar problem a while back with my British `en_GB` keyboard - and my Ubuntu MATE has always been set to be `en_GB`. Like you, after using macros, it somehow became `en_US` until the daemon was stopped.

My workaround was to add this command to **Start-up Applications** as soon as I log on, or once when the problem starts:

    setxkbmap -layout 'gb'

I presume it would be `de` in your case, does that fix the issue?

terrycain, 2017-02-15:
Macros are "experimental", hope to rewrite all of the macro code at some point

GabMus, 2017-02-15:
I just happened to have implemented macro support in razercommander, my frontend for these drivers (https://github.com/gabmus/razercommander).
In razercommander I use commands even for keystroke macros, so it should work for you. Note that you will need to use razercommander on the macros_ng branch

Jujuedv, 2017-02-15:
I am using Arch 4.9.8-1-ARCH with i3-gaps as my WM.

Not my keyboard layout got changed by using a macro, only that particular macro got changed.

For example if i record "uiaenrtd" (the home row) as my macro and replay it i get "asdfjklj". 

The daemon was definitely started after the keyboard layout was already set since i have to start it manually (since it only provides a GNOME autostart and not systemd)

As to razercommander, i can only find an option to record a single keystroke as a macro, but i'd like to be able to record a sequence of button presses.
This single keystroke uses the correct layout for some layers of my keyboard layout,  but not all of them. (μ works, ℕ does not)

In case you are wondering what kind of layout that is: [](http://neo-layout.org/grafik/aufsteller/neo20-aufsteller.pdf) It uses CAPS, SHIFT and ALT-GR and combinations of those as modifiers.

terrycain, 2017-02-15:
There's 2 issues here.
1. The keyboard by default __i think__ will send out en-us or en-gb keycodes and then the os will convert, and the daemon reads straight from the event file so would get whatever the kb outputs.

2. Getting the currently set keymap is an arse, especially in a cross distro manner, if thats solved I could easily get the daemon to do key translation for macros. Another option (when apps support it, daemon functionality is there) to just use an app to set macros and dont use the on the fly option. Currently macros in general and especially the on the fly macros are experimental and I plan to rewrite it at some point.

Jujuedv, 2017-02-15:
i don't exactly know how it works, but could you not feed the events through the system, so all keymaps get used? So that the driver does not have to care about layouts at all?

terrycain, 2017-02-15:
The events are fed through the system as they're typed but the daemon reads them too early in the process, if it was/is possible to read them later in the pipeline then im all ears. As I said its buggy and going to get a rewrite. Once I have it rewritten, working and cleaned up, I'll have a look at this. Ideally I'd like a better way to do macros than is currently done but i'll have to dig in some other large codebases for that.

terrycain, 2017-02-15:
Technically keymaps exist as currently it converts the keycode to the uk layout i think. If we could get the keymap in use, we could possibly use that, would probably be the best solution.

Jujuedv, 2017-02-15:
At which point in the process are the macros fed back in? maybe it could help if you created another input device and used that to feed it through the system and the whole keymap conversion

terrycain, 2017-02-15:
Some command probs xte is ran with the keys as arguments. Is possible, I could create a kernel endpoint to accept keycodes which would work but then displaying said macro to end users would present issues, as at that point the keycodes would need to be converted to keys to be presented and currently it would be wrong. Same goes the other way, if a macro was set by a program, keys would need to be converted to keycodes (incorrectly currently), fed to the driver

GabMus, 2017-02-15:
> As to razercommander, i can only find an option to record a single keystroke as a macro

Yeah, right, that's just what most people seem to do, and that's just what I implemented. You could try to use a command instead, like this: `xdotool key `(key combo, keys separated by + signs, ie `xdotool key a+s+d+f+g+h+j+k+l`). It's a bit tedious to setup, but it should work fine.

terrycain, 2017-02-15:
@GabMus I'd need to look at the code but the daemon will already do `xdotool key sgerevdsvsd` for you, but its gonna get redone hopefully on weekend if im not tooo busy

closed 2018-02-28 by stale[bot]