openrazer/openrazer #201: Issues with mice under driver mode

tildearrow, 2017-02-15:
**Description:**
Mice under driver mode do not send events for DPI buttons and wheel left/right tilt.

**Steps to Reproduce:**
Write a 0x03 and a 0x00 to `/sys/bus/hid/drivers/razermouse/0003\:1532\:*/device_mode`.
Then pressing the DPI buttons and wheel left/right won't work.

**What should happen instead:**
Pressing the DPI buttons should produce keycodes, so the daemon can handle them and change the DPI.
And tilting the scroll wheel should produce wheel left/right events.

terrycain, 2017-02-15:
Is a known issue ;-) requires some effort

tildearrow, 2017-03-07:
I think I just did some sort of implementation for it.
It works pretty much like the keyboard one: it catches "0x04" raw events, and it remaps them to F13-F16 keys.
However, what I am trying to do, is to remap the F13-F16 keys to mouse button presses, but it doesn't seem to be working. I have even tried to hack the keybit bitmap to make it work, but it still doesn't.
Or should I leave it as F13-F16? (which I really don't want)

terrycain, 2017-03-07:
I think the issue is that they emit keycodes on one of the interfaces that isnt interface 0 (which would mean they're keyboard interfaces), might be wrong havent looked at the mice stuff as of late

tildearrow, 2017-03-11:
Exactly. I just thought of a workaround:
Let the wheel tilt and DPI buttons be mapped to F13-F16, then catch them in the daemon and execute right actions (wheel tilt does scroll left/right and DPI buttons change DPI).

Is this possible? If so I can create a pull request for the map part.

terrycain, 2017-03-11:
Yeah should be fine. We'd need an exclusive read on the event files so the F13-16 doesn't interfere with other system shortcuts (i remeber something like f23 does touchpad on/off on some systems)

But yeah should be fine

rlifshay, 2018-02-06:
I would like to see side scrolling support, and I just now found this issue. I am commenting because I see that this has just recently been marked as stale.

wirespot, 2018-06-09:
Hi, I've built and installed the driver/daemon/udev etc. from latest master, which contains the patch listed above. After switching the device mode to 3:0, shouldn't I see F13-F16 keys being emitted when testing with xev? I'm getting nothing.

Is there anything else I can map the raw events to in order to make it work (obtain usable events)? Or other patches I can apply?

Orochikaku, 2018-10-26:
Hi, just curious as to whether this will be fixed at any point. I personally use the tilt scroll a lot when in Windows.

G-M0N3Y-2503, 2021-08-28:
I noticed that the implementation for supporting tilt scroll for the Naga 2014 was the same as the Basilisk. perhaps this is a general solution for more than these two devices. :thinking: 

https://github.com/openrazer/openrazer/pull/1597/commits/65ad28898cc8a0abc4253f2c911edd94cdb004c5