openrazer/openrazer #203: How do i use the driver?

gedreher, 2017-02-18:
Hi, i have the razer Abyssus 2014 mice and after downloading the packages for my OS which is Ubuntu 16.10 i couldn't figure out how to use them, i tried looking for the set_mouse_dpi file but it seems it's missing.

terrycain, 2017-02-18:
@gedreher It's just called dpi now

terrycain, 2017-02-18:
Also you shouldnt use the driver directly, the daemon provides a dbus api and theres a python library that interracts with that.

gedreher, 2017-02-18:
oh ok, didn't know that, in which directory i can find that python library?

terrycain, 2017-02-18:
did you install all the debs?

z3ntu, 2017-02-18:
```
from razer.client import DeviceManager
a = DeviceManager()
a.devices
```
gives you an array from razer devices if you mean that

gedreher, 2017-02-19:
yes, i added the repository manually and in the terminal i pasted this "sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc"

terrycain, 2017-02-19:
```
$ python3
>>> from razer.client import DeviceManger
>>>
```
Should work

gedreher, 2017-02-19:
that worked, thank both of you!

closed 2017-02-19 by gedreher

nonetrix, 2019-01-14:
> ```
> $ python3
> >>> from razer.client import DeviceManger
> >>>
> ```
> 
> Should work

ModuleNotFoundError: No module named 'razer'

z3ntu, 2019-01-15:
@nonetrix Change the `razer.client` to `openrazer.client`, the project has had a rename since the comment was written