openrazer/openrazer #204: Unable to change DPI/Polling for Abyssus

lszanto, 2017-02-18:
After razer decided to move the amazing buttons from the bottom of their mouse into a software based solution i've struggled to get my abyssus to work and change the dpi/polling on linux. 

I've tried with the python api but it appears to just give a not implemented error. Is there anything I can do to fix this or potentially change my dpi another way? If not I fear I will have to go back to using my old mouse.

z3ntu, 2017-02-18:
Please copy-paste the stacktrace here, otherwise there isn't much we can do.

lszanto, 2017-02-19:
`➜  ~ python
Python 3.6.0 (default, Jan 16 2017, 12:12:55) 
[GCC 6.3.1 20170109] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import razer.client as a
>>> b = a.DeviceManager()
>>> c = b.devices[0]
>>> c.name
'Razer Abyssus 2014'
>>> c.poll_rate(1000)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.6/site-packages/razer/client/devices/mice.py", line 80, in poll_rate
    raise NotImplementedError()
NotImplementedError
>>> c.dpi(3000)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.6/site-packages/razer/client/devices/mice.py", line 36, in dpi
    raise NotImplementedError()
NotImplementedError
>>> 
`

terrycain, 2017-02-19:
do the same as above and then run this, what does it give
```
b.version
b.daemon_version
c.version
c._available_features
```

lszanto, 2017-02-19:
`>>> c = b.devices[0]
>>> c.name
'Razer Abyssus 2014'
>>> b.version
'1.1.8'
>>> b.daemon_version
'1.1.8'
>>> c.version
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'RazerMouse' object has no attribute 'version'
>>> c._available_features
{'razer.device.misc': ['getSerial', 'suspendDevice', 'getDeviceMode', 'getRazerUrls', 'setDeviceMode', 'resumeDevice', 'getVidPid', 'getDriverVersion', 'hasDedicatedMacroKeys', 'getFirmware', 'getMatrixDimensions', 'hasMatrix', 'getDeviceName', 'getDeviceType'], 'razer.device.lighting.logo': ['setLogoActive', 'getLogoActive']}
>>> 
`

terrycain, 2017-02-19:
Ahh the abyssus doesnt have dpi support in the driver, fancy adding it
You'll need to follow this - https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol

lszanto, 2017-02-19:
Yeah for sure i'll take a look next weekend when I get the chance, thanks for your help!

closed 2017-02-19 by lszanto