openrazer/openrazer #207: Abyssus mouse - service fails to initialize

memoryleakno1, 2017-02-20:
I am not sure what happened, I made a post earlier today, but it seems to be gone.

I have problems running with Abyssus. Service crash during initialisation:

> 17-02-20 15:50:28 | razer.device0                  | INFO     | Initialising device.0 RazerAbyssus
> Traceback (most recent call last):
>   File "/usr/bin/razer-service", line 71, in <module>
>     run()
>   File "/usr/bin/razer-service", line 65, in run
>     daemonize(**daemon_args)
>   File "/usr/lib/python3.6/site-packages/razer_daemon/daemon.py", line 114, in daemonize
>     daemon = RazerDaemon(verbose, log_dir, console_log, run_dir, config_file, test_dir=test_dir)
>   File "/usr/lib/python3.6/site-packages/razer_daemon/daemon.py", line 217, in __init__
>     self._load_devices(first_run=True)
>   File "/usr/lib/python3.6/site-packages/razer_daemon/daemon.py", line 430, in _load_devices
>     razer_device = device_class(sys_path, device_number, self._config, testing=self._test_dir is not None, additional_interfaces=sorted(additional_interfaces))
>   File "/usr/lib/python3.6/site-packages/razer_daemon/hardware/mouse.py", line 113, in __init__
>     super(RazerAbyssus, self).__init__(*args, **kwargs)
>   File "/usr/lib/python3.6/site-packages/razer_daemon/hardware/device_base.py", line 60, in __init__
>     self.serial = self.get_serial()
>   File "/usr/lib/python3.6/site-packages/razer_daemon/hardware/device_base.py", line 194, in get_serial
>     with open(serial_path, 'r') as serial_file:
> PermissionError: [Errno 13] Permission denied: '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.2/1-2.2:1.0/0003:1532:0042.0001/device_serial'
> Exception ignored in: <object repr() failed>
> Traceback (most recent call last):
>   File "/usr/lib/python3.6/site-packages/razer_daemon/hardware/device_base.py", line 412, in __del__
>     self.close()
>   File "/usr/lib/python3.6/site-packages/razer_daemon/hardware/device_base.py", line 325, in close
>     if not self._is_closed:
> AttributeError: 'RazerAbyssus' object has no attribute '_is_closed'

terrycain, 2017-02-20:
@memoryleakno1 Remind me tonight and I'll have a look

memoryleakno1, 2017-02-20:
I have checked spaces, looks fine.

terrycain, 2017-02-20:
@memoryleakno1 Are you running it from master?

memoryleakno1, 2017-02-20:
From AUR package version 1.1.8-1. Wasn't lucky compiling it from git. cmake and I are not best friends.

z3ntu, 2017-02-20:
Just use the packages suffixed with `-git` ;)

memoryleakno1, 2017-02-20:
Ok, will give it another try :).

memoryleakno1, 2017-02-20:
Ok, I just downloaded and rebuild everything, but am a bit confused. There are 3 different '-git' packages: python-razer, razer-daemon and razer-driver-dkms. After compilation they all produce same 3 packages to install. Is it safe to assume the content is identical, so it does not matter which one I choose? :-).

terrycain, 2017-02-20:
are you a member of plugdev group?

memoryleakno1, 2017-02-20:
yepp, did that one.

terrycain, 2017-02-20:
reboot then run 
```
ls -lah /sys/bus/hid/drivers/
ls -lah /sys/bus/hid/drivers/razermouse/
ls -lah /sys/bus/hid/drivers/razermouse/0*/
```

memoryleakno1, 2017-02-20:
Sorry not so random. It just locks on TAB key while in console :-)

z3ntu, 2017-02-20:
@memoryleakno1 Yeah, I implementier that stuff as a split package so one PKGBUILD produces 3 packages.

z3ntu, 2017-02-20:
*implemented

(Can't edit comments on mobile)

memoryleakno1, 2017-02-20:
Don't worry, I hate typing on phone. I understand now how it works.
ls lah /sys/bus/hid/driver
totalt 0

> drwxr-xr-x 4 root root 0 20 feb 17.34 .
> drwxr-xr-x 4 root root 0 20 feb 17.34 ..
> drwxr-xr-x 2 root root 0 20 feb 17.34 hid-generic
> drwxr-xr-x 2 root root 0 20 feb 17.34 razermouse

ls -lah /sys/bus/hid/drivers/razermouse/
totalt 0

> drwxr-xr-x 2 root root    0 20 feb 17.34 .
> drwxr-xr-x 4 root root    0 20 feb 17.34 ..
> lrwxrwxrwx 1 root root    0 20 feb 17.34 0003:1532:0042.0001 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.2/1-2.2:1.0/0003:1532:0042.0001
> --w------- 1 root root 4,0K 20 feb 17.34 bind
> lrwxrwxrwx 1 root root    0 20 feb 17.36 module -> ../../../../module/razermouse
> --w------- 1 root root 4,0K 20 feb 17.36 new_id
> --w------- 1 root root 4,0K 20 feb 17.36 uevent
> --w------- 1 root root 4,0K 20 feb 17.36 unbind

And ls -lah /sys/bus/hid/drivers/razermouse/0* >
lrwxrwxrwx 1 root root 0 20 feb 17.34 /sys/bus/hid/drivers/razermouse/0003:1532:0042.0001 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.2/1-2.2:1.0/0003:1532:0042.0001

PS, Coulnd't I just reload kernel driver instead of rebooting? :) It's not virtual machine :)

memoryleakno1, 2017-02-20:
ok I have reinstalled all packages, am sure the old ones were uninstalled clean. I added some try/catches to isolate those "attribute.not found" errors, and I see new ones are fresh. I get now error in file daemon.py, line 395:
  File "/usr/lib/python3.6/site-packages/razer_daemon/daemon.py", line 395, in _load_devices
    self.logger.debug(format_str.format(cls.__name__ + ' ', cls.USB_VID, cls.USB_PID))
AttributeError: type object 'DBusService' has no attribute 'USB_VID'

z3ntu, 2017-02-20:
`ls -lah /sys/bus/hid/drivers/razermouse/0*/` (mind the `/` at the end!)

memoryleakno1, 2017-02-20:
> totalt 0
> drwxr-xr-x 5 root plugdev    0 20 feb 18.01 .
> drwxr-xr-x 5 root root       0 20 feb 17.34 ..
> -r--r--r-- 1 root root    4,0K 20 feb 18.01 country
> -rw-rw---- 1 root root    4,0K 20 feb 18.01 device_mode
> -r--r----- 1 root root    4,0K 20 feb 18.01 device_serial
> -r--r----- 1 root root    4,0K 20 feb 18.01 device_type
> lrwxrwxrwx 1 root root       0 20 feb 18.01 driver -> ../../../../../../../../bus/hid/drivers/razermouse
> -r--r----- 1 root root    4,0K 20 feb 18.01 firmware_version
> drwxr-xr-x 3 root root       0 20 feb 18.01 hidraw
> drwxr-xr-x 3 root root       0 20 feb 18.01 input
> -rw-rw---- 1 root root    4,0K 20 feb 18.01 logo_led_state
> -r--r--r-- 1 root plugdev 4,0K 20 feb 17.34 modalias
> drwxr-xr-x 2 root plugdev    0 20 feb 17.34 power
> -r--r--r-- 1 root plugdev 4,0K 20 feb 17.34 report_descriptor
> lrwxrwxrwx 1 root plugdev    0 20 feb 17.34 subsystem -> ../../../../../../../../bus/hid
> --w--w---- 1 root root    4,0K 20 feb 18.01 test
> -rw-r--r-- 1 root plugdev 4,0K 20 feb 17.34 uevent
> -r--r----- 1 root root    4,0K 20 feb 18.01 version

z3ntu, 2017-02-20:
Yeah these file in that dir should be owned by `plugdev` but aren't out of some reason. 
Also please post the output of `sudo journalctl -b | grep -i razer_moun`

memoryleakno1, 2017-02-20:
> feb 20 17:34:55 pascal razer_mount[308]: Driver razermouse
> feb 20 17:34:55 pascal razer_mount[309]: Device_ID 0003:1532:0042.0001
> feb 20 17:34:55 pascal razer_mount[310]: Unbinding 0003:1532:0042.0001 from hid-generic
> feb 20 17:34:55 pascal razer_mount[312]: Binding 0003:1532:0042.0001 to razermouse
> feb 20 17:34:55 pascal razer_mount[317]: Finished binding 0003:1532:0042.0001
> feb 20 17:34:55 pascal razer_mount[318]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0042.0001/
> feb 20 17:34:55 pascal razer_mount[320]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0042.0001/

memoryleakno1, 2017-02-20:
I just checked that I am in plugdev groups
> ± |makepkg U:3 ?:5 ✗| → groups myusername
> wheel storage power plugdev users

z3ntu, 2017-02-20:
Mind trying to run the script https://raw.githubusercontent.com/terrycain/razer-drivers/master/scripts/troubleshoot.sh ? Maybe it sees something I don't know about ^^

memoryleakno1, 2017-02-20:
>  |makepkg U:4 ?:5 ✗| → scripts/troubleshoot.sh
> This script will now try to determine why the razer driver isn't working correctly.
> Please enter your 'sudo' password for priviliged operations.
> [sudo] lösenord för myname:
> No razer-service currently running.
> Traceback (most recent call last):
>   File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 175, in activate_name_owner
>     return self.get_name_owner(bus_name)
>   File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 361, in get_name_owner
>     's', (bus_name,), **keywords)
>   File "/usr/lib/python3.6/site-packages/dbus/connection.py", line 651, in call_blocking
>     message, timeout)
> dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.razer': no such name
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/usr/lib/python3.6/site-packages/razer/client/__init__.py", line 20, in __init__
>     self._dbus = session_bus.get_object("org.razer", "/org/razer")
>   File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 241, in get_object
>     follow_name_owner_changes=follow_name_owner_changes)
>   File "/usr/lib/python3.6/site-packages/dbus/proxies.py", line 248, in __init__
>     self._named_service = conn.activate_name_owner(bus_name)
>   File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 180, in activate_name_owner
>     self.start_service_by_name(bus_name)
>   File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 278, in start_service_by_name
>     'su', (bus_name, flags)))
>   File "/usr/lib/python3.6/site-packages/dbus/connection.py", line 651, in call_blocking
>     message, timeout)
> dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.razer was not provided by any .service files
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "<string>", line 1, in <module>
>   File "/usr/lib/python3.6/site-packages/razer/client/__init__.py", line 22, in __init__
>     raise DaemonNotFound("Could not connect to daemon")
> razer.client.DaemonNotFound: Could not connect to daemon
> The script didn't find an error. If you still think something's wrong, please create an issue on the GitHub page. Exiting.

z3ntu, 2017-02-20:
Okay :( So it didn't find an obvious error... 😞

z3ntu, 2017-02-20:
Do you have Telegram? If yes, please write me at @z3ntu

memoryleakno1, 2017-02-20:
Of sorry I dont' have Telegram, never heard of that one to be honest

z3ntu, 2017-02-20:
Please try a `sudo chown -R 777 /sys/bus/hid/drivers/razermouse/0*/` and try to start the daemon again (`razer-service`). That should circumvent all possible permission issues...

memoryleakno1, 2017-02-20:
> ○ → ls -lah /sys/bus/hid/drivers/razermouse/0*/
> totalt 0
> drwxr-xr-x 5  777 plugdev    0 20 feb 18.04 .
> drwxr-xr-x 5 root root       0 20 feb 17.34 ..
> -r--r--r-- 1  777 root    4,0K 20 feb 18.26 country
> -rw-rw---- 1  777 root    4,0K 20 feb 18.26 device_mode
> -r--r----- 1  777 root    4,0K 20 feb 18.26 device_serial
> -r--r----- 1  777 root    4,0K 20 feb 18.26 device_type
> lrwxrwxrwx 1  777 root       0 20 feb 18.26 driver -> ../../../../../../../../bus/hid/drivers/razermouse
> -r--r----- 1  777 root    4,0K 20 feb 18.26 firmware_version
Ok, I have changed permissions:

> drwxr-xr-x 3  777 root       0 20 feb 18.26 hidraw
> drwxr-xr-x 3  777 root       0 20 feb 18.26 input
> -rw-rw---- 1  777 root    4,0K 20 feb 18.26 logo_led_state
> -r--r--r-- 1  777 plugdev 4,0K 20 feb 17.34 modalias
> drwxr-xr-x 2  777 plugdev    0 20 feb 17.34 power
> -r--r--r-- 1  777 plugdev 4,0K 20 feb 17.34 report_descriptor
> lrwxrwxrwx 1  777 plugdev    0 20 feb 17.34 subsystem -> ../../../../../../../../bus/hid
> --w--w---- 1  777 root    4,0K 20 feb 18.26 test
> -rw-r--r-- 1  777 plugdev 4,0K 20 feb 17.34 uevent
> -r--r----- 1  777 root    4,0K 20 feb 18.26 version

And I got same error on same line:
 

> 2017-02-20 18:27:08 ⌚  pascal in ~/rztest/razer-driver-dkms-git
> ○ → 2017-02-20 18:27:08 | razer                          | INFO     | Initialising Daemon (v1.1.8). Pid: 20914
> 2017-02-20 18:27:08 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
> Traceback (most recent call last):
>   File "/usr/bin/razer-service", line 102, in <module>
>     run()
>   File "/usr/bin/razer-service", line 96, in run
>     daemonize(**daemon_args)
>   File "/usr/lib/python3.6/site-packages/razer_daemon/daemon.py", line 114, in daemonize
>     daemon = RazerDaemon(verbose, log_dir, console_log, run_dir, config_file, test_dir=test_dir)
>   File "/usr/lib/python3.6/site-packages/razer_daemon/daemon.py", line 217, in __init__
>     self._load_devices(first_run=True)
>   File "/usr/lib/python3.6/site-packages/razer_daemon/daemon.py", line 395, in _load_devices
>     self.logger.debug(format_str.format(cls.__name__ + ' ', cls.USB_VID, cls.USB_PID))
> AttributeError: type object 'DBusService' has no attribute 'USB_VID'
> ^C
> [2]+  Avslut 1                /usr/bin/razer-service --foreground --verbose --run-dir ~/.razer-service/data --log-dir ~/.razer-service/logs --pid-file ~/.razer-service/daemon.pid

z3ntu, 2017-02-20:
Please revert your changes to the daemon, because it's not making anything better ;)

> -rw-rw---- 1 777 root 4,0K 20 feb 18.26 device_mode
-r--r----- 1 777 root 4,0K 20 feb 18.26 device_serial
-r--r----- 1 777 root 4,0K 20 feb 18.26 device_type

they are still owned by root.... 

And Telegram is a WhatsApp-like (but much better 😉) messaging service where we have real-time communication which is pretty convenient.

memoryleakno1, 2017-02-20:
I just got 35 hits from yaourt on telegram so I understand it is quite popular :-). I will install it later.

Hmm .. that a bit odd, I just repeated command, and I see it's still root owning them.

> ± |makepkg U:4 ?:6 ✗| → _ chown -R 777 /sys/bus/hid/drivers/razermouse/0*/
> [sudo] lösenord för :
> 
>  2017-02-20 18:43:26 ⌚  pascal in ~/rztest/razer-driver-dkms-git/src/razer-drivers-git
> ± |makepkg U:4 ?:6 ✗| → ls -lah /sys/bus/hid/drivers/razermouse/0*/
> totalt 0
> drwxr-xr-x 5  777 plugdev    0 20 feb 18.04 .
> drwxr-xr-x 5 root root       0 20 feb 17.34 ..
> -r--r--r-- 1  777 root    4,0K 20 feb 18.26 country
> -rw-rw---- 1  777 root    4,0K 20 feb 18.26 device_mode
> -r--r----- 1  777 root    4,0K 20 feb 18.26 device_serial
> -r--r----- 1  777 root    4,0K 20 feb 18.26 device_type
> lrwxrwxrwx 1  777 root       0 20 feb 18.26 driver -> ../../../../../../../../bus/hid/drivers/razermouse
> -r--r----- 1  777 root    4,0K 20 feb 18.26 firmware_version
> drwxr-xr-x 3  777 root       0 20 feb 18.26 hidraw
> drwxr-xr-x 3  777 root       0 20 feb 18.26 input
> -rw-rw---- 1  777 root    4,0K 20 feb 18.26 logo_led_state
> -r--r--r-- 1  777 plugdev 4,0K 20 feb 17.34 modalias
> drwxr-xr-x 2  777 plugdev    0 20 feb 17.34 power
> -r--r--r-- 1  777 plugdev 4,0K 20 feb 17.34 report_descriptor
> lrwxrwxrwx 1  777 plugdev    0 20 feb 17.34 subsystem -> ../../../../../../../../bus/hid
> --w--w---- 1  777 root    4,0K 20 feb 18.26 test
> -rw-r--r-- 1  777 plugdev 4,0K 20 feb 17.34 uevent
> -r--r----- 1  777 root    4,0K 20 feb 18.26 version
> 
>  2017-02-20 18:43:43 ⌚  pascal in ~/rztest/razer-driver-dkms-git/src/razer-drivers-git
If I log in as a root I get another can of worms:

> [root@pascal razer-drivers-git]# /usr/bin/razer-service --foreground --verbose --run-dir ~/.razer-service/data --log-dir ~/.razer-service/logs --pid-file ~/.razer-service/daemon.pid
> Traceback (most recent call last):
>   File "/usr/bin/razer-service", line 102, in <module>
>     run()
>   File "/usr/bin/razer-service", line 96, in run
>     daemonize(**daemon_args)
>   File "/usr/lib/python3.6/site-packages/razer_daemon/daemon.py", line 114, in daemonize
>     daemon = RazerDaemon(verbose, log_dir, console_log, run_dir, config_file, test_dir=test_dir)
>   File "/usr/lib/python3.6/site-packages/razer_daemon/daemon.py", line 174, in __init__
>     DBusService.__init__(self, self.BUS_PATH, '/org/razer')
>   File "/usr/lib/python3.6/site-packages/razer_daemon/dbus_services/service.py", line 81, in __init__
>     bus_object = dbus.service.BusName(bus_name, bus=dbus.SessionBus())
>   File "/usr/lib/python3.6/site-packages/dbus/_dbus.py", line 211, in __new__
>     mainloop=mainloop)
>   File "/usr/lib/python3.6/site-packages/dbus/_dbus.py", line 100, in __new__
>     bus = BusConnection.__new__(subclass, bus_type, mainloop=mainloop)
>   File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 122, in __new__
>     bus = cls._new_for_bus(address_or_type, mainloop=mainloop)
> dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

But I think I read somewhere in "issues" that service is not supposed to be run as root?

terrycain, 2017-02-20:
@memoryleakno1 Yeah don't run it as root

z3ntu, 2017-02-20:
DBus stuff is pretty environment specific and doesn't work with sudo without additional configuration.

z3ntu, 2017-02-20:
Do you have ANY additional security stuff configured, like selinux, secure boot or anything like that?

memoryleakno1, 2017-02-20:
Jesus, what do they pull in Qt-Wayland as a dependency to install telegram on my X11 linux???

Yeah, I know you can't just simply sudo and use d-bus, but I have another tty to login as root.

memoryleakno1, 2017-02-20:
Njah, for the current it is just plaing Arch. A new installation on a new computer.

terrycain, 2017-02-20:
sudo chmod -R 777 /sys/bus/hid/drivers/razermouse/0*/
sudo chown -R root:plugdev /sys/bus/hid/drivers/razermouse/0*/

z3ntu, 2017-02-20:
lol sry, I meant `chmod` instead of `chown` before, my bad :( listen to @terrycain ^^

memoryleakno1, 2017-02-20:
Njah, I see no change:

>  2017-02-20 18:54:41 ⌚  pascal in ~/rztest/razer-driver-dkms-git/src/razer-drivers-git
> ± |makepkg U:4 ?:6 ✗| → sudo chmod -R 777 /sys/bus/hid/drivers/razermouse/0*/
> 
>  2017-02-20 18:54:54 ⌚  pascal in ~/rztest/razer-driver-dkms-git/src/razer-drivers-git
> ± |makepkg U:4 ?:6 ✗| → sudo chown -R root:plugdev /sys/bus/hid/drivers/razermouse/0*/
> 
>  2017-02-20 18:55:04 ⌚  pascal in ~/rztest/razer-driver-dkms-git/src/razer-drivers-git
> ± |makepkg U:4 ?:6 ✗| → ls -lah /sys/bus/hid/drivers/razermouse/0*/
> totalt 0
> drwxrwxrwx 5 root plugdev    0 20 feb 18.04 .
> drwxr-xr-x 5 root root       0 20 feb 17.34 ..
> -rwxrwxrwx 1 root plugdev 4,0K 20 feb 18.26 country
> -rwxrwxrwx 1 root plugdev 4,0K 20 feb 18.26 device_mode
> -rwxrwxrwx 1 root plugdev 4,0K 20 feb 18.26 device_serial
> -rwxrwxrwx 1 root plugdev 4,0K 20 feb 18.26 device_type
> lrwxrwxrwx 1 root plugdev    0 20 feb 18.26 driver -> ../../../../../../../../bus/hid/drivers/razermouse
> -rwxrwxrwx 1 root plugdev 4,0K 20 feb 18.26 firmware_version
> drwxrwxrwx 3 root plugdev    0 20 feb 18.26 hidraw
> drwxrwxrwx 3 root plugdev    0 20 feb 18.26 input
> -rwxrwxrwx 1 root plugdev 4,0K 20 feb 18.26 logo_led_state
> -rwxrwxrwx 1 root plugdev 4,0K 20 feb 17.34 modalias
> drwxrwxrwx 2 root plugdev    0 20 feb 17.34 power
> -rwxrwxrwx 1 root plugdev 4,0K 20 feb 17.34 report_descriptor
> lrwxrwxrwx 1 root plugdev    0 20 feb 17.34 subsystem -> ../../../../../../../../bus/hid
> -rwxrwxrwx 1 root plugdev 4,0K 20 feb 18.26 test
> -rwxrwxrwx 1 root plugdev 4,0K 20 feb 17.34 uevent
> -rwxrwxrwx 1 root plugdev 4,0K 20 feb 18.26 version

I can't explain, I am a bit noob on those matters :-).

z3ntu, 2017-02-20:
Try starting the daemon / service again

memoryleakno1, 2017-02-20:
Still same.

z3ntu, 2017-02-20:
So a permission denied error? Or your `AttributeError: type object 'DBusService' has no attribute 'USB_VID'` one? If the second, did you revert your changes you made earlier to the daemon? If no, we can't help you with the issues you are experiencing.

memoryleakno1, 2017-02-20:
It's the attribute one 'usb ..

Yepp, I reverted changes at time we started. I uninstalled everything razer, than downloaded packages a new and built them a new. I also checked that newly installed python files didn't had my changes. Changes I did was just to add try-catch blocks around those attributes that were rapported as missing - though I see tham declared in constructor. Anyway, there are no my changes, since I installed clean.

z3ntu, 2017-02-20:
Does the daemon crash too when no device is connected?

memoryleakno1, 2017-02-20:
No it doesn't. Since the driver knows only for one device - Abyssus mouse, I did unload kernel module and int that case daemon runs. I can connect to it from python shell list devices (an emoty array) and stop it.

z3ntu, 2017-02-20:
The crashlog you sent before is from https://github.com/terrycain/razer-drivers/blob/master/daemon/razer_daemon/daemon.py#L395 which happens or doesn't happen no matter if a device is connected or not... Really weird...

z3ntu, 2017-02-20:
`pacman -Ql razer-daemon-git` please

memoryleakno1, 2017-02-20:
>  → pacman -Ql razer-daemon-git
> razer-daemon-git /etc/
> razer-daemon-git /etc/xdg/
> razer-daemon-git /etc/xdg/autostart/
> razer-daemon-git /etc/xdg/autostart/razer-service.desktop
> razer-daemon-git /usr/
> razer-daemon-git /usr/bin/
> razer-daemon-git /usr/bin/razer-service
> razer-daemon-git /usr/lib/
> razer-daemon-git /usr/lib/python3.6/
> razer-daemon-git /usr/lib/python3.6/site-packages/
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon-1.1.8-py3.6.egg-info/
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon-1.1.8-py3.6.egg-info/PKG-INFO
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon-1.1.8-py3.6.egg-info/SOURCES.txt
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon-1.1.8-py3.6.egg-info/dependency_links.txt
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon-1.1.8-py3.6.egg-info/top_level.txt
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/__init__.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/__pycache__/
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/__pycache__/__init__.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/__pycache__/daemon.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/__pycache__/device.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/__pycache__/keyboard.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/daemon.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/__init__.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/__pycache__/
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/__pycache__/__init__.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/__pycache__/service.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__init__.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/__init__.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/all.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/bw2013.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/chroma_keyboard.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/deathadder_chroma.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/kraken.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/macro.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/mamba.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/mug.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/nagahex.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/nagahexv2.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/tartarus.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/all.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/bw2013.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/chroma_keyboard.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/deathadder_chroma.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/kraken.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/macro.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/mamba.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/mug.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/nagahex.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/nagahexv2.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/tartarus.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/service.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/device.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/hardware/
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/hardware/__init__.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/hardware/__pycache__/
[... cut ...]
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/misc/ripple_effect.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/misc/screensaver_monitor.py
> razer-daemon-git /usr/share/
> razer-daemon-git /usr/share/man/
> razer-daemon-git /usr/share/man/man5/
> razer-daemon-git /usr/share/man/man5/razer.conf.5.gz
> razer-daemon-git /usr/share/man/man8/
> razer-daemon-git /usr/share/man/man8/razer-service.8.gz
> razer-daemon-git /usr/share/razer-service/
> razer-daemon-git /usr/share/razer-service/razer.conf.example

z3ntu, 2017-02-20:
Okay so apparently the packaging script failed and removed all underscores from the file/directory paths... eg it should be `/usr/lib/python3.6/site-packages/razer_daemon/__init__.py` instead of `/usr/lib/python3.6/site-packages/razer_daemon/init.py`
left: yours - right: mine
[image]
What are you using to build/install the packages? (You mentioned yaourt before.)

memoryleakno1, 2017-02-20:
I have downloaded pkgbuild file from yaourt and building from git. Nothing special, just standard makepkg  && pacman -U *.xz 

I have just rebuild package will check how it look again.

z3ntu, 2017-02-20:
Is there anything special about the directory you are building it in? For example a space in the directoryname or anything else?

memoryleakno1, 2017-02-20:
>  2017-02-20 19:53:00 ⌚  pascal in ~/rztest/razer-driver-dkms-git/src/razer-drivers-git
> ± |makepkg U:4 ?:6 ✗| → pacman -Ql razer-daemon-git
> razer-daemon-git /etc/
> razer-daemon-git /etc/xdg/
> razer-daemon-git /etc/xdg/autostart/
> razer-daemon-git /etc/xdg/autostart/razer-service.desktop
> razer-daemon-git /usr/
> razer-daemon-git /usr/bin/
> razer-daemon-git /usr/bin/razer-service
> razer-daemon-git /usr/lib/
> razer-daemon-git /usr/lib/python3.6/
> razer-daemon-git /usr/lib/python3.6/site-packages/
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon-1.1.8-py3.6.egg-info/
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon-1.1.8-py3.6.egg-info/PKG-INFO
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon-1.1.8-py3.6.egg-info/SOURCES.txt
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon-1.1.8-py3.6.egg-info/dependency_links.txt
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon-1.1.8-py3.6.egg-info/top_level.txt
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/__init__.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/__pycache__/
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/__pycache__/__init__.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/__pycache__/daemon.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/__pycache__/device.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/__pycache__/keyboard.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/daemon.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/__init__.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/__pycache__/
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/__pycache__/__init__.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/__pycache__/service.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__init__.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/__init__.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/all.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/bw2013.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/chroma_keyboard.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/deathadder_chroma.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/kraken.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/macro.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/mamba.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/mug.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/nagahex.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/nagahexv2.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/__pycache__/tartarus.cpython-36.pyc
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/all.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/bw2013.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/chroma_keyboard.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/deathadder_chroma.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/kraken.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/macro.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/mamba.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/mug.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/nagahex.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/nagahexv2.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/dbus_methods/tartarus.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/dbus_services/service.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/device.py
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/hardware/
> razer-daemon-git /usr/lib/python3.6/site-packages/razer_daemon/hardware/__init__.py
[... cut ...]
Njah, it shoulnd't be. Just standard pacman. 

However this new build is my first build with patched code so I understand if you would like to give up here since it is no longer vanilla install. 

I have Deathstalker kbd so I am trying to get it to work. It is pretty much same as Deathstalker Chroma, minus chroma part. I have just loaded newly build kbd & mouse driver, and thus fur they don't crash. But it does not mean they work :-). Would be interesting if I get deamon up to see if it works as intended.

memoryleakno1, 2017-02-20:
AHA Underscores are there, but it is Github & markdown that spooks! Underscores dissaper in github markup. I thought that quoting eliminates this. I am terribly sorry for the inconvenience here. Didn't realize this myself untill now.

z3ntu, 2017-02-20:
Oh wait, I think github removed the underscores by doing markdown formatting 😠 
Please use codeblocks (triple backticks `) from now on...

memoryleakno1, 2017-02-20:
:-) yeah. we just realized it both 👍

z3ntu, 2017-02-20:
Okay I would suggest you just using my 'unofficial repo'.
Put into `/etc/pacman.conf`:
```
[home_z3ntu_Arch_Extra]
Server = http://anorien.csc.warwick.ac.uk/mirrors/download.opensuse.org/repositories/home:/z3ntu/Arch_Extra/$arch/
```
And run
```
# download the signing key
sudo pacman-key --lsign-key 9C6B8662CF4D5535
# trust the signing key
sudo pacman-key -r 9C6B8662CF4D5535
```
Then you can install/update the packages
```
sudo pacman -Syu razer-driver-dkms razer-daemon python-razer
```
If the download server is slow, I can give you others (they are from the opensuse build service)

memoryleakno1, 2017-02-20:
Ok, thank you that is very kind from you. I will try a bit more to see if I can get this to work somehow tomorrow before I try your packages. I think I fine for today too, it's getting late. Thank you for your very kind help.

memoryleakno1, 2017-02-21:
Can the error has something to do that with d-bus and multithreading as described here:

http://stackoverflow.com/questions/11685936/python-attributeerror-object-has-no-attribute ?

terrycain, 2017-02-21:
Most likely not as master works for other arch users and the location of that error "should" hit everyone

z3ntu, 2017-02-21:
@memoryleakno1 So my prebuilt packages didn't work for you? 😞

memoryleakno1, 2017-02-21:
Oh I am sorry I didn't tested them yet. I didn't do so much last night before sleep, and as you saw, i am more interested to get my kbd extra keys to work. To be honest I am not sure if really care for d-bus integration and those extra applications like razercommander & polycolor.

z3ntu, 2017-02-21:
Okay so let's start from the beginning:
**You have:**
* DeathStalker Expert
* Abyssus

**Your problem(s):**
* DeathStalker Expert isn't supported by the driver yet (see the new branch).
* The daemon crashes when the Absyssus is connected.

Your system:
* Arch Linux (with yaourt)

Did I get this correct for now?

p.s. we would really appreciate if you could help test the devices because then the experience will be better for you and others!

[5 more replies]

closed 2017-06-11 by z3ntu