openrazer/openrazer #208: razercommander fails to launch

greyltc, 2017-02-20:
When I launch it with my mouse (Naga HexV2) plugged in:
```
Traceback (most recent call last):
  File "/usr/share/razercommander/main.py", line 74, in <module>
    refreshDevices()
  File "/usr/share/razercommander/main.py", line 70, in refreshDevices
    initDevices()
  File "/usr/share/razercommander/main.py", line 36, in initDevices
    newdev = device.Device(dev)
  File "/usr/share/razercommander/device.py", line 30, in __init__
    self.device)
  File "/usr/share/razercommander/macro_logic.py", line 73, in make_device
    n_device = MacroDevice(mk_arr, device_uid, device)
  File "/usr/share/razercommander/macro_logic.py", line 14, in __init__
    self.set_all_current_macros()
  File "/usr/share/razercommander/macro_logic.py", line 32, in set_all_current_macros
    key.set_current_macro(self.device)
  File "/usr/share/razercommander/macro_logic.py", line 50, in set_current_macro
    self._set_macro(device, self.macro)
  File "/usr/share/razercommander/macro_logic.py", line 43, in _set_macro
    device.macro.del_macro(self.key)
AttributeError: 'RazerMouse' object has no attribute 'macro'
```

memoryleakno1, 2017-02-21:
Seems like a same issue as what I have yesterday with my Abyss mouse. I haven't got it to work either. :(. I don't know if this post might be a hind to what happeneds:
http://stackoverflow.com/questions/11685936/python-attributeerror-object-has-no-attribute

terrycain, 2017-02-21:
@memoryleakno1 its a separate issue. This is to do with the python library not the daemon

Yeah my bad, I haven't added the lines to the mouse part of the library that allows interracting with macros, should get that done by end of week

terrycain, 2017-03-09:
Done but not released.

z3ntu, 2017-03-09:
Keeping this open until the next release 🙂

terrycain, 2017-04-15:
Packages are building in launchpad, will be available soon

commit 984cbb22 (2017-03-09): Added macro capability to be used by all devices that need it.

Fixes #208
closed 2017-04-15 by terrycain