openrazer/openrazer #210: Deathstalker Keyboard

memoryleakno1, 2017-02-21:
There are by now different Deathstalker's keyboards, but I am sitting on the old one with monochromatic light. Keyboard seem to be same layout as Deathstalkerchroma minus chroma part.

I tested yesterday to patch code to run Deathstalkerchroma driver on it and thus part, the kernel driver seems to work in sense that it loads correctly on boot and does not crash or do anything weird with basic functionality. 

 2017-02-21 06:52:42 ⌚  pascal in ~/rzrtmp
```sh
○ → fmod razer
razerkbd               49152  0
razermouse             53248  0
hid                   114688  5 i2c_hid,razermouse,hid_generic,usbhid,razerkbd
usbcore               208896  8 razermouse,uvcvideo,usbhid,razerkbd,snd_usb_audio,xhci_pci,snd_usbmidi_lib,xhci_hcd

 2017-02-21 06:53:08 ⌚  pascal in ~/rzrtmp
○ → modinfo razerkbd
filename:       /lib/modules/4.9.11-1-ARCH/kernel/drivers/hid/razerkbd.ko
license:        GPL v2
description:    Razer Keyboard Device Driver
author:         Terry Cain <terry@terrys-home.co.uk>
alias:          hid:b0003g*v00001532p0000010F
alias:          hid:b0003g*v00001532p0000021F
alias:          hid:b0003g*v00001532p0000021E
alias:          hid:b0003g*v00001532p0000021A
alias:          hid:b0003g*v00001532p00000216
alias:          hid:b0003g*v00001532p00000209
alias:          hid:b0003g*v00001532p00000204
alias:          hid:b0003g*v00001532p00000202
alias:          hid:b0003g*v00001532p00000203
alias:          hid:b0003g*v00001532p00000208
alias:          hid:b0003g*v00001532p00000210
alias:          hid:b0003g*v00001532p0000020F
alias:          hid:b0003g*v00001532p00000220
alias:          hid:b0003g*v00001532p00000205
alias:          hid:b0003g*v00001532p00000217
alias:          hid:b0003g*v00001532p00000214
alias:          hid:b0003g*v00001532p0000011A
alias:          hid:b0003g*v00001532p0000010D
alias:          hid:b0003g*v00001532p0000010E
alias:          hid:b0003g*v00001532p0000011B
alias:          hid:b0003g*v00001532p00000113
depends:        hid,usbcore
vermagic:       4.9.11-1-ARCH SMP preempt mod_unload modversions

○ → ls -lah /sys/bus/hid/drivers/razerkbd/
totalt 0
drwxr-xr-x 2 root root    0 20 feb 21.09 .
drwxr-xr-x 5 root root    0 20 feb 21.09 ..
lrwxrwxrwx 1 root root    0 21 feb 06.04 0003:1532:0202.0006 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.4/1-2.4:1.0/0003:1532:0202.0006
lrwxrwxrwx 1 root root    0 21 feb 06.04 0003:1532:0202.0007 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.4/1-2.4:1.1/0003:1532:0202.0007
lrwxrwxrwx 1 root root    0 21 feb 06.04 0003:1532:0202.0008 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.4/1-2.4:1.2/0003:1532:0202.0008
--w------- 1 root root 4,0K 21 feb 06.04 bind
lrwxrwxrwx 1 root root    0 21 feb 06.58 module -> ../../../../module/razerkbd
--w------- 1 root root 4,0K 21 feb 06.58 new_id
--w------- 1 root root 4,0K 20 feb 21.09 uevent
--w------- 1 root root 4,0K 21 feb 06.58 unbind

○ → ls -lah /sys/bus/hid/drivers/razerkbd/0*/
'/sys/bus/hid/drivers/razerkbd/0003:1532:0202.0006/':
totalt 0
drwxr-xr-x 5 root plugdev    0 21 feb 06.04 .
drwxr-xr-x 5 root root       0 21 feb 06.04 ..
-r--r--r-- 1 root plugdev 4,0K 21 feb 06.04 country
lrwxrwxrwx 1 root plugdev    0 21 feb 06.04 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x 3 root plugdev    0 21 feb 06.04 hidraw
drwxr-xr-x 3 root plugdev    0 21 feb 06.04 input
-rw-rw---- 1 root plugdev 4,0K 21 feb 06.04 key_alt_f4
-rw-rw---- 1 root plugdev 4,0K 21 feb 06.04 key_alt_tab
-rw-rw---- 1 root plugdev 4,0K 21 feb 06.04 key_super
-r--r--r-- 1 root plugdev 4,0K 21 feb 06.04 modalias
drwxr-xr-x 2 root plugdev    0 21 feb 06.04 power
-r--r--r-- 1 root plugdev 4,0K 21 feb 06.04 report_descriptor
lrwxrwxrwx 1 root plugdev    0 21 feb 06.04 subsystem -> ../../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4,0K 21 feb 06.04 uevent

'/sys/bus/hid/drivers/razerkbd/0003:1532:0202.0007/':
totalt 0
drwxr-xr-x 5 root plugdev    0 21 feb 06.04 .
drwxr-xr-x 5 root root       0 21 feb 06.04 ..
-r--r--r-- 1 root plugdev 4,0K 21 feb 06.04 country
lrwxrwxrwx 1 root plugdev    0 21 feb 06.04 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x 3 root plugdev    0 21 feb 06.04 hidraw
drwxr-xr-x 3 root plugdev    0 21 feb 06.04 input
-rw-rw---- 1 root plugdev 4,0K 21 feb 06.04 key_alt_f4
-rw-rw---- 1 root plugdev 4,0K 21 feb 06.04 key_alt_tab
-rw-rw---- 1 root plugdev 4,0K 21 feb 06.04 key_super
-r--r--r-- 1 root plugdev 4,0K 21 feb 06.04 modalias
drwxr-xr-x 2 root plugdev    0 21 feb 06.04 power
-r--r--r-- 1 root plugdev 4,0K 21 feb 06.04 report_descriptor
lrwxrwxrwx 1 root plugdev    0 21 feb 06.04 subsystem -> ../../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4,0K 21 feb 06.04 uevent

'/sys/bus/hid/drivers/razerkbd/0003:1532:0202.0008/':
totalt 0
drwxr-xr-x 5 root plugdev    0 21 feb 06.04 .
drwxr-xr-x 5 root root       0 21 feb 06.04 ..
-r--r--r-- 1 root plugdev 4,0K 21 feb 06.04 country
-rw-rw---- 1 root plugdev 4,0K 21 feb 06.04 device_mode
-r--r----- 1 root plugdev 4,0K 21 feb 06.04 device_serial
-r--r----- 1 root plugdev 4,0K 21 feb 06.04 device_type
lrwxrwxrwx 1 root plugdev    0 21 feb 06.04 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
-r--r----- 1 root plugdev 4,0K 21 feb 06.04 firmware_version
-rw-rw---- 1 root plugdev 4,0K 21 feb 06.04 game_led_state
drwxr-xr-x 3 root plugdev    0 21 feb 06.04 hidraw
drwxr-xr-x 3 root plugdev    0 21 feb 06.04 input
-rw-rw---- 1 root plugdev 4,0K 21 feb 06.04 macro_led_effect
-rw-rw---- 1 root plugdev 4,0K 21 feb 06.04 macro_led_state
-rw-rw---- 1 root plugdev 4,0K 21 feb 06.04 matrix_brightness
--w--w---- 1 root plugdev 4,0K 21 feb 06.04 matrix_custom_frame
--w--w---- 1 root plugdev 4,0K 21 feb 06.04 matrix_effect_breath
--w--w---- 1 root plugdev 4,0K 21 feb 06.04 matrix_effect_custom
--w--w---- 1 root plugdev 4,0K 21 feb 06.04 matrix_effect_none
--w--w---- 1 root plugdev 4,0K 21 feb 06.04 matrix_effect_spectrum
--w--w---- 1 root plugdev 4,0K 21 feb 06.04 matrix_effect_static
--w--w---- 1 root plugdev 4,0K 21 feb 06.04 matrix_effect_wave
-r--r--r-- 1 root plugdev 4,0K 21 feb 06.04 modalias
drwxr-xr-x 2 root plugdev    0 21 feb 06.04 power
-r--r--r-- 1 root plugdev 4,0K 21 feb 06.04 report_descriptor
lrwxrwxrwx 1 root plugdev    0 21 feb 06.04 subsystem -> ../../../../../../../../bus/hid
-rw-rw---- 1 root plugdev 4,0K 21 feb 06.04 test
-rw-r--r-- 1 root plugdev 4,0K 21 feb 06.04 uevent
-r--r----- 1 root plugdev 4,0K 21 feb 06.04 version
```
Now I suppose if I write to stuff ment for rgb lighting it won't work or may even crash the driver, but I would like to see if I can get get Fn key and logo and backliting volume to work? Currently keyboard backlits as soon as it gets power, but it seems to preserve it's light volume from what was set in Windows. 

I am aware Fn key on it's own does nothing, but keyboard should send different keys when a combination of say Fn+F1 is pressed.

terrycain, 2017-02-21:
Get some usb packet captures of it working on Windows them we can see

memoryleakno1, 2017-02-21:
Hmm, it will take a while before I load into windows,, I'll come back to that one. Anyway I am getting problem with binding driver to the device:
```sh
○ → ls -l /sys/bus/hid/devices/
totalt 0
lrwxrwxrwx 1 root root 0 21 feb 06.04 0003:1532:0042.0005 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.2/1-2.2:1.0/0003:1532:0042.0005
lrwxrwxrwx 1 root root 0 21 feb 06.04 0003:1532:0202.0006 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.4/1-2.4:1.0/0003:1532:0202.0006
lrwxrwxrwx 1 root root 0 21 feb 06.04 0003:1532:0202.0007 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.4/1-2.4:1.1/0003:1532:0202.0007
lrwxrwxrwx 1 root root 0 21 feb 06.04 0003:1532:0202.0008 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.4/1-2.4:1.2/0003:1532:0202.0008

○ → udevadm info -a /sys/bus/hid/devices/0003:1532:0202.0008 | grep bInterfaceNumber
    ATTRS{bInterfaceNumber}=="02"

○ → ls -l /sys/bus/hid/drivers/hid-generic/
totalt 0
--w------- 1 root root 4096 21 feb 08.18 bind
lrwxrwxrwx 1 root root    0 21 feb 08.11 module -> ../../../../module/hid_generic
--w------- 1 root root 4096 21 feb 08.11 new_id
--w------- 1 root root 4096 21 feb 08.11 uevent
--w------- 1 root root 4096 21 feb 08.16 unbind

○ → echo -n "0003:1532:0202.0008" > bind
bash: bind: Åtkomst nekas

 2017-02-21 08:24:01 ⌚  pascal in /sys/bus/hid/drivers/hid-generic
○ → su root
Lösenord:
[root@pascal hid-generic]# echo -n "0003:1532:0202.0008" > bind
bash: echo: skrivfel: Enheten finns inte
[root@pascal hid-generic]# LANG=en
[root@pascal hid-generic]# echo -n "0003:1532:0202.0008" > bind
bash: echo: skrivfel: Enheten finns inte
[root@pascal hid-generic]#
```

z3ntu, 2017-02-21:
you can force english output with `LC_ALL= <command>` ;) And didn't you say, the device isn't supported yet? 

Also is the keyboard called **Razer DeathStalker Expert**? Because that one has the same usb pid as this one

z3ntu, 2017-02-21:
If yes, this is a duplicate of #185

memoryleakno1, 2017-02-21:
Yeah, I was just lazy, for some apps LANG= use to suffice, anyway no idea about Stalker Expert. 

Now when I look at pictures and checked the thread, it does look completely same, so now I am a proud owner of Deathstalker Export keyboard without knowing it :-). Anyway, I didn't see pid 0202 in the table of supported devices so I assumed I need to hack it. I don't see any changes in master branch (I dowlnoaded it yesterday and hacked):
```sh
 |master U:6 ?:7 ✗| → ag deathstalker
daemon/razer_daemon/hardware/keyboards.py
700:class RazerDeathStalkerChroma(_MacroKeyboard):
702:    Class for the DeathStalker Chroma
704:    EVENT_FILE_REGEX = re.compile(r'.*DeathStalker_Chroma(-if01)?-event-kbd')
717:        "store": "https://www.razerzone.com/gb-en/store/razer-deathstalker-chroma",
718:        "top_img": "https://assets.razerzone.com/eeimages/products/22563/rzr_deathstalker_chroma_05.png",
719:        "side_img": "https://assets.razerzone.com/eeimages/products/22563/rzr_deathstalker_chroma_03.png",
720:        "perspective_img": "https://assets.razerzone.com/eeimages/products/22563/rzr_deathstalker_chroma_02.png"
724:        super(RazerDeathStalkerChroma, self).__init__(*args, **kwargs)
736:class RazerDeathStalker(_MacroKeyboard):
738:    Class for the DeathStalker
740:    EVENT_FILE_REGEX = re.compile(r'.*DeathStalker(-if01)?-event-kbd')
753:        "store": "https://www.razerzone.com/gb-en/store/razer-deathstalker-chroma",
754:        "top_img": "https://assets.razerzone.com/eeimages/products/22563/rzr_deathstalker_chroma_05.png",
755:        "side_img": "https://assets.razerzone.com/eeimages/products/22563/rzr_deathstalker_chroma_03.png",
756:        "perspective_img": "https://assets.razerzone.com/eeimages/products/22563/rzr_deathstalker_chroma_02.png"
760:        super(RazerDeathStalker, self).__init__(*args, **kwargs)

driver/razerkbd_driver.h
45:#ifndef USB_DEVICE_ID_RAZER_DEATHSTALKER
46:    #define USB_DEVICE_ID_RAZER_DEATHSTALKER 0x0202
53:#ifndef USB_DEVICE_ID_RAZER_DEATHSTALKER_CHROMA
54:    #define USB_DEVICE_ID_RAZER_DEATHSTALKER_CHROMA 0x0204

driver/razerkbd_driver.c
319:        case USB_DEVICE_ID_RAZER_DEATHSTALKER:
320:            device_type = "Razer DeathStalker Chroma\n";
323:        case USB_DEVICE_ID_RAZER_DEATHSTALKER_CHROMA:
324:            device_type = "Razer DeathStalker Chroma\n";
734:        case USB_DEVICE_ID_RAZER_DEATHSTALKER:
735:        case USB_DEVICE_ID_RAZER_DEATHSTALKER_CHROMA:
1177:                   case USB_DEVICE_ID_RAZER_DEATHSTALKER:
1178:                   case USB_DEVICE_ID_RAZER_DEATHSTALKER_CHROMA:
1623:            case USB_DEVICE_ID_RAZER_DEATHSTALKER:
1624:            case USB_DEVICE_ID_RAZER_DEATHSTALKER_CHROMA:
1842:            case USB_DEVICE_ID_RAZER_DEATHSTALKER:
1843:            case USB_DEVICE_ID_RAZER_DEATHSTALKER_CHROMA:
1898:    { HID_USB_DEVICE(USB_VENDOR_ID_RAZER,USB_DEVICE_ID_RAZER_DEATHSTALKER) },
1899:    { HID_USB_DEVICE(USB_VENDOR_ID_RAZER,USB_DEVICE_ID_RAZER_DEATHSTALKER_CHROMA) },

debian/changelog
24:  * Added Device - Razer DeathStalker Chroma

README.md
25:| Razer DeathStalker                            |  1532:0202  |   ✔    |   ✔    |      |
26:| Razer DeathStalker Chroma                     |  1532:0204  |   ✔    |   ✔    |      |
``'
(I have aded 0202 to rules in 99-razer.rules).

I will change title of thread as well.

z3ntu, 2017-02-21:
I'm just gonna point you to the branch [`feature_deathstalker_expert_0x0202`](https://github.com/terrycain/razer-drivers/tree/feature_deathstalker_expert_0x0202) 😉

terrycain, 2017-02-21:
Duplicate #185

closed 2017-02-21 by terrycain

memoryleakno1, 2017-02-21:
Aha. 👍 Thanks.

memoryleakno1, 2017-02-21:
Oki, after recompile and fixing my bad changes, I can now confirm that Fn button works fine. Pressing F1 with Fn generates now correctly XF85AudioMute. Before Fn had no effect. 

Backlitting brightness now works too, I was able to change it by writing different values to it:
```sh
○ → cat /sys/bus/hid/drivers/razerkbd/0*/matrix_brightness
255

 2017-02-21 09:34:31 ⌚  pascal in ~
○ → echo "0" > /sys/bus/hid/drivers/razerkbd/0*/matrix_brightness

 2017-02-21 09:34:51 ⌚  pascal in ~
○ → echo "255" > /sys/bus/hid/drivers/razerkbd/0*/matrix_brightness

 2017-02-21 09:34:58 ⌚  pascal in ~
○ → echo "128" > /sys/bus/hid/drivers/razerkbd/0*/matrix_brightness

 2017-02-21 09:35:07 ⌚  pascal in ~
○ → echo "50" > /sys/bus/hid/drivers/razerkbd/0*/matrix_brightness

 2017-02-21 09:35:19 ⌚  pascal in ~
○ → echo "256" > /sys/bus/hid/drivers/razerkbd/0*/matrix_brightness

 2017-02-21 09:35:29 ⌚  pascal in ~
○ → echo "255" > /sys/bus/hid/drivers/razerkbd/0*/matrix_brightness
```
I could also swith on/off game and macro leds (not that it matters for any purpose) but as the other user I can't see the way to turn on/off logo.