openrazer/openrazer #212: Razer starts with zero brightness after computer sleeping and/or screensaver activation

epertinez, 2017-02-21:
I always have to click fn+F12 to put correct brightness after computer goes to sleep and or screensaver is activated.
I do use  RazerOrnataChroma
Daemon (v1.1.8)
May it be that  get_brightness is returning zero so set_brightness sets it to zero after wake up?

2017-02-21 15:09:10 | razer.device0                  | DEBUG    | DBus call get_brightness
2017-02-21 15:29:22 | razer.screensaver              | DEBUG    | Received screensaver active signal
2017-02-21 15:29:22 | razer.device0                  | INFO     | Suspending RazerOrnataChroma
2017-02-21 15:29:22 | razer.device0                  | DEBUG    | DBus call get_brightness
2017-02-21 15:29:22 | razer.device0                  | DEBUG    | DBus call set_brightness
2017-02-21 15:29:22 | razer.screensaver              | DEBUG    | Received screensaver active signal
2017-02-21 15:29:22 | razer.device0                  | INFO     | Suspending RazerOrnataChroma
2017-02-21 15:29:22 | razer.device0                  | DEBUG    | DBus call get_brightness
2017-02-21 15:29:22 | razer.device0                  | DEBUG    | DBus call set_brightness
2017-02-21 16:28:32 | razer.screensaver              | DEBUG    | Received screensaver inactive signal
2017-02-21 16:28:32 | razer.device0                  | INFO     | Resuming RazerOrnataChroma
2017-02-21 16:28:32 | razer.device0                  | DEBUG    | DBus call set_brightness
2017-02-21 16:28:32 | razer.screensaver              | DEBUG    | Received screensaver inactive signal
2017-02-21 16:28:32 | razer.device0                  | INFO     | Resuming RazerOrnataChroma
2017-02-21 16:28:32 | razer.device0                  | DEBUG    | DBus call set_brightness
2017-02-21 16:28:34 | razer.device0                  | DEBUG    | DBus call set_brightness
2017-02-21 16:28:34 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerOrnataChroma:SI1649F12600138, 'setBrightness', 51)
2017-02-21 16:28:35 | razer.device0                  | DEBUG    | DBus call set_brightness
2017-02-21 16:28:35 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerOrnataChroma:SI1649F12600138, 'setBrightness', 102)
2017-02-21 16:28:35 | razer.device0                  | DEBUG    | DBus call set_brightness
2017-02-21 16:28:35 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerOrnataChroma:SI1649F12600138, 'setBrightness', 153)
2017-02-21 16:28:35 | razer.device0                  | DEBUG    | DBus call set_brightness
2017-02-21 16:28:35 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerOrnataChroma:SI1649F12600138, 'setBrightness', 204)
2017-02-21 16:28:35 | razer.device0                  | DEBUG    | DBus call set_brightness
2017-02-21 16:28:35 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerOrnataChroma:SI1649F12600138, 'setBrightness', 255)

z3ntu, 2017-02-21:
Yeah I experience the same issue; it's a more-or-less known problem
Arch Linux, KDE, SDDM

terrycain, 2017-02-21:
Ahh it receives the active signal multiple times. So theoretically after then first it gets brightness and stores it then sets brightness to 0, sets active signal again, gets brightness (which is 0) stores it and sets brightness to 0. Upon inactive signal it will restore the last known brightness which is 0. Will have a look

z3ntu, 2017-02-21:
```
2017-02-20 21:20:02 | razer.screensaver              | DEBUG    | Received screensaver active signal
2017-02-20 21:20:02 | razer.device0                  | INFO     | Suspending RazerDeathadderChroma
2017-02-20 21:20:02 | razer.device0                  | DEBUG    | DBus call get_logo_brightness
2017-02-20 21:20:02 | razer.device0                  | DEBUG    | DBus call get_scroll_brightness
2017-02-20 21:20:02 | razer.device0                  | DEBUG    | DBus call set_logo_brightness
2017-02-20 21:20:02 | razer.device0                  | DEBUG    | DBus call set_scroll_brightness
2017-02-20 21:20:02 | razer.device1                  | INFO     | Suspending RazerBlackWidowChroma
2017-02-20 21:20:02 | razer.device1                  | DEBUG    | DBus call get_brightness
2017-02-20 21:20:02 | razer.device1                  | DEBUG    | DBus call set_brightness
2017-02-20 21:20:02 | razer.screensaver              | DEBUG    | Received screensaver active signal
2017-02-20 21:20:02 | razer.device0                  | INFO     | Suspending RazerDeathadderChroma
2017-02-20 21:20:02 | razer.device0                  | DEBUG    | DBus call get_logo_brightness
2017-02-20 21:20:02 | razer.device0                  | DEBUG    | DBus call get_scroll_brightness
2017-02-20 21:20:02 | razer.device0                  | DEBUG    | DBus call set_logo_brightness
2017-02-20 21:20:02 | razer.device0                  | DEBUG    | DBus call set_scroll_brightness
2017-02-20 21:20:02 | razer.device1                  | INFO     | Suspending RazerBlackWidowChroma
2017-02-20 21:20:02 | razer.device1                  | DEBUG    | DBus call get_brightness
2017-02-20 21:20:02 | razer.device1                  | DEBUG    | DBus call set_brightness
2017-02-20 21:20:04 | razer.screensaver              | DEBUG    | Received screensaver inactive signal
2017-02-20 21:20:04 | razer.device0                  | INFO     | Resuming RazerDeathadderChroma
2017-02-20 21:20:04 | razer.device0                  | DEBUG    | DBus call set_logo_brightness
2017-02-20 21:20:04 | razer.device0                  | DEBUG    | DBus call set_scroll_brightness
2017-02-20 21:20:04 | razer.device1                  | INFO     | Resuming RazerBlackWidowChroma
2017-02-20 21:20:04 | razer.device1                  | DEBUG    | DBus call set_brightness
2017-02-20 21:20:04 | razer.screensaver              | DEBUG    | Received screensaver inactive signal
2017-02-20 21:20:04 | razer.device0                  | INFO     | Resuming RazerDeathadderChroma
2017-02-20 21:20:04 | razer.device0                  | DEBUG    | DBus call set_logo_brightness
2017-02-20 21:20:04 | razer.device0                  | DEBUG    | DBus call set_scroll_brightness
2017-02-20 21:20:04 | razer.device1                  | INFO     | Resuming RazerBlackWidowChroma
2017-02-20 21:20:04 | razer.device1                  | DEBUG    | DBus call set_brightness
```
for me.
@epertinez what desktop environment / login manager / distro are you using?

z3ntu, 2017-02-21:
Fixed in 3343bdcca3fd21717bb93f15e3e9f7a833cc8d52

commit d6aa71e7 (2017-03-04): Merge branch 'master' into stable

Closes #212
Closes #184
Closes #169
Closes #149
Closes #116
closed 2017-03-04 by terrycain