openrazer/openrazer #214: Razer Blade 14 late 2016 support

snatella, 2017-02-25:
Bus 001 Device 004: ID 1532:0224 Razer USA, Ltd

[chroma-lsusb.txt](https://github.com/terrycain/razer-drivers/files/801305/chroma-lsusb.txt)

Happy to provide more info if needed.

z3ntu, 2017-02-26:
Please record packet captures of the keyboard effects from Synapse.

ryanisnan, 2017-03-07:
I'm a new owner of a razer blade 2017 and the apps don't seem to have driver support yet, would be interested in helping out with this project. Would love to chat with someone about getting started.

z3ntu, 2017-03-07:
Please post the output of `lsusb` 
And you can join our chat at Telegram (dunno how right now but will tell you later)

z3ntu, 2017-03-07:
Do you have a username set on Telegram? If yes please post it and I will add you. I don't know if it's possible without a username right now as we don't have a 'join' link afaik

ryanisnan, 2017-03-07:
I'll try to sign up - we'll see what happens. If not, my email is simply my
username at gmail.

z3ntu, 2017-03-07:
Username isn't the email (you don't even need an email address for Telegram)

lezed1, 2017-03-07:
I have the same issue on my Razer Blade Stealth Late 2016. I can post a packet capture of ripple with a few different parameters if that helps.

z3ntu, 2017-03-07:
@lezed1 the usb id of yours is `1532:0224`? And yes, packet captures are needed.

lezed1, 2017-03-07:
@z3ntu No, it's a Razer Blade Stealth (Late 2016) with id `1532:0220`. However, it's the same issue on a different but also recent model. I'm currently typing out another issue, so I'll make a capture in a few minutes.

lezed1, 2017-03-07:
Here is a capture of a few different ripples. It starts with red (`0xFF0000`), green (`0x00FF00`) and blue (`0x0000FF`). It ends with a ripple of (`0xDDDDDD`).

From looking at the packets, this seems to just be using the individual key settings, so this may be solved by getting individual lighting working (#223).

z3ntu, 2017-03-07:
Please note, that this issue is about 1532:0224 and not 1532:0220 :)

lezed1, 2017-03-07:
Yes, but it appears to affect both. Would you like me to make a new issue?

z3ntu, 2017-03-07:
But it's still not for this issue. #166 is probably what you are looking for.

lezed1, 2017-03-07:
Oh, that's definitely better. I missed it by only searching for "stealth", my mistake.

z3ntu, 2017-03-07:
I'll delete the comments here to keep it tidy and the messages were completely unrelated

z3ntu, 2017-03-07:
@ryanisnan btw your "Blade 2017" is actually "Late 2016" 😉 so you are in the correct issue

ryanisnan, 2017-03-07:
Thanks @z3ntu - I wish Razer named these things slightly less confusing ;)

snatella, 2017-03-28:
Sorry, this is still on my radar to do at some point - just will take a bit of time and I'm fresh out of it!

z3ntu, 2017-03-28:
We have a Telegram chat where afaik most communication happen. Hit me up @z3ntu and I can add you to the group.

juliancoleman, 2017-04-06:
Has anyone made an attempt to reverse engineer this device?

z3ntu, 2017-04-06:
Hopefully we have something to test in the next couple of days.

terrycain, 2017-04-06:
Still need screenshots and packet captures

Retic1337, 2017-04-08:
Attempted to create packet captures [https://drive.google.com/open?id=0B9mxquoXdSjrVjA5Vm5SUzF3d1U](url)

lezed1, 2017-04-08:
No, it's split into folders that have what effect each capture is though.

terrycain, 2017-04-08:
@lezed1 Ahh cool

terrycain, 2017-04-11:
Had brief look at them, looks good from here, hopefully will get some time this week to work on it, should be a copy/paste job

terrycain, 2017-04-13:
Who actaully has this that can test, and then what distro you on

terrycain, 2017-04-13:
And can this laptop also toggle the logo led on the laptop lid?

fgblomqvist, 2017-04-13:
I have it (or rather I have the 2017 Blade but it has the same keyboard signature). I'm running Ubuntu 17.04 with GNOME. Would love to test it. Have had the laptop for a month now and as far as I know, you cannot toggle the lid led (I have not read anything about it). But I might be wrong.

terrycain, 2017-04-13:
Ahh can i get the product and model number of your laptop for @z3ntu's records, theres a spreadsheet somewhere.

Do you have telegram/can you get it? Me and a fair few other razer people hang out there now, would be easier for me to spam you with instructions there, if not the issue will do :wink:

juliancoleman, 2017-04-13:
I have a late 2016 Razer Blade with a 7th gen Intel core i7-7700HQ (same as @fgblomqvist). I'm currently on Ubuntu 16.04, using the Nvidia driver 381 (non-LTS). I can confirm that not even Synapse on Windows allows you to toggle the LED on the lid, so I'm unsure if this is possible, or if there's some workaround.

terrycain, 2017-04-13:
Nope thats cool, I just know some laptops do some dont.

terrycain, 2017-04-13:
Ok grab debs from [here](https://drive.google.com/drive/folders/0BydYgD2LzDUCSi1BdExpSlRlNm8?usp=sharing)
install with `sudo dpkg -i debname`, install deps with `sudo apt-get install -f` then reboot

then hopefully post output of `lsmod | grep razer`

juliancoleman, 2017-04-13:
`lsmod | grep razer` outputs an empty response for me. However, the install process seemed to have gone through just fine, so that's good.

terrycain, 2017-04-13:
`sudo modprobe razerkbd`
`lsmod | grep razer`

juliancoleman, 2017-04-13:
*derp*... forgot to `modprobe`..

```
razerkbd               53248  0
hid                   122880  5 i2c_hid,hid_generic,usbhid,hid_rmi,razerkbd
```

terrycain, 2017-04-13:
ok in theory udev should do that for us but we can deal with udev another time.

`ls /sys/bus/hid/drivers/hid-generic`
`ls /sys/bus/hid/drivers/razerkbd`

juliancoleman, 2017-04-13:
`ls /sys/bus/hid/drivers/hid-generic`

```
0003:046D:C537.0001  0003:046D:C537.0002  0003:1532:0224.0003  0003:1532:0224.0004  0003:1532:0224.0005  bind  module  new_id  uevent  unbind
```

`ls /sys/bus/hid/drivers/razerkbd`

```
bind  module  new_id  uevent  unbind
```

terrycain, 2017-04-13:
ok are you in the plugdev group? if not add yourself, log out and log back in.

Then sudo to root
```
/lib/udev/razer_mount razerkbd 0003:1532:0224.0003
/lib/udev/razer_mount razerkbd 0003:1532:0224.0003
```

terrycain, 2017-04-13:
Then `ls /sys/bus/hid/drivers/razerkbd`

juliancoleman, 2017-04-13:
when I ran the command, it said I was added to that group. Secure Boot has been disabled this entire time, so I can't imagine something happened with that. Let me keep going with these and I'll get back to you in a second.

juliancoleman, 2017-04-13:
```
sudo /lib/udev/razer_mount razerkbd 0003:1532:0224.0003
[sudo] password for julian:
Driver razerkbd
Device_ID 0003:1532:0224.0003
Unbinding 0003:1532:0224.0003 from hid-generic
Binding 0003:1532:0224.0003 to razerkbd
sh: printf: I/O error
```

terrycain, 2017-04-13:
ooh thats an odd one

juliancoleman, 2017-04-13:
Alright, so running any of those commands disabled my keyboard until a reboot

terrycain, 2017-04-13:
Yeah it would do, it managed to unbind from the general usb keyboard driver, but failed to bind it to the razer one

juliancoleman, 2017-04-13:
The backlights still work, so there's that :smile:

terrycain, 2017-04-13:
Ha

juliancoleman, 2017-04-13:
Haha okay, where do we go from here? Do you still want me to `ls /sys/bus/hid/drivers/razerkbd`? Or will that not have changed?

terrycain, 2017-04-13:
Ok install the razer-kernel-moduels deb from here - https://drive.google.com/open?id=0BydYgD2LzDUCSi1BdExpSlRlNm8

then reboot
```
ls /sys/bus/hid/drivers/hid-generic
```

juliancoleman, 2017-04-13:
add user to plugdev group again?

terrycain, 2017-04-13:
only have to do that once

juliancoleman, 2017-04-13:
```
0003:046D:C537.0001  0003:046D:C537.0002  bind  module  new_id  uevent  unbind
```

terrycain, 2017-04-13:
ooh
that looks more promising

terrycain, 2017-04-13:
ls /sys/bus/hid/drivers/razerkbd

juliancoleman, 2017-04-13:
Yesssss. Let's do this.

`ls /sys/bus/hid/drivers/razerkbd`

```
0003:1532:0224.0003  0003:1532:0224.0004  0003:1532:0224.0005  bind  module  new_id  uevent  unbind
```

terrycain, 2017-04-13:
ls /sys/bus/hid/drivers/razerkbd/0003:1532:0224.0005

juliancoleman, 2017-04-13:
`ls /sys/bus/hid/drivers/razerkbd/0003:1532:0224.0005`

```
country        firmware_version   matrix_custom_frame     matrix_effect_spectrum   power              version
device_mode    fn_toggle          matrix_effect_breath    matrix_effect_starlight  report_descriptor
device_serial  hidraw             matrix_effect_custom    matrix_effect_static     subsystem
device_type    input              matrix_effect_none      matrix_effect_wave       test
driver         matrix_brightness  matrix_effect_reactive  modalias                 uevent
```

terrycain, 2017-04-13:
```
sudo -i
cd /sys/bus/hid/drivers/razerkbd/0003:1532:0224.0005/
cat version
cat device_mode
cat device_serial
cat device_type
cat firmware_version
car matrix_brightness
```

juliancoleman, 2017-04-13:
`cat version`

```
1.1.9
```

`cat device_mode`

```
0:0
```

`cat device_serial`

```
BY1711A34901954
```

`cat device_type`

```
Razer Blade (Late 2016)
```

`cat firmware_version`

```
v1.0#
```

`cat matrix_brightness`

```
255
```

z3ntu, 2017-04-13:
Sure, that there is a hash after the firmware_version?

[49 more replies]

closed 2017-04-15 by terrycain