openrazer/openrazer #215: Custom effects on Razer Mamba do not work

tildearrow, 2017-02-25:
Doing the following:

```
echo -ne "\x00\x00\x0e\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff" > matrix_custom_frame
echo -ne "0" > matrix_effect_custom
```

should set the mouse to white. However, it has no effect, and instead displays a message in the kernel logs:

```
razermouse: Command failed. Start Marker: 03 id: 80 Num Params: 46 Reserved: 03 Command: 0b Params: ff00000effffffffffffffffffffffff .
```

So I decided to capture packets for this mouse under a Windows VM, and it seems like the Razer guys decided to use a different packet format for mice.

I don't know if anything can be done about it, but here is the possible spec:

The packet ID seems to be `0c` rather than `0b`, and its arguments most likely are:

```C
struct args {
  unsigned char start;
  unsigned char end;
  unsigned char values[3*(end-start+1)];
}
```

If packet captures are needed I think I can provide them. See you.

terrycain, 2017-02-25:
@tildearrow Yeah attach packet captures to the thread and ill look into it, i've seen 0x0c come up a few times

tildearrow, 2017-02-25:
Done.

tildearrow, 2017-02-26:
Closing as I added a pull request.

closed 2017-02-26 by tildearrow