openrazer/openrazer #217: Deathadder chroma RZ01-0121 support

vbujanca, 2017-02-28:
Hi again,

Could you please tell me whether this mouse is supported and if so how to make it work. It is recognized under razercommander and polychromatic (however the latter says next to it new device found unlike for my razer keyboard) but there is no settings/effect box.

I have restarted razer-service

here is some more info:
lsusb
`Bus 001 Device 007: ID 1532:0221 Razer USA, Ltd `

razer-service log

```
385:2017-02-28 08:25:36 | razer                          | INFO     | Found device.1: 0003:1532:0043.0017
2386-2017-02-28 08:25:36 | razer.device1                  | INFO     | Initialising device.1 RazerDeathadderChroma
2387-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding getSerial method to DBus
2388-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.misc.suspendDevice method to DBus
2389-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.misc.getDeviceMode method to DBus
2390-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.misc.getRazerUrls method to DBus
2391-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.misc.setDeviceMode method to DBus
2392-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.misc.resumeDevice method to DBus
2393-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.misc.getVidPid method to DBus
2394-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.misc.version method to DBus
2395-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.misc.hasDedicatedMacroKeys method to DBus
2396-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.misc.getFirmware method to DBus
2397-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.misc.getMatrixDimensions method to DBus
2398-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.misc.hasMatrix method to DBus
2399-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.misc.getDeviceName method to DBus
2400-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.misc.getDeviceType method to DBus
2401-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.lighting.logo.setLogoActive method to DBus
2402-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.lighting.logo.getLogoActive method to DBus
2403-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.lighting.logo.getLogoEffect method to DBus
2404-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.lighting.logo.getLogoBrightness method to DBus
2405-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.lighting.logo.setLogoBrightness method to DBus
2406-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.lighting.logo.setLogoStatic method to DBus
2407-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.lighting.logo.setLogoPulsate method to DBus
2408-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.lighting.logo.setLogoBlinking method to DBus
2409-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.lighting.logo.setLogoSpectrum method to DBus
2410-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.lighting.scroll.setScrollActive method to DBus
2411-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.lighting.scroll.getScrollActive method to DBus
2412-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.lighting.scroll.getScrollEffect method to DBus
2413-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.lighting.scroll.getScrollBrightness method to DBus
2414-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.lighting.scroll.setScrollBrightness method to DBus
2415-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.lighting.scroll.setScrollStatic method to DBus
2416-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.lighting.scroll.setScrollPulsate method to DBus
2417-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.lighting.scroll.setScrollBlinking method to DBus
2418-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.lighting.scroll.setScrollSpectrum method to DBus
2419-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.dpi.getDPI method to DBus
2420-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.dpi.setDPI method to DBus
2421-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.misc.getPollRate method to DBus
2422-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Adding razer.device.misc.setPollRate method to DBus
2423-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | DBus call set_logo_brightness
2424-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Sending observer message: ('effect', RazerDeathadderChroma:xxxxxxx, 'setBrightness', 255)
2425-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | DBus call set_scroll_brightness
2426-2017-02-28 08:25:36 | razer.device1                  | DEBUG    | Sending observer message: ('effect', RazerDeathadderChroma:xxxxxxx, 'setBrightness', 255)
```

ls -al /sys/bus/hid/drivers/razermouse/

```
total 0
drwxr-xr-x 2 root root    0 Feb 28 08:24 .
drwxr-xr-x 5 root root    0 Feb 28 07:56 ..
lrwxrwxrwx 1 root root    0 Feb 28 08:42 0003:1532:0043.0017 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/0003:1532:0043.0017
lrwxrwxrwx 1 root root    0 Feb 28 08:42 0003:1532:0043.0018 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.1/0003:1532:0043.0018
lrwxrwxrwx 1 root root    0 Feb 28 08:42 0003:1532:0043.0019 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.2/0003:1532:0043.0019
--w------- 1 root root 4096 Feb 28 08:13 bind
lrwxrwxrwx 1 root root    0 Feb 28 08:42 module -> ../../../../module/razermouse
--w------- 1 root root 4096 Feb 28 08:42 new_id
--w------- 1 root root 4096 Feb 28 08:42 uevent
--w------- 1 root root 4096 Feb 28 08:42 unbind
```

Lemme know if you want me to post d-feet content, it looks good to me though. Thanks for any help on this.

z3ntu, 2017-02-28:
1. Yes, the DeathAdder Chroma is supported and works.
2. Please report issues for polychromatic & razercommander in the respective repositories.
3. The usb id you sent (`1532:0221`) is for the BlackWidow Chroma V2, for that one see #184 (if you can't wait for the next release of razer-drivers, I can send you instructions on how to install the package from git-master if you tell me your distro)

vbujanca, 2017-02-28:
Thanks for your reply. 
Indeed the usb id I sent is wrong, the keyboard works fine.

In that case, if the problem resides with the applications I will try there instead. Cheers

closed 2017-02-28 by vbujanca