openrazer/openrazer #218: Orbweaver (non-chroma)

RD777, 2017-03-02:
The current version of the driver does not seem to work with my Orbweaver non-Chroma

I have installed the driver on arch. The driver is loaded:

```
$ lsmod | grep razer
razerkbd               45056  0
hid                   114688  3 hid_generic,usbhid,razerkbd
usbcore               208896  8 usbhid,razerkbd,usb_storage,ehci_hcd,xhci_pci,uas,xhci_hcd,ehci_pci
```

However, the daemon seems to exit after this exception:

```
$ /usr/sbin/razer-service --foreground --verbose --run-dir ~/.razer-service/data --log-dir ~/.razer-service/logs --pid-file ~/.razer-service/daemon.pid     
2017-03-02 11:20:39 | razer                          | INFO     | Initialising Daemon (v1.1.8). Pid: 5368
2017-03-02 11:20:39 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2012 --------------------- (1532:010D)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderChroma ------------------- (1532:0043)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderElite -------------------- (1532:005C)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerFireFly ---------------------------- (1532:0C00)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerImperiator ------------------------- (1532:002F)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerKrakenV2 --------------------------- (1532:0510)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaTE ---------------------- (1532:0046)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexV2 -------------------------- (1532:0050)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaver -------------------------- (1532:0113)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerOrnata ----------------------------- (1532:021F)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerOrnataChroma ----------------------- (1532:021E)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerOrochiWired ------------------------ (1532:0048)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerOuroboros -------------------------- (1532:0032)
2017-03-02 11:20:39 | razer                          | DEBUG    | Loaded device specification: RazerTartarus --------------------------- (1532:0208)
2017-03-02 11:20:39 | razer                          | INFO     | Found device.0: 0003:1532:0113.0003
2017-03-02 11:20:39 | razer.device0                  | INFO     | Initialising device.0 RazerOrbweaver
Traceback (most recent call last):
  File "/usr/sbin/razer-service", line 71, in <module>
    run()
  File "/usr/sbin/razer-service", line 65, in run
    daemonize(**daemon_args)
  File "/usr/lib/python3.6/site-packages/razer_daemon/daemon.py", line 114, in daemonize
    daemon = RazerDaemon(verbose, log_dir, console_log, run_dir, config_file, test_dir=test_dir)
  File "/usr/lib/python3.6/site-packages/razer_daemon/daemon.py", line 217, in __init__
    self._load_devices(first_run=True)
  File "/usr/lib/python3.6/site-packages/razer_daemon/daemon.py", line 430, in _load_devices
    razer_device = device_class(sys_path, device_number, self._config, testing=self._test_dir is not None, additional_interfaces=sorted(additional_interfaces))
  File "/usr/lib/python3.6/site-packages/razer_daemon/hardware/keyboards.py", line 100, in __init__
    super(RazerOrbweaver, self).__init__(*args, **kwargs)
  File "/usr/lib/python3.6/site-packages/razer_daemon/hardware/device_base.py", line 60, in __init__
    self.serial = self.get_serial()
  File "/usr/lib/python3.6/site-packages/razer_daemon/hardware/device_base.py", line 194, in get_serial
    with open(serial_path, 'r') as serial_file:
PermissionError: [Errno 13] Permission denied: '/sys/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.2/0003:1532:0113.0003/device_serial'
Exception ignored in: <object repr() failed>
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/razer_daemon/hardware/device_base.py", line 412, in __del__
    self.close()
  File "/usr/lib/python3.6/site-packages/razer_daemon/hardware/device_base.py", line 325, in close
    if not self._is_closed:
AttributeError: 'RazerOrbweaver' object has no attribute '_is_closed'
```

z3ntu, 2017-03-02:
> PermissionError: [Errno 13] Permission denied: '/sys/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.2/0003:1532:0113.0003/device_serial'

Are you in the `plugdev` group?

RD777, 2017-03-02:
..... omg.

Sorry about that yes, that's it.

closed 2017-03-02 by RD777