openrazer/openrazer #219: Error install

gersontk, 2017-03-04:
I followed the tutorial, but the error occurred:

Errors were encountered while processing:
 /var/cache/apt/archives/razer-kernel-modules-dkms_1.1.8-0ubuntu1_amd64.deb
 /var/cache/apt/archives/python3-razer_1.1.8-0ubuntu1_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

what should I do??

terrycain, 2017-03-04:
what linux distribution?
What commands did you run?

z3ntu, 2017-03-04:
There should be more output with the actual error before that what you sent.

gersontk, 2017-03-04:
Linux Mint 17.2 Rafaela

i try:
sudo add-apt-repository ppa:terrz/razerutils
 sudo apt update
 sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc

not success...

sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
Reading package lists... Done
Building dependency tree       
Reading state information... Done
razer-doc is already the newest version.
razer-daemon is already the newest version.
The following NEW packages will be installed:
  python3-razer razer-kernel-modules-dkms
0 upgraded, 2 newly installed, 0 to remove and 21 not upgraded.
1 not fully installed or removed.
Need to get 0 B/54,5 kB of archives.
After this operation, 447 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
(Reading database ... 234622 files and directories currently installed.)
Preparing to unpack .../razer-kernel-modules-dkms_1.1.8-0ubuntu1_amd64.deb ...
Unpacking razer-kernel-modules-dkms (1.1.8-0ubuntu1) ...
dpkg: error processing archive /var/cache/apt/archives/razer-kernel-modules-dkms_1.1.8-0ubuntu1_amd64.deb (--unpack):
 trying to overwrite '/usr/src/razer_chroma_driver-1.0.0/dkms.conf', which is also in package razer-chroma-driver 1.0.0
Preparing to unpack .../python3-razer_1.1.8-0ubuntu1_all.deb ...
Unpacking python3-razer (1.1.8-0ubuntu1) ...
dpkg: error processing archive /var/cache/apt/archives/python3-razer_1.1.8-0ubuntu1_all.deb (--unpack):
 trying to overwrite '/usr/lib/python3/dist-packages/razer/__init__.py', which is also in package razer-chroma-driver 1.0.0
Errors were encountered while processing:
 /var/cache/apt/archives/razer-kernel-modules-dkms_1.1.8-0ubuntu1_amd64.deb
 /var/cache/apt/archives/python3-razer_1.1.8-0ubuntu1_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

z3ntu, 2017-03-04:
Uninstall the package  razer-chroma-driver

closed 2017-03-04 by z3ntu

gersontk, 2017-03-04:
not working.

 
Setting up razer-kernel-modules-dkms (1.1.9-0ubuntu1) ...
Removing old razer_chroma_driver-1.0.0 DKMS files...
Loading new razer_chroma_driver-1.0.0 DKMS files...
Error! DKMS tree already contains: razer_chroma_driver-1.0.0
You cannot add the same module/version combo more than once.
dpkg: error processing package razer-kernel-modules-dkms (--configure):
 subprocess installed post-installation script returned error exit status 3
dpkg: dependency problems prevent configuration of razer-daemon:
 razer-daemon depends on razer-kernel-modules-dkms; however:
  Package razer-kernel-modules-dkms is not configured yet.

dpkg: error processing package razer-daemon (--configure):
 dependency problems - leaving unconfigured
E: Sub-process /usr/bin/dpkg returned an error code (1)

gersontk, 2017-03-04:
polychromatic not work, razer orochi stop work...

i execute sudo modprobe razerkbd nothing happens

lah7, 2017-03-04:
Apt is trying to install the correct driver/daemon, but the old one is still in the way. Try this:
```
sudo dpkg --configure -a
sudo apt remove python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
sudo apt purge razer_chroma_driver
sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
```

Once it's installed properly, restart the computer, then the apps and daemon should start to work.

gersontk, 2017-03-04:
sudo dpkg --configure -a
Setting up razer-kernel-modules-dkms (1.1.9-0ubuntu1) ...
Removing old razer_chroma_driver-1.0.0 DKMS files...
Loading new razer_chroma_driver-1.0.0 DKMS files...
Error! DKMS tree already contains: razer_chroma_driver-1.0.0
You cannot add the same module/version combo more than once.
dpkg: error processing package razer-kernel-modules-dkms (--configure):
 subprocess installed post-installation script returned error exit status 3
dpkg: dependency problems prevent configuration of razer-daemon:
 razer-daemon depends on razer-kernel-modules-dkms; however:
  Package razer-kernel-modules-dkms is not configured yet.

dpkg: error processing package razer-daemon (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 razer-kernel-modules-dkms
 razer-daemon

gersontk, 2017-03-04:
Installed on another computer did not have error but does not work the polychromatic controler

z3ntu, 2017-03-04:
`sudo dkms remove razer_chroma_driver/1.0.0`

gersontk, 2017-03-04:
Nothing works I give up...

sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  python3-razer razer-daemon razer-doc razer-kernel-modules-dkms
0 upgraded, 4 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/106 kB of archives.
After this operation, 793 kB of additional disk space will be used.
Selecting previously unselected package python3-razer.
(Reading database ... 234517 files and directories currently installed.)
Preparing to unpack .../python3-razer_1.1.9-0ubuntu1_all.deb ...
Unpacking python3-razer (1.1.9-0ubuntu1) ...
Selecting previously unselected package razer-kernel-modules-dkms.
Preparing to unpack .../razer-kernel-modules-dkms_1.1.9-0ubuntu1_amd64.deb ...
Unpacking razer-kernel-modules-dkms (1.1.9-0ubuntu1) ...
Selecting previously unselected package razer-daemon.
Preparing to unpack .../razer-daemon_1.1.9-0ubuntu1_amd64.deb ...
Unpacking razer-daemon (1.1.9-0ubuntu1) ...
Selecting previously unselected package razer-doc.
Preparing to unpack .../razer-doc_1.1.9-0ubuntu1_all.deb ...
Unpacking razer-doc (1.1.9-0ubuntu1) ...
Processing triggers for man-db (2.6.7.1-1ubuntu1) ...
Setting up python3-razer (1.1.9-0ubuntu1) ...
Setting up razer-kernel-modules-dkms (1.1.9-0ubuntu1) ...
Removing old razer_chroma_driver-1.0.0 DKMS files...
Loading new razer_chroma_driver-1.0.0 DKMS files...
Error! DKMS tree already contains: razer_chroma_driver-1.0.0
You cannot add the same module/version combo more than once.
dpkg: error processing package razer-kernel-modules-dkms (--configure):
 subprocess installed post-installation script returned error exit status 3
dpkg: dependency problems prevent configuration of razer-daemon:
 razer-daemon depends on razer-kernel-modules-dkms; however:
  Package razer-kernel-modules-dkms is not configured yet.

dpkg: error processing package razer-daemon (--configure):
 dependency problems - leaving unconfigured
Setting up razer-doc (1.1.9-0ubuntu1) ...
Errors were encountered while processing:
 razer-kernel-modules-dkms
 razer-daemon
E: Sub-process /usr/bin/dpkg returned an error code (1)

gersontk, 2017-03-04:
sudo dkms remove razer_chroma_driver/1.0.0
Error! Invalid number of parameters passed.
Usage: remove <module>/<module-version> --all
   or: remove <module>/<module-version> -k <kernel-version>

terrycain, 2017-03-04:
`dpkg --get-selections | grep razer`

gersontk, 2017-03-04:
dpkg --get-selections | grep razer
python3-razer					install
razer-chroma-driver				install
razer-daemon					install
razer-doc					install
razer-kernel-modules-dkms			install

that it correct?
because polichromatic dont open it yet

terrycain, 2017-03-04:
nope, `razer-chroma-driver` shouldnt be there,

Firstly,
```
sudo apt-get remove -y python3-razer razer-chroma-driver razer-daemon razer-doc razer-kernel-modules-dkms
```

Next run an `sudo apt-get update` and paste the output on pastebin

gersontk, 2017-03-04:
what the means output pastebin..

terrycain, 2017-03-04:
http://pastebin.com/

gersontk, 2017-03-04:
sudo apt-get remove -y python3-razer razer-chroma-driver razer-daemon razer-doc razer-kernel-modules-dkms
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package 'razer-chroma-driver' is not installed, so not removed
The following packages were automatically installed and are no longer required:
  libxdo3 python3-evdev python3-notify2 python3-numpy python3-pyudev
  python3-setproctitle xautomation xdotool
Use 'apt-get autoremove' to remove them.
The following packages will be REMOVED:
  python3-razer razer-daemon razer-doc razer-kernel-modules-dkms
0 upgraded, 0 newly installed, 4 to remove and 0 not upgraded.
2 not fully installed or removed.
After this operation, 793 kB disk space will be freed.
(Reading database ... 234609 files and directories currently installed.)
Removing razer-daemon (1.1.9-0ubuntu1) ...
Removing razer-kernel-modules-dkms (1.1.9-0ubuntu1) ...
Removing python3-razer (1.1.9-0ubuntu1) ...
Removing razer-doc (1.1.9-0ubuntu1) ...
Processing triggers for man-db (2.6.7.1-1ubuntu1) ...

terrycain, 2017-03-04:
> Next run `sudo apt-get update` and paste the output on pastebin

gersontk, 2017-03-04:
sudo apt-get update
[sudo] password for samurai: 
Ign http://dl.google.com stable InRelease
Hit http://repo.steampowered.com precise InRelease                             
Hit http://dl.google.com stable Release.gpg                                    
Hit http://repo.steampowered.com precise/steam Sources                         
Hit http://repo.steampowered.com precise/steam amd64 Packages                  
Hit http://dl.google.com stable Release                                        
Hit http://repo.steampowered.com precise/steam i386 Packages                   
Hit http://dl.google.com stable/main amd64 Packages                            
Ign http://archive.ubuntu.com rafaela InRelease                                
Hit http://security.ubuntu.com trusty-security InRelease                       
Ign http://packages.linuxmint.com rafaela InRelease                            
Ign http://extra.linuxmint.com rafaela InRelease                               
Hit https://repo.skype.com stable InRelease                                    
Ign http://archive.canonical.com trusty InRelease                              
Hit https://repo.skype.com stable/main amd64 Packages                          
Hit http://ppa.launchpad.net trusty InRelease                                  
Hit http://packages.linuxmint.com rafaela Release.gpg                          
Ign http://archive.ubuntu.com trusty InRelease                                 
Hit http://extra.linuxmint.com rafaela Release.gpg                             
Ign http://repo.vivaldi.com stable InRelease                                   
Hit http://security.ubuntu.com trusty-security/main amd64 Packages             
Hit http://archive.canonical.com trusty Release.gpg                            
Ign http://ppa.launchpad.net trusty InRelease                                  
Hit http://archive.ubuntu.com trusty-updates InRelease                         
Hit http://security.ubuntu.com trusty-security/restricted amd64 Packages       
Hit http://archive.canonical.com trusty Release                                
Ign http://dl.google.com stable/main Translation-en_US                         
Hit http://packages.linuxmint.com rafaela Release                              
Ign http://ppa.launchpad.net trusty InRelease                                  
Hit http://extra.linuxmint.com rafaela Release                                 
Ign http://dl.google.com stable/main Translation-en                            
Ign http://repo.vivaldi.com stable InRelease                                   
Ign http://archive.ubuntu.com rafaela Release.gpg                              
Hit http://security.ubuntu.com trusty-security/universe amd64 Packages         
Hit http://archive.canonical.com trusty/partner amd64 Packages                 
Hit http://repo.vivaldi.com stable Release.gpg                                 
Hit http://ppa.launchpad.net trusty InRelease                                  
Get:1 https://repo.skype.com stable/main Translation-en_US                     
Hit http://archive.ubuntu.com trusty Release.gpg                               
Hit http://security.ubuntu.com trusty-security/multiverse amd64 Packages       
Hit http://packages.linuxmint.com rafaela/main amd64 Packages                  
Hit http://repo.vivaldi.com stable Release.gpg                                 
Hit http://extra.linuxmint.com rafaela/main amd64 Packages                     
Hit http://archive.canonical.com trusty/partner i386 Packages                  
Hit http://ppa.launchpad.net trusty InRelease                                  
Hit http://repo.vivaldi.com stable Release                                     
Hit http://archive.ubuntu.com trusty-updates/main amd64 Packages               
Hit http://security.ubuntu.com trusty-security/main i386 Packages              
Hit http://archive.canonical.com trusty/partner Translation-en                 
Hit http://repo.vivaldi.com stable Release                                     
Hit http://ppa.launchpad.net trusty InRelease                                  
Hit http://packages.linuxmint.com rafaela/upstream amd64 Packages              
Hit http://archive.ubuntu.com trusty-updates/restricted amd64 Packages         
Hit http://security.ubuntu.com trusty-security/restricted i386 Packages        
Hit http://extra.linuxmint.com rafaela/main i386 Packages                      
Hit http://repo.vivaldi.com stable/main amd64 Packages                         
Hit http://ppa.launchpad.net trusty InRelease                                  
Hit http://archive.ubuntu.com trusty-updates/universe amd64 Packages           
Hit http://repo.vivaldi.com stable/main i386 Packages                          
Hit http://security.ubuntu.com trusty-security/universe i386 Packages          
Hit http://packages.linuxmint.com rafaela/import amd64 Packages                
Hit http://ppa.launchpad.net trusty InRelease                                  
Hit http://repository.spotify.com stable InRelease                             
Hit http://archive.ubuntu.com trusty-updates/multiverse amd64 Packages         
Hit http://security.ubuntu.com trusty-security/multiverse i386 Packages        
Ign http://ppa.launchpad.net trusty InRelease                                  
Hit http://archive.ubuntu.com trusty-updates/main i386 Packages                
Hit http://repository.spotify.com stable/non-free amd64 Packages               
[... 168 lines cut ...]
[... cut ...]
W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/rafaela/multiverse/binary-i386/Packages  404  Not Found [IP: 91.189.88.162 80]

W: Failed to fetch http://ppa.launchpad.net/artfwo/ppa/ubuntu/dists/trusty/main/source/Sources  404  Not Found

W: Failed to fetch http://ppa.launchpad.net/artfwo/ppa/ubuntu/dists/trusty/main/binary-amd64/Packages  404  Not Found

W: Failed to fetch http://ppa.launchpad.net/artfwo/ppa/ubuntu/dists/trusty/main/binary-i386/Packages  404  Not Found

E: Some index files failed to download. They have been ignored, or old ones used instead.

terrycain, 2017-03-04:
Ok now follow https://terrycain.github.io/razer-drivers/#ubuntu

gersontk, 2017-03-04:
error contenue.

sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  python3-razer razer-daemon razer-doc razer-kernel-modules-dkms
0 upgraded, 4 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/106 kB of archives.
After this operation, 793 kB of additional disk space will be used.
Selecting previously unselected package python3-razer.
(Reading database ... 234519 files and directories currently installed.)
Preparing to unpack .../python3-razer_1.1.9-0ubuntu1_all.deb ...
Unpacking python3-razer (1.1.9-0ubuntu1) ...
Selecting previously unselected package razer-kernel-modules-dkms.
Preparing to unpack .../razer-kernel-modules-dkms_1.1.9-0ubuntu1_amd64.deb ...
Unpacking razer-kernel-modules-dkms (1.1.9-0ubuntu1) ...
Selecting previously unselected package razer-daemon.
Preparing to unpack .../razer-daemon_1.1.9-0ubuntu1_amd64.deb ...
Unpacking razer-daemon (1.1.9-0ubuntu1) ...
Selecting previously unselected package razer-doc.
Preparing to unpack .../razer-doc_1.1.9-0ubuntu1_all.deb ...
Unpacking razer-doc (1.1.9-0ubuntu1) ...
Processing triggers for man-db (2.6.7.1-1ubuntu1) ...
Setting up python3-razer (1.1.9-0ubuntu1) ...
Setting up razer-kernel-modules-dkms (1.1.9-0ubuntu1) ...
Removing old razer_chroma_driver-1.0.0 DKMS files...
Loading new razer_chroma_driver-1.0.0 DKMS files...
Error! DKMS tree already contains: razer_chroma_driver-1.0.0
You cannot add the same module/version combo more than once.
dpkg: error processing package razer-kernel-modules-dkms (--configure):
 subprocess installed post-installation script returned error exit status 3
dpkg: dependency problems prevent configuration of razer-daemon:
 razer-daemon depends on razer-kernel-modules-dkms; however:
  Package razer-kernel-modules-dkms is not configured yet.

dpkg: error processing package razer-daemon (--configure):
 dependency problems - leaving unconfigured
Setting up razer-doc (1.1.9-0ubuntu1) ...
Errors were encountered while processing:
 razer-kernel-modules-dkms
 razer-daemon
E: Sub-process /usr/bin/dpkg returned an error code (1)

terrycain, 2017-03-04:
ok try
```
apt-get remove -y razer-kernel-modules-dkms
rm --vrf /usr/src/razer_chroma_driver-1.0.0
apt-get install -y razer-kernel-modules-dkms
```

gersontk, 2017-03-04:
sudo apt-get install -y razer-kernel-modules-dkms
Reading package lists... Done
Building dependency tree       
Reading state information... Done
razer-kernel-modules-dkms is already the newest version.
The following packages were automatically installed and are no longer required:
  libxdo3 python3-evdev python3-notify2 python3-pyudev python3-setproctitle
  xautomation xdotool
Use 'apt-get autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up razer-kernel-modules-dkms (1.1.9-0ubuntu1) ...
Removing old razer_chroma_driver-1.0.0 DKMS files...
Error! Could not locate dkms.conf file.
File:  does not exist.
First Installation: checking all kernels...
Building only for 3.16.0-38-generic
Error! Could not locate dkms.conf file.
File:  does not exist.
Building initial module for 3.16.0-38-generic
Error! Could not locate dkms.conf file.
File:  does not exist.
-e 
ADD YOUR USER TO THE PLUGDEV GROUP 'sudo gpasswd -a <USERNAME> plugdev'

terrycain, 2017-03-04:
ok thats progress, try this to see if it makes dkms any happier
```
apt-get remove -y razer-kernel-modules-dkms
apt-get install -y razer-kernel-modules-dkms
```

gersontk, 2017-03-04:
not work...

terrycain, 2017-03-04:
output...

gersontk, 2017-03-04:
sudo apt-get install -y razer-kernel-modules-dkms
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libxdo3 python3-evdev python3-notify2 python3-pyudev python3-setproctitle
  xautomation xdotool
Use 'apt-get autoremove' to remove them.
The following NEW packages will be installed:
  razer-kernel-modules-dkms
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/39,7 kB of archives.
After this operation, 388 kB of additional disk space will be used.
Selecting previously unselected package razer-kernel-modules-dkms.
(Reading database ... 234538 files and directories currently installed.)
Preparing to unpack .../razer-kernel-modules-dkms_1.1.9-0ubuntu1_amd64.deb ...
Unpacking razer-kernel-modules-dkms (1.1.9-0ubuntu1) ...
Setting up razer-kernel-modules-dkms (1.1.9-0ubuntu1) ...
Removing old razer_chroma_driver-1.0.0 DKMS files...
Loading new razer_chroma_driver-1.0.0 DKMS files...
Error! DKMS tree already contains: razer_chroma_driver-1.0.0
You cannot add the same module/version combo more than once.
dpkg: error processing package razer-kernel-modules-dkms (--configure):
 subprocess installed post-installation script returned error exit status 3
Errors were encountered while processing:
 razer-kernel-modules-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)

terrycain, 2017-03-04:
try
```
ls /var/lib/initramfs-tools | sudo xargs -n1 /usr/lib/dkms/dkms_autoinstaller start
```

gersontk, 2017-03-04:
ls /var/lib/initramfs-tools | sudo xargs -n1 /usr/lib/dkms/dkms_autoinstaller start
 * dkms: running auto installation service for kernel 3.16.0-38-generic                                                                   [ OK ]

terrycain, 2017-03-04:
```
sudo modprobe razerkbd
sudo lsmod | grep razer
```

gersontk, 2017-03-04:
sudo lsmod | grep razer
razerkbd               25116  0 
hid                   110426  3 hid_generic,razerkbd,usbhid

terrycain, 2017-03-04:
```
sudo apt-get install -y python3-razer razer-daemon razer-doc
```
Then reboot

gersontk, 2017-03-04:
why this error?

sudo apt-get install -y python3-razer razer-daemon razer-doc
Reading package lists... Done
Building dependency tree       
Reading state information... Done
razer-doc is already the newest version.
python3-razer is already the newest version.
The following NEW packages will be installed:
  razer-daemon
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
Need to get 0 B/44,5 kB of archives.
After this operation, 297 kB of additional disk space will be used.
Selecting previously unselected package razer-daemon.
(Reading database ... 234565 files and directories currently installed.)
Preparing to unpack .../razer-daemon_1.1.9-0ubuntu1_amd64.deb ...
Unpacking razer-daemon (1.1.9-0ubuntu1) ...
Processing triggers for man-db (2.6.7.1-1ubuntu1) ...
Setting up razer-kernel-modules-dkms (1.1.9-0ubuntu1) ...
Removing old razer_chroma_driver-1.0.0 DKMS files...
Loading new razer_chroma_driver-1.0.0 DKMS files...
Error! DKMS tree already contains: razer_chroma_driver-1.0.0
You cannot add the same module/version combo more than once.
dpkg: error processing package razer-kernel-modules-dkms (--configure):
 subprocess installed post-installation script returned error exit status 3
dpkg: dependency problems prevent configuration of razer-daemon:
 razer-daemon depends on razer-kernel-modules-dkms; however:
  Package razer-kernel-modules-dkms is not configured yet.

dpkg: error processing package razer-daemon (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 razer-kernel-modules-dkms
 razer-daemon
E: Sub-process /usr/bin/dpkg returned an error code (1)

terrycain, 2017-03-04:
try
```
sudo dkms remove -m razer_chroma_driver -v 1.0.0 --all
```
if that works
```
dpkg-reconfigure razer-kernel-modules-dkms
```

gersontk, 2017-03-04:
sudo dpkg-reconfigure razer-kernel-modules-dkms
/usr/sbin/dpkg-reconfigure: razer-kernel-modules-dkms is broken or not fully installed

terrycain, 2017-03-04:
```
dkms status
```

gersontk, 2017-03-04:
dkms status
bbswitch, 0.7, 3.16.0-38-generic, x86_64: installed
ndiswrapper, 1.59, 3.16.0-38-generic, x86_64: installed (WARNING! Diff between built and installed module!)
nvidia-346, 346.72, 3.16.0-38-generic, x86_64: installed
razer_chroma_driver, 1.0.0: added
virtualbox-guest, 4.3.36, 3.16.0-38-generic, x86_64: installed (original_module exists)

terrycain, 2017-03-04:
```
apt-get remove -y razer-kernel-modules-dkms
dkms status
```

gersontk, 2017-03-04:
dkms status
bbswitch, 0.7, 3.16.0-38-generic, x86_64: installed
ndiswrapper, 1.59, 3.16.0-38-generic, x86_64: installedError! Could not locate dkms.conf file.
File:  does not exist.
 (WARNING! Diff between built and installed module!)
nvidia-346, 346.72, 3.16.0-38-generic, x86_64: installed

terrycain, 2017-03-04:
ok so now dkms _should_ be happy
```
apt-get install -y razer-kernel-modules-dkms
```

gersontk, 2017-03-04:
nonono...

same error.

sudo apt-get install -y razer-kernel-modules-dkms
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libxdo3 python3-evdev python3-notify2 python3-pyudev python3-setproctitle
  xautomation xdotool
Use 'apt-get autoremove' to remove them.
The following NEW packages will be installed:
  razer-kernel-modules-dkms
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/39,7 kB of archives.
After this operation, 388 kB of additional disk space will be used.
Selecting previously unselected package razer-kernel-modules-dkms.
(Reading database ... 234538 files and directories currently installed.)
Preparing to unpack .../razer-kernel-modules-dkms_1.1.9-0ubuntu1_amd64.deb ...
Unpacking razer-kernel-modules-dkms (1.1.9-0ubuntu1) ...
Setting up razer-kernel-modules-dkms (1.1.9-0ubuntu1) ...
Removing old razer_chroma_driver-1.0.0 DKMS files...
Loading new razer_chroma_driver-1.0.0 DKMS files...
Error! DKMS tree already contains: razer_chroma_driver-1.0.0
You cannot add the same module/version combo more than once.
dpkg: error processing package razer-kernel-modules-dkms (--configure):
 subprocess installed post-installation script returned error exit status 3
Errors were encountered while processing:
 razer-kernel-modules-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)

terrycain, 2017-03-04:
sudo modprobe razercore
sudo lsmod | grep razer

gersontk, 2017-03-04:
sudo modprobe razercore
modprobe: FATAL: Module razercore not found.

terrycain, 2017-03-04:
Ok something is broke with dkms, for some reason it wont remove the old razer entry from whatever database it has

gersontk, 2017-03-04:
thanks for help.

gersontk, 2017-03-04:
I will reinstall mint and try again later

gersontk, 2017-03-05:
I installed in ubuntu it run right, in mint it did not work why?

z3ntu, 2017-03-05:
It's some problem on your machine (dkms issue right now). There were some other problems before that we fixed yesterday.

gersontk, 2017-03-05:
se eu tentar o processo novamente vai funcionar será? em linux mint

z3ntu, 2017-03-05:
Afaik nobody here speaks Portuguese...

gersontk, 2017-03-05:
sorry.
If I try the process again it will work will it? In linux mint

z3ntu, 2017-03-05:
So do you still have your Mint installation from yesterday? If yes, install Teamviewer (if you know a better software we can use that) and I will try to fix it...

z3ntu, 2017-03-05:
But you have to at least try to be as cooperative as possible 😉

MGThePro, 2017-06-04:
Is there any fix for that problem? I still can't fix it.

z3ntu, 2017-06-04:
@MGThePro 

* What distro are you running?  
* Please post the output of `sudo dkms status razer_chroma_driver/1.0.0`

MGThePro, 2017-06-04:
1. Ubuntu Mate 16.04
2.razer_chroma_driver, 1.0.0: added

z3ntu, 2017-06-04:
Please post the output of `python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"` and of `sudo modprobe razerkbd`.

MGThePro, 2017-06-04:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 175, in activate_name_owner
    return self.get_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 361, in get_name_owner
    's', (bus_name,), **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.razer': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer/client/__init__.py", line 20, in __init__
    self._dbus = session_bus.get_object("org.razer", "/org/razer")
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 248, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 180, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 278, in start_service_by_name
    'su', (bus_name, flags)))
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.razer was not provided by any .service files

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/razer/client/__init__.py", line 22, in __init__
    raise DaemonNotFound("Could not connect to daemon")
razer.client.DaemonNotFound: Could not connect to daemon

sudo modprobe razerkbd doesn't output anything

z3ntu, 2017-06-04:
Quick question: Did you rebooted already after installing?
If no, please do it, otherwise please run `razer-service -F -v`

[29 more replies]