openrazer/openrazer #221: Razer Ouroboros - 'Configure your Device' page empty

aaronkmiller, 2017-03-07:
The device and serial number is detected but if I click on its tab there are no configuration options. Also all the options in the tray icon popup menu are grayed out.

I tried rebooting and restarting the daemon but it did not help.

    [aaronm@wk1 ~]$ lsusb | grep Razer
    Bus 005 Device 003: ID 1532:0032 Razer USA, Ltd 

```
[aaronm@wk1 ~]$ cat ~/.razer-service/logs/razer.log
2017-03-06 18:44:09 | razer                          | INFO     | Initialising Daemon (v1.1.9). Pid: 1306
2017-03-06 18:44:09 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2012 --------------------- (1532:010D)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderChroma ------------------- (1532:0043)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderElite -------------------- (1532:005C)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerFireFly ---------------------------- (1532:0C00)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerImperiator ------------------------- (1532:002F)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerKrakenV2 --------------------------- (1532:0510)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaTE ---------------------- (1532:0046)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerNagaHex ---------------------------- (1532:0041)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexV2 -------------------------- (1532:0050)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaver -------------------------- (1532:0113)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerOrnata ----------------------------- (1532:021F)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerOrnataChroma ----------------------- (1532:021E)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerOrochiWired ------------------------ (1532:0048)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerOuroboros -------------------------- (1532:0032)
2017-03-06 18:44:09 | razer                          | DEBUG    | Loaded device specification: RazerTartarus --------------------------- (1532:0208)
2017-03-06 18:44:09 | razer                          | INFO     | Found device.0: 0003:1532:0032.0004
2017-03-06 18:44:09 | razer.device0                  | INFO     | Initialising device.0 RazerOuroboros
2017-03-06 18:44:09 | razer.device0                  | DEBUG    | Adding getSerial method to DBus
2017-03-06 18:44:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.suspendDevice method to DBus
2017-03-06 18:44:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceMode method to DBus
2017-03-06 18:44:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.getRazerUrls method to DBus
2017-03-06 18:44:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.setDeviceMode method to DBus
2017-03-06 18:44:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.resumeDevice method to DBus
2017-03-06 18:44:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.getVidPid method to DBus
2017-03-06 18:44:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.version method to DBus
2017-03-06 18:44:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.hasDedicatedMacroKeys method to DBus
2017-03-06 18:44:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.getFirmware method to DBus
2017-03-06 18:44:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.getMatrixDimensions method to DBus
2017-03-06 18:44:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.hasMatrix method to DBus
2017-03-06 18:44:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceName method to DBus
2017-03-06 18:44:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceType method to DBus
2017-03-06 18:44:09 | razer.device0                  | DEBUG    | Adding razer.device.dpi.getDPI method to DBus
2017-03-06 18:44:09 | razer.device0                  | DEBUG    | Adding razer.device.dpi.setDPI method to DBus
2017-03-06 18:44:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.getPollRate method to DBus
[... cut ...]
2017-03-06 18:59:49 | razer.device0                  | DEBUG    | DBus call get_device_name
2017-03-06 18:59:49 | razer.device0                  | DEBUG    | DBus call get_device_type
2017-03-06 18:59:49 | razer.device0                  | DEBUG    | DBus call get_firmware
2017-03-06 18:59:49 | razer.device0                  | DEBUG    | DBus call version
2017-03-06 18:59:49 | razer.device0                  | DEBUG    | DBus call has_matrix
2017-03-06 18:59:49 | razer.device0                  | DEBUG    | DBus call has_matrix
```

terrycain, 2017-03-07:
This sounds like an app related issue, please go to their respective repos on github. Sounds like your referring to polychromatic - https://github.com/lah7/polychromatic

closed 2017-03-07 by terrycain

lah7, 2017-03-07:
Definitely sounds like Polychromatic. It's awaiting a new release. In the meantime, you can try the latest `master` version which adds support for `logo` and ` scroll_wheel`.

aaronkmiller, 2017-03-08:
Thanks @lah7 I installed 0.3.7 and it lets me adjust the dpi and brightness now. The lighting effects dropdown is empty but I think the Ouroboros may not support those?

Sorry for the noise.