openrazer/openrazer #223: Late 2016 individual key RGB not working

lezed1, 2017-03-07:
It appears that individual keys cannot be set on the Razer Blade Stealth (Late 2016) - id `1532:0220`. If I set the custom colors in Windows and then attempt to set any individual color in Linux, the last custom Windows colors appear (it seems to go to the correct mode) but no key can be updated. Additionally, enabling Ripple mode has the same effect - the keyboard reverts to the last custom state from Windows (This was mentioned in [#214](https://github.com/terrycain/razer-drivers/issues/214#issuecomment-284703711)).

I have a packet capture from a Windows 10 VM. I've selected ~10 seconds of it so it's of manageable size. The colors are static (but through the Chroma Configurator so they are constantly sent). The top (fn, row id `0x0`) row is all color `0x333333`, row `0x1` set to `0x444444`, down to `0x888888`.
[row-by-row-packets-filtered.pcapng.zip](https://github.com/terrycain/razer-drivers/files/824315/row-by-row-packets-filtered.pcapng.zip)

Let me know if you want any more information or different captures. Also, I am running Arch Linux.

terrycain, 2017-03-07:
@lezed1 "Should" be fine

lezed1, 2017-03-07:
@terrycain What do you mean?

terrycain, 2017-03-08:
I meant what you've done should be ok, haven't had a chance to look at it yet but seeing as you can see some rgb in the capture I reckon it'll be fine.

terrycain, 2017-03-09:
what distro you on?

lezed1, 2017-03-09:
Arch

terrycain, 2017-03-09:
@z3ntu instructions on how to get branch `bug_lighting_0x0220` plz.

lezed1, 2017-03-09:
I can build off that branch, sure

terrycain, 2017-03-09:
build, make sure that specific version of the driver is running and that the devices are bound to it.

normally just
make
sudo rmmod razerkbd
sudo rmmod razerkbd
sudo insmod ./driver/razerkbd.ko

(by this point the daemon will be unhappy as all the files in /sys/bus/hid/drivers/razerkbd/*/ are owned by root)
sudo chgrp plugdev /sys/bus/hid/drivers/razerkbd/0*/*

z3ntu, 2017-03-09:
You probably will get some dependency issues because some packages needed are only available in the AUR -> install them first (AUR helper or whatever you like)
**Run the following commands:**
```
git clone https://aur.archlinux.org/razer-drivers-git.git
cd razer-drivers-git
curl https://gist.githubusercontent.com/z3ntu/28cb958ce9a778de0975e54c54b77318/raw/3149acc7d7e046adf33846f65d514d74d516b651/PKGBUILD.bug0220 -o PKGBUILD
makepkg -sr
sudo pacman -U python-razer-git-*.tar* razer-daemon-git-*.tar* razer-driver-dkms-git-*.tar*
```
If you need additional help, you can reach me on Telegram @z3ntu.

lezed1, 2017-03-10:
This worked correctly on a0caf3993d37c4996a78a0d35cf6784811909a73.

lezed1, 2017-04-08:
This actually works properly on master! This script (https://gist.github.com/terrycain/c51cb364529d9586514f8201d4ea9dd9) fixed the problem, not the changes on the bugfix branch.

closed 2017-04-08 by lezed1

terrycain, 2017-04-08:
@lezed1 :-)

terrycain, 2017-04-11:
Will get the changes into the daemon soonish then :smile: