openrazer/openrazer #224: Razer Taipan (1532:0034) Support

lah7, 2017-03-08:
Another mice wants to join the driver/daemon:

**Razer Taipan**

* ID `1532:0034`
* Model No. `RZ01-0078`
* Product No. `RZ01-00780100`

-----------
**`lsusb -d 1532:0034`**:

```
Bus 003 Device 006: ID 1532:0034 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0034 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

------------
**`ls -lah /dev/input/by-id/`**
```
drwxr-xr-x 2 root root 360 Mar  8 20:20 .
drwxr-xr-x 4 root root 740 Mar  8 20:20 ..
lrwxrwxrwx 1 root root   9 Mar  8 20:20 usb-Razer_Razer_Taipan-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 Mar  8 20:20 usb-Razer_Razer_Taipan-if01-event-kbd -> ../event8
lrwxrwxrwx 1 root root  10 Mar  8 20:20 usb-Razer_Razer_Taipan-if02-event-kbd -> ../event25
lrwxrwxrwx 1 root root   9 Mar  8 20:20 usb-Razer_Razer_Taipan-mouse -> ../mouse1
```

-----------
[image]

[image]

* Device only has green, no RGB.
* Logo actually pulsates when **on**.

Here's some captures from Synapse, it's a pretty simple mouse:

* Setting DPI to 1000 then 2500
* Setting polling rate from 125 to 500 to 1000 (only three options available)
* Toggle logo off, then on again
* Toggle scroll wheel off, then on again

[razer-taipan-pcapng-captures.tar.gz](https://github.com/terrycain/razer-drivers/files/828912/razer-taipan-pcapng-captures.tar.gz)

Thanks @terrycain :+1:

z3ntu, 2017-03-08:
I can't even post any part of my default response 😞

terrycain, 2017-03-09:
@lah7 try out - https://drive.google.com/open?id=0BydYgD2LzDUCbkg4OTRMLUhkV2s

lah7, 2017-03-09:
Thanks @terrycain :+1: -- almost everything works -- just one thing -- setting the `scroll_wheel` active changes the logo instead -- happens both from the daemon and driver (D-Feet)

terrycain, 2017-04-15:
Packages are building in launchpad, will be available soon

closed 2017-04-15 by terrycain