openrazer/openrazer #226: Failed to toggle logo for Razer Naga Hex V2

simonszu, 2017-03-09:
Hi,

for some reason, Polychromatic shows a UI element for toggling the logo of my mouse. But it doesn't seem to be able to toggle it - it starts as "switched off", although the logo on the mouse is lighted up, and after klicking it, it reports the unableness to toggle the logo. 

I suppose, Polychromatic just uses an API which your driver package offers, so i am reporting this issue here. I don't know if the Naga Hex V2 really has the proper hardware support to switch off the logo, but if it has, your driver is somehow not working - and if it hasn't, then your driver reports the wrong feature set of the hardware. 

And...the color selection only affects the color around the button circle at the thumb position. Is this the desired behaviour, or does the logo and scroll wheel light also have RGB capabilites, and should change as well?

z3ntu, 2017-03-09:
daemon output (`~/.razer-service/logs/razer.log`) and polychromatic (open it in a terminal) output, please :)

simonszu, 2017-03-09:
Of course. Here's the daemon output: http://pastebin.com/LNEYe1TR
And that's the terminal output of polychromatic after trying to switch the logo on (the UI switch does indeed switch to the ON position, and stays there) and trying to switch the logo off:
```
 $ polychromatic-controller 
 ** Preference 'brightness' in 'PM1625H00503024' non-existent. Using default 'None' instead.
Failed to toggle logo for Razer Naga Hex V2.
Exception: can't set attribute
Failed to toggle logo for Razer Naga Hex V2.
Exception: can't set attribute
```
If Razer could have some internal hardware modifications done, be assured it should be the newest revision - i bought the mouse around 3 hours ago :P
Maybe it's not relevant, but the mouse appears as two devices in the `xinput list` command.

z3ntu, 2017-03-09:
@lah7 Any idea?

simonszu, 2017-03-09:
Some additional info: The mouse does not even appear as two devices to xinput, but even as 4 devices to udev. This is part of the content of my `/dev/input/by-id`
```
lrwxrwxrwx 1 root root 9  9. Mär 20:43 usb-Razer_Razer_Naga_Hex_V2-event-mouse -> ../event9
lrwxrwxrwx 1 root root 9  9. Mär 20:43 usb-Razer_Razer_Naga_Hex_V2-if01-event-kbd -> ../event7
lrwxrwxrwx 1 root root 9  9. Mär 20:43 usb-Razer_Razer_Naga_Hex_V2-if02-event-kbd -> ../event8
lrwxrwxrwx 1 root root 9  9. Mär 20:43 usb-Razer_Razer_Naga_Hex_V2-mouse -> ../mouse1
```
Is this desired behaviour? I am asking, because i tried to customize the thumb buttons - apparently they are registered as a keyboard, and `xev` does not recognize them. A short search on the interned shows that this is the default behaviour of the Razer Nagas. However, i followed the [Arch Wiki](https://wiki.archlinux.org/index.php/extra_keyboard_keys) and tried to get the button IDs via `evtest`. It showed that the if01-event-kbd does not give any commands at all, while the `fuser` command of the if02-event-kbd is in use by `razer-service`. 

So my next question is: Should the razer driver daemon, and polychromatic be able to make custom keybindings to the extra buttons as well? I am a little suspicious, because polychromatic renders a quite big black window, with the effect list on the left side, and an empty space on the right side. And, i tried razercommander as well, but it crashes with `AttributeError: 'RazerMouse' object has no attribute 'macro'
`. 

If the driver daemon and polychromatic don't support custom key mappings (yet?), how could i be able to make custom keymaps if the device file is constantly in use by the driver daemon? Do i have to decide between the driver daemon and custom key bindings? 

Sorry for these questions, i know they aren't really targeting the initial problem, but maybe they show that there is indeed some kind of miscommunication between my OS (it's Arch, btw) and the mouse.

lah7, 2017-03-09:
I can't comment on the driver behaviour, but Polychromatic _should_ display brightness sliders for your mouse (at least, this is what happens under the "fake" driver for testing devices), however the daemon is telling Polychromatic it has `lighting_logo_active` (logo toggle), which overrides `lighting_logo_brightness` (sliders).

You can find out what the daemon supports by opening a terminal and running these commands in `python3`:

```
import razer.client
a = razer.client.DeviceManager()
b = a.devices[0]      # Assuming this is the first and only device
b.capabilities
```

You _should_ be able to set your brightness with:

    b.fx.misc.logo.brightness = 50

But the daemon is saying it supports logo toggle (which it obviously doesn't)  so an attribute error happens:

    b.fx.misc.logo.active = True

-----------------

Polychromatic doesn't have any front-end custom key mappings yet -- I don't think the daemon handles them either at the moment. However, macros in the daemon are being re-written, as far as I'm aware.

simonszu, 2017-03-09:
@lah7 Yep, you're right. 
`b.fx.misc.logo.brightness = <value>` works very well, however `b.fx.misc.logo.active = <value>` results in 
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.6/site-packages/razer/client/fx.py", line 522, in active
    func(True)
  File "/usr/lib/python3.6/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3.6/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownMethod: Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/dbus/service.py", line 654, in _message_cb
    (candidate_method, parent_method) = _method_lookup(self, method_name, interface_name)
  File "/usr/lib/python3.6/site-packages/dbus/service.py", line 246, in _method_lookup
    raise UnknownMethodException('%s is not a valid method of interface %s' % (method_name, dbus_interface))
dbus.exceptions.UnknownMethodException: org.freedesktop.DBus.Error.UnknownMethod: Unknown method: setLogoActive is not a valid method of interface razer.device.lighting.logo
```
So it was a good idea to create this issue here and not on polychromatic's repository. 
Let's hope @terrycain can fix this some time. However, the logo's brightness isn't as important to me as key mappings.

terrycain, 2017-03-09:
@simonszu Will have a look in a bit, I have a hex v2 so will go find it. Key mappings are being actively worked on but not currently available

simonszu, 2017-03-09:
Thanks. That's great news.
I really appreciate your work. The clerks in our local electronic store
actually did recommend the Razer Mouse to me because "there is some kind of
open source driver project". :)

z3ntu, 2017-03-09:
lol

lah7, 2017-03-09:
I see the problem --- you're using an older version of Polychromatic. 

@z3ntu is the [maintainer for Polychromatic's Arch packages](https://aur.archlinux.org/packages/polychromatic/), and it's still on **0.3.6** -- the latest version with improved device support is **0.3.7**. It's not an issue with the daemon.

terrycain, 2017-03-09:
:smile:

simonszu, 2017-03-10:
@lah7 Suddenly: Over 9000 more settings in Polychromatic available. Awesome work.

closed 2017-03-10 by terrycain