openrazer/openrazer #230: Debian issue

artificial-immersion, 2017-03-12:
Compiled from source under Debian.

"sudo modprobe razerkbd" ends in:
ERROR: could not insert 'razerkbd': Exec format error

I used make ubuntu_install and install. Both didn't work.

z3ntu, 2017-03-12:
Do you have it installed with dkms or already-compiled kernel modules?

artificial-immersion, 2017-03-12:
Not sure. I think I have DKMS on my system. But I just used the Makefile with both mentioned instructions. If I should could you link a page or just dump a tiny how to here?

z3ntu, 2017-03-12:
Theoretically you could also build yourself a .deb package. You would have to edit the `debian/control` file and replace `linux-headers-generic` with the kernel header package on debian (dunno how it's called) and then run some magic debian command to build the package. maybe @terrycain can help a bit there (i'm not familiar with debian packaging)

artificial-immersion, 2017-03-12:
linux-headers-generic would be replaced by linux-headers-amd64. But already again ran into some trouble. 

But now ld_cleanup fails with:
make[3]: *** /lib/modules//build: No such file or directory.  Stop

Complet log:
`dpkg-buildpackage: info: source package razer                                   
dpkg-buildpackage: info: source version 1.1.9-0ubuntu1                                                                                                         
dpkg-buildpackage: info: source distribution xenial                             
dpkg-buildpackage: info: source changed by Terry Cain <terry@terrys-home.co.uk> 
 dpkg-source --before-build razer                                               
dpkg-buildpackage: info: host architecture amd64                                
 fakeroot debian/rules clean                                                    
dh clean --with-python3                                                         
   dh_testdir -O--with-python3                                                  
   debian/rules override_dh_auto_clean                                          
make[1]: Entering directory '/tmp/tmp.pNQ7ITJcIx_deb_build_tmp/razer'           
/usr/bin/make lp_driver_clean                                                   
make[2]: Entering directory '/tmp/tmp.pNQ7ITJcIx_deb_build_tmp/razer'           
make -C "/lib/modules//build" SUBDIRS="/tmp/tmp.pNQ7ITJcIx_deb_build_tmp/razer/driver" clean
make[3]: Entering directory '/tmp/tmp.pNQ7ITJcIx_deb_build_tmp/razer'           
make[3]: *** /lib/modules//build: No such file or directory.  Stop.             
make[3]: Leaving directory '/tmp/tmp.pNQ7ITJcIx_deb_build_tmp/razer'            
Makefile:64: recipe for target 'lp_driver_clean' failed                         
make[2]: *** [lp_driver_clean] Error 2                                          
make[2]: Leaving directory '/tmp/tmp.pNQ7ITJcIx_deb_build_tmp/razer'            
debian/rules:20: recipe for target 'override_dh_auto_clean' failed              
make[1]: *** [override_dh_auto_clean] Error 2                                   
make[1]: Leaving directory '/tmp/tmp.pNQ7ITJcIx_deb_build_tmp/razer'            
debian/rules:23: recipe for target 'clean' failed                               
make: *** [clean] Error 2                                                       
dpkg-buildpackage: error: fakeroot debian/rules clean gave error exit status 2
`

artificial-immersion, 2017-03-12:
Ah I see between the double slashes there is the headers missing.

artificial-immersion, 2017-03-12:
this whole "KERNELDIR?=/lib/modules/$(shell uname -r)/build" isn't working at all under debian.

artificial-immersion, 2017-03-12:
Botched it by replacing every line with "/lib/.../build" with the correct directory. If you guys need someone for the debian department, I would offer myself haha.

artificial-immersion, 2017-03-12:
I did all this shebang on the Debian Testing Distro maybe that causes issues. If @terrycain will read this, just leave a comment here I can upload source + compiled .debs on regular base if needed.

z3ntu, 2017-03-17:
I just installed debian testing (`debian-stretch-DI-rc2-amd64-netinst.iso`) in a VM. I then install `git` and `linux-headers-amd64`, cloned the repo, did `make driver` and `insmod driver/razerkbd.ko` and the module shows up in `lsmod`. No idea what the problem with your machine is.

z3ntu, 2017-03-17:
[image]
[image]

If you need help, you can install something like Teamviewer and I'll try to help you (I'm gmt+1 so evening in my timezone works good for me)

z3ntu, 2017-04-03:
@artificial-immersion Are you still interested? Otherwise this issue will be closed.

closed 2017-04-06 by z3ntu