openrazer/openrazer #232: Razer Bade Stealth 2016/early 2017. Trackpad not detected in Debian.

DMJC, 2017-03-16:
Hi, I've got a Razerblade Stealth late 2016. I've noticed Razer have implemented the thunderbolt security patch the community has been recommending to get the core working on Linux. My laptop arrived two days ago, and the trackpad is not being detected under debian. I got the polychromatic driver to work fine with the keyboard, but I can't find any information online about getting the trackpad working. As this is a laptop not having a mouse is frustrating as hell. Here's the information I dumped from Ubuntu and a summary of the software on my laptop:

UBUNTU LIVE USB:
lspci:
00:00.0 Host bridge: Intel Corporation Device 5904 (rev 02)
00:02.0 VGA compatible controller: Intel Corporation Device 5916 (rev 02)
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
00:14.2 Signal processing controller: Intel Corporation Sunrise Point-LP Thermal subsystem (rev 21)
00:15.0 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #0 (rev 21)
00:15.1 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #1 (rev 21)
00:16.0 Communication controller: Intel Corporation Sunrise Point-LP CSME HECI #1 (rev 21)
00:1c.0 PCI bridge: Intel Corporation Device 9d12 (rev f1)
00:1c.4 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #5 (rev f1)
00:1d.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #9 (rev f1)
00:1e.0 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO UART Controller #0 (rev 21)
00:1f.0 ISA bridge: Intel Corporation Device 9d58 (rev 21)
00:1f.2 Memory controller: Intel Corporation Sunrise Point-LP PMC (rev 21)
00:1f.3 Audio device: Intel Corporation Device 9d71 (rev 21)
00:1f.4 SMBus: Intel Corporation Sunrise Point-LP SMBus (rev 21)
01:00.0 Network controller: Qualcomm Atheros QCA6174 802.11ac Wireless Network Adapter (rev 32)
3b:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd Device a804

lsusb:
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 04f3:223c Elan Microelectronics Corp. 
Bus 001 Device 006: ID 1532:0220 Razer USA, Ltd 
Bus 001 Device 005: ID 0bda:579f Realtek Semiconductor Corp. 
Bus 001 Device 004: ID 0cf3:e300 Atheros Communications, Inc. 
Bus 001 Device 003: ID 0461:4d15 Primax Electronics, Ltd Dell Optical Mouse
Bus 001 Device 002: ID 6557:0021  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

lsmod | grep i2c
i2c_designware_platform    16384  0
i2c_designware_core    20480  1 i2c_designware_platform
i2c_algo_bit           16384  1 i915
i2c_hid                20480  0
hid                   118784  5 i2c_hid,hid_generic,usbhid,hid_rmi,hid_multitouch

event devices:
mouse3
event12

synclient:
Device '15320205:00 06CB:5F41':
    Device Enabled (140):    1
    Coordinate Transformation Matrix (142):    1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
    Device Accel Profile (265):    1
    Device Accel Constant Deceleration (266):    2.500000
    Device Accel Adaptive Deceleration (267):    1.000000
    Device Accel Velocity Scaling (268):    12.500000
    Synaptics Edges (303):    163, 3902, 130, 2276
    Synaptics Finger (304):    25, 30, 0
    Synaptics Tap Time (305):    180
    Synaptics Tap Move (306):    207
    Synaptics Tap Durations (307):    180, 100, 100
    Synaptics ClickPad (308):    1
    Synaptics Middle Button Timeout (309):    0
    Synaptics Two-Finger Pressure (310):    282
    Synaptics Two-Finger Width (311):    7
    Synaptics Scrolling Distance (312):    -94, -94
    Synaptics Edge Scrolling (313):    0, 0, 0
    Synaptics Two-Finger Scrolling (314):    1, 1
    Synaptics Move Speed (315):    1.000000, 1.750000, 0.042373, 0.000000
    Synaptics Off (316):    2
    Synaptics Locked Drags (317):    0
    Synaptics Locked Drags Timeout (318):    5000
    Synaptics Tap Action (319):    2, 3, 0, 0, 1, 3, 0
    Synaptics Click Action (320):    1, 3, 0
    Synaptics Circular Scrolling (321):    0
[... 76 lines cut ...]
i2c_hid                20480  0
hid                   122880  5 i2c_hid,hid_generic,usbhid,hid_multitouch,razerkbd

synclient:
Couldn't find synaptics properties. No synaptics driver loaded?

X Server Version:
X.Org X Server 1.19.2
Release Date: 2017-03-02
X Protocol Version 11, Revision 0
Build Operating System: Linux 3.16.0-4-amd64 x86_64 Debian
Current Operating System: Linux James-Work 4.10.0-rc6-amd64 #1 SMP Debian 4.10~rc6-1~exp2 (2017-02-02) x86_64
Kernel command line: BOOT_IMAGE=/boot/vmlinuz-4.10.0-rc6-amd64 root=UUID=f1d89c16-0738-4744-bfbe-588c69a40c50 ro quiet acpi_osi=Darwin
Build Date: 03 March 2017  03:14:41PM
xorg-server 2:1.19.2-1 (https://www.debian.org/support) 

/sys/bus find . | grep 1532
./platform/devices/15320205:00
./hid/devices/0003:1532:0220.0004
./hid/devices/0003:1532:0220.0002
./hid/devices/0003:1532:0220.0003
./hid/drivers/razerkbd/0003:1532:0220.0004
./hid/drivers/razerkbd/0003:1532:0220.0002
./hid/drivers/razerkbd/0003:1532:0220.0003
./acpi/devices/15320205:00

DMJC, 2017-03-16:
Attempting to bind 15320205 to the i2c-hid module fails with error no such device:
/sys/bus/i2c/drivers/i2c_hid# echo -n "15320205\:00" > bind
bash: echo: write error: No such device

DMJC, 2017-03-17:
I request this issue be closed/resolved. I messed around with my grub settings trying to fix my framebuffer and it suddenly came good. Probably an old Apple ACPI setting or something coming over. (this installation was transferred from a macbook pro). All features are now working and the trackpad even has two finger scroll.

closed 2017-03-17 by z3ntu

acca301, 2018-04-11:
Man, you made my day!

For completeness: the command line **must** contain
```shell
acpi_osi='!Windows 2015'
```
to cope with another ACPI quirk, see [#764 in Bumblebee](https://github.com/Bumblebee-Project/Bumblebee/issues/764)