openrazer/openrazer #233: How to make it work?

ShadowSigyn, 2017-03-19:
Firstly I'm really sorry, it will be a very noob-like question but I just don't know any other way to figure it out.
I've installed the driver on Debian testing following your guide.
Everything went flawlessly but how do I activate it? 
My keyboard (blackwidow chroma) went completely dark. Before it at least had default lightning.

Also when I try to install razerCommander:
    dpkg: dependency problems prevent configuration of razercommander:
     razercommander depends on python3-razer; however:
      Package python3-razer is not installed.
     razercommander depends on razer-daemon; however:
      Package razer-daemon is not installed.

Even though the razer-driver package is installed and I restarted PC.

z3ntu, 2017-03-19:
You didn't install a .deb package but the polychromatic & razercommander packages depend on them (technically). I don't know if it's possible to skip dependency checks with apt/dpkg but we should really have binary packages for Debian (@terrycain 😉 ).

You could try to install polychromatic or razercommander manually, which should work.

z3ntu, 2017-04-03:
@ShadowSigyn Are you still interested? Otherwise this issue will be closed.

closed 2017-04-06 by z3ntu