openrazer/openrazer #234: Odd behavior on Arch: Deathadder chroma random movement and stutter occasiounal

krutoileshii, 2017-03-20:
Swapped my install from fedora to Arch (4.10.3-1-ARCH) got the drivers installed and all. however, lately the mouse was behaving oddly such as:

1 - Random mouse movement (on it's own, hand not even on the mouse)
2 - stutters occasionally

DPI is currently set to 3200@1000 (seems to help a bit)
stutters more when around links or haven't been moved.
likely a libinput issue. not sure where to go to diagnose.

below are the messages from when it happens.

`[    2.379278] razermouse: loading out-of-tree module taints kernel.
[    2.597206] razermouse 0003:1532:0043.0003: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer DeathAdder Chroma] on usb-0000:00:14.0-13/input1
[    2.597235] razermouse 0003:1532:0043.0002: input,hidraw2: USB HID v1.11 Mouse [Razer Razer DeathAdder Chroma] on usb-0000:00:14.0-13/input0
[    2.657242] razermouse 0003:1532:0043.0004: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer DeathAdder Chroma] on usb-0000:00:14.0-13/input2
[    3.107351] razerkbd 0003:1532:0214.0007: input,hidraw4: USB HID v1.11 Mouse [Razer Razer BlackWidow Ultimate 2016] on usb-0000:00:14.0-14/input2
[    3.163885] razerkbd 0003:1532:0214.0005: input,hidraw5: USB HID v1.11 Keyboard [Razer Razer BlackWidow Ultimate 2016] on usb-0000:00:14.0-14/input0
[    3.267140] razerkbd 0003:1532:0214.0006: input,hidraw6: USB HID v1.11 Keyboard [Razer Razer BlackWidow Ultimate 2016] on usb-0000:00:14.0-14/input1
[  259.731886] razer driver: Device data transfer failed.
[  259.740797] razer driver: Invalid USB repsonse. USB Report length: -32
[  259.740799] razermouse: Invalid Report Length. Start Marker: 00 id: 00 Num Params: 00 Reserved: 00 Command: 00 Params: 00000000000000000000000000000000 .
[  259.742228] razer driver: Device data transfer failed.
[  259.743665] razer driver: Device data transfer failed.
[  259.745085] razer driver: Device data transfer failed.
[  259.746563] razermouse: Response doesnt match request. Start Marker: 02 id: ff Num Params: 03 Reserved: 03 Command: 03 Params: 0104ff00000000000000000000000000 .
[  259.746583] razer driver: Device data transfer failed.
[  259.748011] razermouse: Response doesnt match request. Start Marker: 02 id: ff Num Params: 03 Reserved: 03 Command: 03 Params: 0104ff00000000000000000000000000 .
[  259.748055] razer driver: Device data transfer failed.
[  259.749466] razermouse: Response doesnt match request. Start Marker: 02 id: ff Num Params: 03 Reserved: 03 Command: 03 Params: 0104ff00000000000000000000000000 .
[  259.749531] razer driver: Device data transfer failed.
[  259.750902] razermouse: Response doesnt match request. Start Marker: 02 id: ff Num Params: 03 Reserved: 03 Command: 03 Params: 0104ff00000000000000000000000000 .`

terrycain, 2017-03-20:
Very weird
To rule out an Arch issue, have you tried it on another linux distro?

krutoileshii, 2017-03-20:
Not yet. But I will as soon as I get home. Have a laptop with fedora sitting here as well as as raspbian

krutoileshii, 2017-03-21:
Tried on my fedora laptop, so far no issues. What could be causing it on arch?

z3ntu, 2017-03-21:
What kernel does your Fedora install have (version)?

krutoileshii, 2017-03-22:
Fedora is on 4.9.11-200.fc25.x86_64.

What else would help diagnose this?

z3ntu, 2017-03-22:
So you were running 4.9.x on Arch and had that issue? I hoped it was the 4.10 kernel upgrade...

krutoileshii, 2017-03-22:
No 4.9 is good 4.1 is bad. I have 4.9 on fedora install I tested on to see if it's arch specific

terrycain, 2017-03-22:
So its not an issue with the driver :smiley: :+1:

krutoileshii, 2017-03-22:
Doesn't look like it so far but could be for a newer kernel  Some setting maybe

krutoileshii, 2017-03-22:
Also I meant 4.10

krutoileshii, 2017-03-24:
Should I text LTS kernel on arch just to make sure? It's 4.9.17

z3ntu, 2017-03-24:
Yeah that would be good.

krutoileshii, 2017-03-25:
LTS kernel wouldn't even boot.... I'll give it another go maybe on Sunday.

krutoileshii, 2017-03-26:
Ok, installed elementary instead of gnome, and it seems to not happen. but haven't used it enough to say for sure. Let me test it for a few days and i will let you know.

krutoileshii, 2017-03-30:
Installed latest libinput will report back the results soon

krutoileshii, 2017-04-05:
Ok, still persists. i have a feeling this has to do with the kernel or libinput as it stutters now as well especially in non-full-screen applications.

krutoileshii, 2017-04-08:
Testing latest kernel. Will report in a bit

krutoileshii, 2017-04-23:
I think i found a solution to this one... in a form of a $9 mouse pad which is odd as it wasn't doing this before that on fedora. Can you please close this one for now.

tildearrow, 2017-04-23:
That means your mouse's lens could be getting dirty. Try cleaning the lens up, then try again?

krutoileshii, 2017-04-23:
Just did. So far so good. Thanks

z3ntu, 2017-04-23:
😀

closed 2017-04-23 by z3ntu