openrazer/openrazer #235: Razer Tartarus (1532:0201) Support

torusnc, 2017-03-20:
**Razer Deathadder 3500**

`ID 1532:0201`
`Model No. RZ07-0103`
`Product No. RZ01-01030100`

**lsusb -d 1532:0201 -v**

```
Bus 001 Device 003: ID 1532:0201 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0201 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```

**ls -lah /dev/input/by-id/**

```
drwxr-xr-x 2 root root 200 Mar 20 11:45 .
drwxr-xr-x 4 root root 560 Mar 20 11:45 ..
lrwxrwxrwx 1 root root   9 Mar 20 10:49 usb-Razer_Razer_Tartarus-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 Mar 20 10:49 usb-Razer_Razer_Tartarus-if01-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 Mar 20 10:49 usb-Razer_Razer_Tartarus-if02-event-mouse -> ../event6
lrwxrwxrwx 1 root root   9 Mar 20 10:49 usb-Razer_Razer_Tartarus-if02-mouse -> ../mouse0
```

z3ntu, 2017-03-20:
Please post screenshots of all available options (eg lighting) in Synapse! (I also edited your post to change DeathAdder 3500 to Tartarus 😉 )

terrycain, 2017-03-20:
^^ packet captures too

z3ntu, 2017-06-11:
@RecursiveProgrammer
* Please read [this](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol) because we probably need packet captures.
* Screenshots of all available options on Razer Synapse are needed.

RecursiveProgrammer, 2017-06-11:
Do you really want screenshots of all of the options, or would you like a video? There are a lot of options, but I can run the video I made through ffmpeg to generate the jpg's, if you prefer.

terrycain, 2017-06-11:
@RecursiveProgrammer sure video would be fine

RecursiveProgrammer, 2017-06-11:
Do you want the packet captures in csv format, or does it matter?

terrycain, 2017-06-11:
@RecursiveProgrammer preferably in raw pcapng form would be most useful

RecursiveProgrammer, 2017-06-11:
Here is the video (mkv was not allowed, so I zipped it). My goodness, the packet capture process is daunting! Is there any specific thing that you would like me to focus on? I did the changes to dim the light and I am working on doing changes to key bindings to one of the keypad keys. I will do a keybind changes to the three side buttons. Once I get that far, I will probably take a break and see if you need me to go further with the packet captures.

By the way: I never new wireshark captured usb transmissions (I thought it was only for network traffic), so I am learning something new :-)

[tartarus.zip](https://github.com/terrycain/razer-drivers/files/1066725/tartarus.zip)

terrycain, 2017-06-11:
Ok here goes

* 1 capture of changing keymaps (so the keymap leds change)
* 1 capture of switching between pulsate and brightness
* 1 capture of sliding brightness up and down

That covers all the hardware related options

RecursiveProgrammer, 2017-06-11:
Well...I guess I was doing more than I needed to :-) I will only upload the three that you are requesting. I did a capture of changing all of the keymaps, a capture of switching between pulsate and brightness, and a capture of sliding the brightness up and down. Beyond that, I did separate captures of each of them (for example: one is changing the dim feature from zero to 29 and another capture is from 29 to zero). I was not sure if the separate captures might be helpful.

$ lsusb | grep -i "razer"
Bus 004 Device 003: ID 1532:0201 Razer USA, Ltd 

My filter was:

usb.bus_id == 4 && usb.device_address == 3

[relevant_ones.zip](https://github.com/terrycain/razer-drivers/files/1066752/relevant_ones.zip)

terrycain, 2017-06-11:
@RecursiveProgrammer 👍 Will have a look soonish

terrycain, 2017-06-16:
What distro you on?

RecursiveProgrammer, 2017-06-16:
Arch Linux. I would send you the uname -a, but cox communication did not see it fit to fix a cable that was obstructing traffic and someone (after two hours) hit it and snapped the line... #winning, so I am replying from my phone: I will update this as soon as I get Internet back, if you need the output.

z3ntu, 2017-06-16:
You probably will get some dependency issues because some packages needed are only available in the AUR -> install them first (AUR helper or whatever you like)
**Run the following commands:**
```
git clone https://aur.archlinux.org/razer-drivers-git.git
cd razer-drivers-git
curl https://gist.githubusercontent.com/z3ntu/28cb958ce9a778de0975e54c54b77318/raw/2ab8b01357d1e0759daf850146ec422aec5a1a57/PKGBUILD.tartarus -o PKGBUILD
makepkg -sri
```
If you need additional help, you can reach me on Telegram @z3ntu.

RecursiveProgrammer, 2017-06-17:
I believe this is what you are looking for (if not, please let me know what I should respond with to let you know that it is working):

Here is what I get, now:

$ sudo journalctl -b | grep -i razer_mount
Jun 16 18:14:01 Desk razer_mount[520]: Driver razerkbd
Jun 16 18:14:01 Desk razer_mount[521]: Driver razerkbd
Jun 16 18:14:01 Desk razer_mount[519]: Driver razerkbd
Jun 16 18:14:01 Desk razer_mount[523]: Device_ID 0003:1532:0201.0003
Jun 16 18:14:01 Desk razer_mount[522]: Device_ID 0003:1532:0201.0001
Jun 16 18:14:01 Desk razer_mount[525]: Unbinding 0003:1532:0201.0001 from hid-generic
Jun 16 18:14:01 Desk razer_mount[524]: Device_ID 0003:1532:0201.0002
Jun 16 18:14:01 Desk razer_mount[526]: Unbinding 0003:1532:0201.0003 from hid-generic
Jun 16 18:14:01 Desk razer_mount[527]: Unbinding 0003:1532:0201.0002 from hid-generic
Jun 16 18:14:01 Desk razer_mount[541]: Binding 0003:1532:0201.0001 to razerkbd
Jun 16 18:14:01 Desk razer_mount[543]: Binding 0003:1532:0201.0003 to razerkbd
Jun 16 18:14:01 Desk razer_mount[548]: Binding 0003:1532:0201.0002 to razerkbd
Jun 16 18:14:02 Desk razer_mount[556]: Finished binding 0003:1532:0201.0001
Jun 16 18:14:02 Desk razer_mount[557]: Finished binding 0003:1532:0201.0002
Jun 16 18:14:02 Desk razer_mount[555]: Finished binding 0003:1532:0201.0003
Jun 16 18:14:02 Desk razer_mount[560]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0201.0001/
Jun 16 18:14:02 Desk razer_mount[558]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0201.0002/
Jun 16 18:14:02 Desk razer_mount[559]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0201.0003/
Jun 16 18:14:02 Desk razer_mount[566]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0201.0003/
Jun 16 18:14:02 Desk razer_mount[565]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0201.0002/
Jun 16 18:14:02 Desk razer_mount[567]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0201.0001/

I am currently looking to find one of the front end apps to see if they will let me map keys, then probably open a report with them :-)

RecursiveProgrammer, 2017-06-17:
Hmm, razer-daemon is throwing an error, though. Shall I open a new ticket or append it to this one?

z3ntu, 2017-06-17:
Please don't open a new ticket support for support. (If you want, you can join our telegram group; message me @z3ntu for that). So yes, please upload the log here.

RecursiveProgrammer, 2017-06-17:
$ razer-daemon -Fv &> razer.out.log

Shows this error:

```
AttributeError: 'RazerTartarus' object has no attribute '_serial'
```

[razer.out.log.txt](https://github.com/terrycain/razer-drivers/files/1082674/razer.out.log.txt)

terrycain, 2017-06-17:
Get the absolute latest changes, should fix

RecursiveProgrammer, 2017-06-17:
I apologize, I think I have wandered off the path.

So I did what z3ntu asked me to do (and saw that error), and posted it. When I saw Terrycain's message, I tried to do a:

```
git pull --rebase
```

Got an error saying that the files were changed. So I removed the directory (I know, I should have renamed it) and tried z3ntu's commands again and get an error on:

```
git clone https://aur.archlinux.org/razer-drivers-git.git
```

saying:

```
$ git clone https://aur.archlinux.org/razer-drivers-git.git
Cloning into 'razer-drivers-git'...
fatal: unable to access 'https://aur.archlinux.org/razer-drivers-git.git/': Failed to connect to aur.archlinux.org port 443: Connection refused

```

Which I believe this worked, before. Anyway, to get back to where I was, I downloaded the feature_tartarus_0x0201 branch and ran:

```
sudo make -s all install
```

and rebooted.

All that said, I still get the error and I am still on the version 1.1.13 of the daemon, which makes me think that I still did not get the latest changes.

What did I need to do differently?

Warm regards.

z3ntu, 2017-06-17:
Haha lol, the AUR is down right now (should be back up in an hour or so). And please don't ever again run sudo make install :/ It just breaks things...

RecursiveProgrammer, 2017-06-17:
@z3ntu, Yes, `sudo make install` was a bad idea. I should have caught that. I was blindly following the [other distribution](https://terrycain.github.io/razer-drivers/#other) installation.

RecursiveProgrammer, 2017-06-17:
Sweet, I think we are getting closer.

Now the error is: _is_closed

```
$ razer-daemon -vF &> ~/razer.out.log.txt
```
[razer.out.log.txt](https://github.com/terrycain/razer-drivers/files/1082829/razer.out.log.txt)

terrycain, 2017-06-18:
Diff /usr/lib/python3.6/site-packages/razer_daemon/hardware/device_base.py
With the version from (master)[https://github.com/terrycain/razer-drivers/blob/master/daemon/razer_daemon/hardware/device_base.py]

RecursiveProgrammer, 2017-06-18:
$ diff /usr/lib/python3.6/site-packages/razer_daemon/hardware/device_base.py ./razer-drivers/daemon/razer_daemon/hardware/device_base.py
49a50,52
>         # Serial cache
>         self._serial = None
> 
111,113d113
< 
<         # Serial cache
<         self._serial = None

terrycain, 2017-06-18:
@RecursiveProgrammer yeah replace the one in /usr/lib with the one from master, basically moved that line so its declared earlier

RecursiveProgrammer, 2017-06-18:
@terrycain That appears to have worked. I am going to try to head over to polychromatic or razorcommander (They currently show the tartarus as an unknown device) to see if I can get the keys remapped.  Thanks for getting it working!

terrycain, 2017-06-18:
@RecursiveProgrammer Yeah key mapping doesnt work too good as of yet, but it is in progress

RecursiveProgrammer, 2017-06-18:
@terrycain Would that be something I might be able to help with, here, or at the application (like polychromatic) level?

DreadPirateLynx, 2017-07-01:
Has any of the progress made on this issue made its way into the ppa yet? I just installed the drivers today, and the device isn't being bound to the driver.

terrycain, 2017-07-01:
thought I merged it but looks like I didnt. Will merge and it'll either be released tomorrow or next weekend. Shame as i did a release earlier today

DreadPirateLynx, 2017-07-09:
Still doing my best to patiently await the release with this fix

terrycain, 2017-07-09:
Yeah sorry mate, I meant to release things this weekend but got side tracked, will try and get it out this week

DreadPirateLynx, 2017-07-09:
I know how it is. Really appreciate the work you do

DreadPirateLynx, 2017-07-26:
Still waiting on this. Tried to manually compile the source, but I've been running into problems and can't get it installed that way.

commit f285ba08 (2017-07-30): Merge branch 'master' into stable

Possibly fixes https://github.com/lah7/polychromatic/issues/104
Close #235
Close #144
closed 2017-07-30 by terrycain