openrazer/openrazer #236: Razer Blackwidow Ultimate Stealth 2016 support

ven3k, 2017-03-21:
Hey, I see that you support BW Ultimate 2016, can you also add the "Stealth" version to your driver?
I use Razer keyboard a lot, and I hate reconnecting it to my Windows PC just to change lighting :/

z3ntu, 2017-03-21:
Please post the output of `lsusb | grep 1532`.

lah7, 2017-03-21:
I have the BlackWidow Ultimate **Stealth** 2016 (`1532:0214`) as does @terrycain, it's supported.

Someone correct me if I'm wrong, but it's the same hardware, it's just that the key caps are different on the Stealth version (green/orange).

ven3k, 2017-03-22:
well, in my case every program for controlling LEDs says it's not supported by the driver o.O

z3ntu, 2017-03-22:
@ven3k so the USB ID is `1532:0214`? Please post the output of `ls -al /sys/bus/hid/drivers/razerkbd/`.

ven3k, 2017-03-22:
```
drwxr-xr-x 2 root root    0 mar 22 19:49 .
drwxr-xr-x 4 root root    0 mar 20 21:42 ..
lrwxrwxrwx 1 root root    0 mar 22 19:49 0003:1532:0214.0007 -> ../../../../devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.0/0003:1532:0214.0007
lrwxrwxrwx 1 root root    0 mar 22 19:49 0003:1532:0214.0008 -> ../../../../devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.1/0003:1532:0214.0008
lrwxrwxrwx 1 root root    0 mar 22 19:49 0003:1532:0214.0009 -> ../../../../devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.2/0003:1532:0214.0009
--w------- 1 root root 4096 mar 22 19:49 bind
lrwxrwxrwx 1 root root    0 mar 22 19:49 module -> ../../../../module/razerkbd
--w------- 1 root root 4096 mar 22 19:49 new_id
--w------- 1 root root 4096 mar 20 21:53 uevent
--w------- 1 root root 4096 mar 22 19:49 unbind
```

Yep, USB ID is 1532:0214

z3ntu, 2017-03-22:
The device is bound so the frontends should work properly except there's some issue with the daemon. 
Make sure the daemon is running (eg `ps -aux | grep razer`) and please upload the contents of `~/.razer-service/logs/razer.log`.

ven3k, 2017-03-22:
oh, wait...
restarting daemon fixed it (xD), thanks anyway

closed 2017-03-22 by ven3k

z3ntu, 2017-03-22:
Actually the daemon should be auto-started by logging in. If the daemon isn't running the next time you reboot, please write us 👍