openrazer/openrazer #237: Razer Kraken 7.1 Arch Linux

AWildBeard, 2017-03-22:
TLDR;
- Followed the install guide for Arch Linux here https://terrycain.github.io/razer-drivers/
- Used the second method that detailed installing three packages ( razer-driver-dkms, razer-daemon and python-razer ). Installed their dependancies as well
- rebooted with the device plugged in, no joy according to both polychromatic and razercommander
- rebooted with the device unplugged and then plugged in, no joy.
- attempted to search for a mod to bind to the device, no joy in /lib/modules/ with $: find . | grep razer
- made sure to start $: razer-service each time before attempting to do the stuff. Also used
    $: razer-service --verbose --run-dir ~/.razer-service/data --log-dir ~/.razer-service/logs --pid-file ~/.razer-service/daemon.pid
- What I think would go a long way to fixing this issue: knowing the exact name of the kernel mod to probe to get it to bind, as the kernel does not appear to have it loaded. Although if the confidence in the three Arch packages I installed is through the roof, then this is probably not the right approach.

The TLDR is pretty much got it all, but here are some logs and other statistics

**$: lsusb | grep '1532:'**
Bus 001 Device 006: ID 1532:0510 Razer USA, Ltd

**~/.razer-service/logs/razer.log**

2017-03-22 14:06:57 | razer                          | INFO     | Initialising Daemon (v1.1.9). Pid: 1466
2017-03-22 14:06:57 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2012 --------------------- (1532:010D)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderChroma ------------------- (1532:0043)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderElite -------------------- (1532:005C)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerFireFly ---------------------------- (1532:0C00)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerImperiator ------------------------- (1532:002F)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerKrakenV2 --------------------------- (1532:0510)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaTE ---------------------- (1532:0046)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerNagaHex ---------------------------- (1532:0041)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexV2 -------------------------- (1532:0050)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaver -------------------------- (1532:0113)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerOrnata ----------------------------- (1532:021F)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerOrnataChroma ----------------------- (1532:021E)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerOrochiWired ------------------------ (1532:0048)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerOuroboros -------------------------- (1532:0032)
2017-03-22 14:06:57 | razer                          | DEBUG    | Loaded device specification: RazerTartarus --------------------------- (1532:0208)
2017-03-22 14:06:57 | razer                          | INFO     | Adding razer.devices.getDevices method to DBus
2017-03-22 14:06:57 | razer                          | INFO     | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2017-03-22 14:06:57 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
2017-03-22 14:06:57 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
2017-03-22 14:06:57 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
2017-03-22 14:06:57 | razer                          | INFO     | Adding razer.daemon.version method to DBus
2017-03-22 14:06:57 | razer                          | INFO     | Adding razer.daemon.stop method to DBus
2017-03-22 14:06:57 | razer                          | INFO     | Serving DBus
2017-03-22 14:06:58 | razer                          | DEBUG    | DBus called get_serial_list
2017-03-22 14:07:08 | razer                          | DEBUG    | DBus called get_serial_list

_**Replaced all Hashtag comments with dual / for formatting reasons with GitHub**_
**~/.razer-service/razer.conf**

[General]
// Verbose logging (logs debug messages - lotsa spam)
verbose_logging = True

[Startup]
// Set the sync effects flag to true so any assignment of effects will work across devices
sync_effects_enabled = True

// Turn off the devices when the systems screensaver kicks in
devices_off_on_screensaver = True

// Mouse battery notifier
mouse_battery_notifier = True

[Statistics]
// Collects number of keypresses per hour per key used to generate a heatmap
key_statistics = True

z3ntu, 2017-03-22:
First: Do you have the kernel headers installed? Eg `linux-headers` for the stock `linux` kernel.
Second: `sudo dkms status` please.
Third: `ls -al /sys/bus/hid/drivers/{,razerkraken/}` please.

AWildBeard, 2017-03-26:
@z3ntu Sorry, missed an update.

closed 2017-03-26 by AWildBeard