openrazer/openrazer #239: Mouse not supported

Swabbo, 2017-03-22:
Is there any possiblity for added support for my Razer Naga 2014 Left-handed addition? (ID 1532:0040)

Thanks eitherway! Great work by the way :)

z3ntu, 2017-03-23:
Please post the output of
`lsusb -d 1532:0040 -v`
and
`ls -lah /dev/input/by-id/`
and read [this on how to get packet captures](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol
). Also screenshots of all available options on Razer Synapse are needed.

Swabbo, 2017-03-24:
```
 lsusb -d 1532:0040 -v

Bus 005 Device 002: ID 1532:0040 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0040 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```

```
ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 140 Mar 24 00:04 .
drwxr-xr-x 4 root root 480 Mar 24 00:26 ..
lrwxrwxrwx 1 root root   9 Mar 24 00:04 usb-Logitech_USB_Receiver-if02-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 Mar 24 00:04 usb-Razer_Razer_Naga_2014-event-mouse -> ../event2
lrwxrwxrwx 1 root root   9 Mar 24 00:04 usb-Razer_Razer_Naga_2014-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 Mar 24 00:04 usb-Razer_Razer_Naga_2014-if02-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 Mar 24 00:04 usb-Razer_Razer_Naga_2014-mouse -> ../mouse0

```

Swabbo, 2017-03-24:
[I downloaded Windows 7 and installed it using Oracle's VirtualBox, but after installing Razer Synapse twice it just isn't recognising my Naga :s](http://i.gyazo.com/055056814ff6a5555ab815dfc2fcfec8.png)

terrycain, 2017-03-24:
@Swabbo have you passedthrough the usb?

z3ntu, 2017-03-24:
Did you pass through the USB devices (=mouse)?

z3ntu, 2017-04-06:
@Swabbo Did you try again? If you want, we can help you in our Telegram Chat (if you are interested in joining this chat, message me on Telegram @z3ntu) in getting everything to work.

stuffboxing, 2017-04-15:
When i attempted to passthrough the mouse input, the mouse pointer in the guest os freezes, however the actions actually happen (like you can move the mouse right click the desktop and so on) but it's borderline annoying to work with, and my research done for the past hour has not been giving me anything remotely promising,

z3ntu, 2017-04-15:
* You need 2 mice / input devices
* Pass through the Razer mouse
* Use the other mouse to do everything.
* Record the USB traffic to the Razer mouse (from Synapse) with Wireshark

Is that kinda what you did?

stuffboxing, 2017-04-15:
i'll get me a 2nd mouse... i read keyboard in the tutorial and simply brainfarted... big thanks for the quick reply though 👍

stuffboxing, 2017-04-16:
screenshots and pcap in the archive https://www.dropbox.com/s/8oid829s2b07z7f/razer2.tar.gz?dl=0
i hope this is somewhat usefull, if you need anything else let me know

terrycain, 2017-04-16:
DPI sliding and polling rate change in pcapng format would be good. Pretty sure I know what they are but its nicer to have evidence. Also a capture starting before the vm is started lasting all the way until synapse is open would be useful too

terrycain, 2017-04-16:
The DPI one is more useful as the Naga Hex which is 1 number away from yours uses some nasty dpi rubbish, so hopefully yours is better but they're funamentally different so could do with captures of that at least

stuffboxing, 2017-04-16:
You'll have it in about an hour.

terrycain, 2017-04-16:
:smile: No rush, am going bbq in a bit. But I do have the majority of the driver ready I think. What distro you on?
And if you can join telegram (my user @terrycain) and i'll add you to the razer group where most of us hang out, it'll be a bit easier to test in realtime. Otherwise i'll just spam the issue with instructions

stuffboxing, 2017-04-16:
here you go: https://www.dropbox.com/s/7vt58s2txiq3hxe/razerdpipollingstartup.tar.gz?dl=0

terrycain, 2017-04-17:
New release is out, is in launchpad, will be out soon. Non Ubuntu packages will be out soon after.

closed 2017-04-17 by terrycain