openrazer/openrazer #241: Razer Ouroboros - Service won't start

MrMannWood, 2017-03-28:
Mint 18.1 with kernal version 4.8.0-44-generic

    $ sudo razer-service -F
    2017-03-27 20:45:47 | razer                          | INFO     | Initialising Daemon (v1.1.9). Pid: 3162
    2017-03-27 20:45:47 | razer                          | CRITICAL | User is not a member of the plugdev group

Apparently "root" doesn't count as plugdev, so I

    sudo usermod -a -G plugdev root

    $ sudo razer-service -F
    2017-03-27 20:49:33 | razer                          | INFO     | Initialising Daemon (v1.1.9). Pid: 4050
    2017-03-27 20:49:33 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2012 --------------------- (1532:010D)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderChroma ------------------- (1532:0043)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderElite -------------------- (1532:005C)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerFireFly ---------------------------- (1532:0C00)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerImperiator ------------------------- (1532:002F)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerKrakenV2 --------------------------- (1532:0510)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaTE ---------------------- (1532:0046)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerNagaHex ---------------------------- (1532:0041)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexV2 -------------------------- (1532:0050)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaver -------------------------- (1532:0113)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerOrnata ----------------------------- (1532:021F)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerOrnataChroma ----------------------- (1532:021E)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerOrochiWired ------------------------ (1532:0048)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerOuroboros -------------------------- (1532:0032)
    2017-03-27 20:49:33 | razer                          | DEBUG    | Loaded device specification: RazerTartarus --------------------------- (1532:0208)
    2017-03-27 20:49:33 | razer                          | INFO     | Adding razer.devices.getDevices method to DBus
    2017-03-27 20:49:33 | razer                          | INFO     | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
    2017-03-27 20:49:33 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
    2017-03-27 20:49:33 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
    2017-03-27 20:49:33 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
    2017-03-27 20:49:33 | razer                          | INFO     | Adding razer.daemon.version method to DBus
    2017-03-27 20:49:33 | razer                          | INFO     | Adding razer.daemon.stop method to DBus
    2017-03-27 20:49:33 | razer                          | INFO     | Serving DBus
    2017-03-27 20:49:47 | razer                          | DEBUG    | Device event [add]: /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11:1.1/0003:1532:0032.000E
    2017-03-27 20:49:47 | razer                          | DEBUG    | Device event [add]: /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11:1.2/0003:1532:0032.000F
    2017-03-27 20:49:47 | razer                          | DEBUG    | Device event [add]: /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11:1.0/0003:1532:0032.000D
    2017-03-27 20:49:47 | razer                          | INFO     | Found valid device.0: 0003:1532:0032.000D
    2017-03-27 20:49:47 | razer.device0                  | INFO     | Initialising device.0 RazerOuroboros
    Exception in thread device-monitor:
    Traceback (most recent call last):
      File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
        self.run()
      File "/usr/lib/python3/dist-packages/pyudev/monitor.py", line 515, in run
    self._callback(device)
      File "/usr/lib/python3/dist-packages/razer_daemon/daemon.py", line 536, in _udev_input_event
    self._add_device(device)
[... cut ...]
        DBusService.__init__(self, self.BUS_PATH, object_path)
      File "/usr/lib/python3/dist-packages/razer_daemon/dbus_services/service.py", line 85, in __init__
        super(DBusService, self).__init__(bus_object, object_path)
      File "/usr/lib/python3/dist-packages/dbus/service.py", line 455, in __init__
        validate_object_path(object_path)
    ValueError: Invalid object path '/org/razer/device/': ends with '/' and is not just '/'

The lights on the mouse came on, but it still doesn't actually do anything.

Any help greatly appreciated!

z3ntu, 2017-03-28:
1. Never start the daemon with root.
2. Please do `cat /sys/bus/hid/drivers/razer*/0*/device_serial` and send the output here.

terrycain, 2017-03-28:
And does it do that every time or only occasionally?

terrycain, 2017-03-28:
Also have you tried plugging the mouse in then starting daemon after?

MrMannWood, 2017-03-29:
output of command: PM1435077400318

Classic turn it off and back on and let the computer sleep overnight and then it works.

Thanks for being my rubber duck, and sorry for wasting your time. Thanks for all your work making these awesome drivers!

closed 2017-03-29 by MrMannWood

z3ntu, 2017-03-29:
@terrycain That could be caused by the same issue I experienced... 

@MrMannWood Can you unplug and replug the mouse while the daemon is running and look if the same error appears? If yes, you have to restart the daemon that it works properly again.