openrazer/openrazer #243: razer_mount failed with exit code 1

domsch1988, 2017-03-30:
Having a Problem to get this working under Arch 4.10
I installed razer-driver-meta and all dependencys from AUR. On Reboot i get some systemd-modules failing.

The relevant parts from journalctl:
`Mär 30 18:33:40 Arch-Desktop razer_mount[339]: Driver razerkbd
Mär 30 18:33:40 Arch-Desktop razer_mount[341]: Device_ID 0003:1532:0209.0003
Mär 30 18:33:40 Arch-Desktop razer_mount[345]: Modprobing razerkbd
Mär 30 18:33:40 Arch-Desktop razer_mount[340]: Driver razerkbd
Mär 30 18:33:40 Arch-Desktop razer_mount[348]: Driver razerkbd
Mär 30 18:33:40 Arch-Desktop systemd-udevd[206]: Process 'razer_mount razerkbd 0003:1532:0209.0003' failed with exit code 1.
Mär 30 18:33:40 Arch-Desktop razer_mount[352]: Device_ID 0003:1532:0209.0001
Mär 30 18:33:40 Arch-Desktop razer_mount[354]: Device_ID 0003:1532:0209.0002
Mär 30 18:33:40 Arch-Desktop razer_mount[356]: Modprobing razerkbd
Mär 30 18:33:40 Arch-Desktop razer_mount[357]: Modprobing razerkbd
Mär 30 18:33:40 Arch-Desktop systemd-udevd[208]: Process 'razer_mount razerkbd 0003:1532:0209.0001' failed with exit code 1.
Mär 30 18:33:40 Arch-Desktop systemd-udevd[217]: Process 'razer_mount razerkbd 0003:1532:0209.0002' failed with exit code 1.
`

razercommander says no device connected. Any Idea what went wrong?

z3ntu, 2017-03-30:
Please post the output of the following commands

```
sudo modprobe razerkbd
lsmod | grep razer
sudo dkms status
```

domsch1988, 2017-03-30:
This was fast! Thanks!
looks like a problem... modprobe says:

modprobe: FATAL: Module razerkbd not found in directory /lib/modules/4.10.6-1-ARCH

The other two produce no output.

z3ntu, 2017-03-30:
`pacman -Q razer-driver-dkms`

domsch1988, 2017-03-30:
Package not found.
Is the driver available through the official repository?

z3ntu, 2017-03-30:
No it's from the AUR but should be auto-installed when you installed `razer-driver-meta`... Try to reinstall the meta package or just install `razer-driver-dkms`, `razer-daemon` and `python-razer`

domsch1988, 2017-03-30:
Ok, i installed it manually. Going to reboot and report back.

z3ntu, 2017-03-30:
👍

domsch1988, 2017-03-30:
Ok, pacman can now find the driver. Modprobe still can not. Do i have to move it manually to /lib/modules/?

z3ntu, 2017-03-30:
No don't move anything manually :)

`sudo dkms status` please

domsch1988, 2017-03-30:
Still nothing

z3ntu, 2017-03-30:
Do you have your kernel headers installed (eg `linux-headers` for the stock `linux` kernel)?

Try `sudo dkms install razer_chroma_driver/1.0.0`

domsch1988, 2017-03-30:
Jesus... *HeadmeetsDesk*
linux-headers it was ofc
Easy to miss "small" things on a fresh Arch Installation.

You have been of tremendous help sir! I thank you much!

z3ntu, 2017-03-30:
No problem. (I actually added a post-install message to remind people but no problem 🙂)

closed 2017-03-30 by z3ntu