openrazer/openrazer #244: No devices found in Arch Linux

joonatoona, 2017-04-01:
Latest version of `razer-driver-meta` from the AUR.

Doesn't find my keyboard, worked just fine in ubuntu. The instructions linked in the README didn't work either.

z3ntu, 2017-04-01:
The output of `ls -al /sys/bus/hid/drivers/razer*/0*/` and `sudo dkms status` please

joonatoona, 2017-04-01:
`ls -al /sys/bus/hid/drivers/razer*/0*/` outputs: `ls: cannot access '/sys/bus/hid/drivers/razer*/0*/': No such file or directory`

`sudo dkms status` outputs: ``

z3ntu, 2017-04-01:
`sudo dkms install razer_chroma_driver/1.0.0` and the output of that

joonatoona, 2017-04-01:
`~ >>> sudo dkms install razer_chroma_driver/1.0.0

Creating symlink /var/lib/dkms/razer_chroma_driver/1.0.0/source ->
                 /usr/src/razer_chroma_driver-1.0.0

DKMS: add completed.
Error! echo
Your kernel headers for kernel 4.10.5-1-ARCH cannot be found at
/usr/lib/modules/4.10.5-1-ARCH/build or /usr/lib/modules/4.10.5-1-ARCH/source.`

z3ntu, 2017-04-01:
Do you have the kernel headers installed? They are required by dkms! (or you have upgraded the kernel recently and not rebooted since)

joonatoona, 2017-04-01:
Trying to install kernel headers: `error: failed retrieving file 'linux-headers-4.10.5-1-x86_64.pkg.tar.xz' from mirrors.kernel.org : The requested URL returned error: 404`

z3ntu, 2017-04-01:
`sudo pacman -Syu linux-headers` (4.10.6-1 is already out)

joonatoona, 2017-04-01:
Cool, now it finds the keyboard! Thanks!

Should `razer-service` start on boot?

z3ntu, 2017-04-01:
Yes it should be "launched during startup of the user's desktop environment after the user has logged in" (xdg autostart)

joonatoona, 2017-04-01:
Doesn't start for me.

z3ntu, 2017-04-01:
What DE?

joonatoona, 2017-04-01:
i3

z3ntu, 2017-04-01:
(I don't know if i3 follows the xdg autostart spec)

Could you try manually starting it with `razer-service --verbose` in your terminal?

joonatoona, 2017-04-01:
Manually starting it works, but it kills it whenever I close the terminal window.

z3ntu, 2017-04-01:
So no, i3 doesn't follow the xdg autostart spec so I'm gonna send you a systemd user service to use which works fine but isn't included with the package (@terrycain now there is an incentive 😉 )

joonatoona, 2017-04-01:
Wouldn't systemd make more sense? Why is xdg autostart the default?

z3ntu, 2017-04-01:
So put the following contents into the file `/usr/lib/systemd/user/razer-service.service`:

```
[Unit]
Description=Razer Daemon (User)

[Service]
ExecStart=/usr/bin/razer-service --foreground --verbose --run-dir %h/.razer-service/data --log-dir %h/.razer-service/logs --pid-file %h/.razer-service/daemon.pid

[Install]
WantedBy=default.target
```

And then do `systemctl --user enable --now razer-service`

joonatoona, 2017-04-01:
Trying to start `razer-service` results in `Failed to start razer-service.service: Unit razer-service.service not found.`

terrycain, 2017-04-01:
@joonatoona systemctl daemon-reload
then try starting again

z3ntu, 2017-04-01:
try `systemctl --user daemon-reload`

z3ntu, 2017-04-01:
or just try both :D

joonatoona, 2017-04-01:
Still not found :/

z3ntu, 2017-04-01:
`ls -al /usr/lib/systemd/user/razer-service.service`

joonatoona, 2017-04-01:
Sorry, fixed it. File was owned by root for some reason.

joonatoona, 2017-04-01:
Daemon still stops whenever I close the terminal window.

z3ntu, 2017-04-01:
That shouldn't matter...?
`-rw-r--r-- 1 root root 247 23. Feb 19:48 /usr/lib/systemd/user/razer-service.service` works fine

z3ntu, 2017-04-01:
And when the systemd service is started, nothing should stop when you close the terminal window...?

joonatoona, 2017-04-01:
Interesting, starting it only works if it's owned by me.
Starting it with systemctl works, it just stops whenever I close the term.

joonatoona, 2017-04-01:
Interesting, echoing bytes to the keyboard works even if the driver isn't running?!?

z3ntu, 2017-04-01:
The project consist out of basically 3 components. The kernel driver, the daemon (/service) and a python library. The devices get automatically bound to the kernel driver with udev, the daemon should get autostarted with xdg autostart (which doesn't work on i3^^) and the python library which communicates with the daemon and exposes stuff for python.

z3ntu, 2017-04-01:
And `systemctl --user status razer-service` please

joonatoona, 2017-04-01:
~ >>> systemctl --user status razer-service
● razer-service.service - Razer Daemon (User)
   Loaded: loaded (/usr/lib/systemd/user/razer-service.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2017-04-01 12:45:20 PDT; 1min 46s ago
 Main PID: 601 (razer-service)
   CGroup: /user.slice/user-1000.slice/user@1000.service/razer-service.service
           └─601 razer-service

Apr 01 12:45:27 JoonaBox razer-service[601]: 2017-04-01 12:45:27 | razer.device0.keymanager       | DEBUG    | Starting KeyWatcher
Apr 01 12:45:27 JoonaBox razer-service[601]: 2017-04-01 12:45:27 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionali
Apr 01 12:45:27 JoonaBox razer-service[601]: 2017-04-01 12:45:27 | razer                          | INFO     | Adding razer.devices.getDevices method to DBus
Apr 01 12:45:27 JoonaBox razer-service[601]: 2017-04-01 12:45:27 | razer                          | INFO     | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
Apr 01 12:45:27 JoonaBox razer-service[601]: 2017-04-01 12:45:27 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
Apr 01 12:45:27 JoonaBox razer-service[601]: 2017-04-01 12:45:27 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
Apr 01 12:45:27 JoonaBox razer-service[601]: 2017-04-01 12:45:27 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
Apr 01 12:45:27 JoonaBox razer-service[601]: 2017-04-01 12:45:27 | razer                          | INFO     | Adding razer.daemon.version method to DBus
Apr 01 12:45:27 JoonaBox razer-service[601]: 2017-04-01 12:45:27 | razer                          | INFO     | Adding razer.daemon.stop method to DBus
Apr 01 12:45:27 JoonaBox razer-service[601]: 2017-04-01 12:45:27 | razer                          | INFO     | Serving DBus

z3ntu, 2017-04-01:
It's running (for 1min 46s) so no problem there...

joonatoona, 2017-04-01:
It's running, but trying to echo bytes to `matrix_custom_frame` freezes the entire PC.

z3ntu, 2017-04-01:
Well that's because you shouldn't put random bytes into the driver. Use the python library if you want to do something quick otherwise use a frontend like polychromatic

joonatoona, 2017-04-01:
Well, the exact bytes worked in ubuntu. Idk.
I tried using a frontent (razercommander), but that claims the service isn't running. I'll go file an issue there.

z3ntu, 2017-04-01:
Please run `python -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"`

joonatoona, 2017-04-01:
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 175, in activate_name_owner
    return self.get_name_owner(bus_name)
  File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 361, in get_name_owner
    's', (bus_name,), **keywords)
  File "/usr/lib/python3.6/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.razer': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/razer/client/__init__.py", line 20, in __init__
    self._dbus = session_bus.get_object("org.razer", "/org/razer")
  File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python3.6/site-packages/dbus/proxies.py", line 248, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 180, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 278, in start_service_by_name
    'su', (bus_name, flags)))
  File "/usr/lib/python3.6/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.razer was not provided by any .service files

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3.6/site-packages/razer/client/__init__.py", line 22, in __init__
    raise DaemonNotFound("Could not connect to daemon")
razer.client.DaemonNotFound: Could not connect to daemon

joonatoona, 2017-04-01:
Running `killall razer-service && razer-service` first outputs this: `[<RazerKeyboard IO1619143700259>]`

z3ntu, 2017-04-01:
Maybe do `killall razer-service && systemctl --user start razer-service.service && systemctl --user status razer-service.service`. No idea why systemd is behaving apparently weird with your machine.

joonatoona, 2017-04-01:
Running that claims it's running, but `python -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"` has the same error.

z3ntu, 2017-04-01:
wtf
does `journalctl --user-unit razer-service` (scroll to the bottom) show any error (eg python traceback)?

joonatoona, 2017-04-01:
Nope, last line is:
`Apr 01 12:55:18 JoonaBox razer-service[1641]: 2017-04-01 12:55:18 | razer                          | INFO     | Serving DBus`

z3ntu, 2017-04-16:
Did you try this again?

joonatoona, 2017-04-16:
Still doesn't work, but after every reboot there's
```
huhti 01 13:14:08 JoonaBox systemd[623]: Started Razer Daemon (User).
huhti 01 13:14:14 JoonaBox razer-service[629]: Unable to init server: Could not connect: Connection refused```

z3ntu, 2017-04-16:
Okay that seems like a dbus server issue... No idea if that has something to do with i3...

z3ntu, 2017-04-16:
Does `d-feet` work (=eg looks something like this when started)?
[image]

joonatoona, 2017-04-16:
Yeah, it seems to work just fine and show a similar set of results.

(GitHub doesn't want to let me attach scrot)

z3ntu, 2017-04-16:
Okay so just to recap:
* starting the daemon from the terminal works fine.
* xdg-autostart doesn't work with i3
* the systemd service doesn't like the dbus

?

joonatoona, 2017-04-16:
Yes, except you need to kill the daemon first

Apparently

Yes

z3ntu, 2017-04-16:
Can you please try

```
pkill -e -9 razer
systemctl --user start razer-service
python -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
systemctl --user status razer-service
```

joonatoona, 2017-04-16:
```
~/s/s/doc >>> pkill -e -9 razer
razer-service killed (pid 681)
~/s/s/doc >>> systemctl --user start razer-service
~/s/s/doc >>> python -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 175, in activate_name_owner
    return self.get_name_owner(bus_name)
  File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 361, in get_name_owner
    's', (bus_name,), **keywords)
  File "/usr/lib/python3.6/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.razer': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/razer/client/__init__.py", line 20, in __init__
    self._dbus = session_bus.get_object("org.razer", "/org/razer")
  File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python3.6/site-packages/dbus/proxies.py", line 248, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 180, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 278, in start_service_by_name
    'su', (bus_name, flags)))
  File "/usr/lib/python3.6/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.razer was not provided by any .service files

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3.6/site-packages/razer/client/__init__.py", line 22, in __init__
    raise DaemonNotFound("Could not connect to daemon")
razer.client.DaemonNotFound: Could not connect to daemon
~/s/s/doc >>> systemctl --user status razer-service
● razer-service.service - Razer Daemon (User)
   Loaded: loaded (/usr/lib/systemd/user/razer-service.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2017-04-16 08:48:08 PDT; 10s ago
 Main PID: 16279 (razer-service)
   CGroup: /user.slice/user-1000.slice/user@1000.service/razer-service.service
           └─16279 razer-service

huhti 16 08:48:09 JoonaBox razer-service[16279]: 2017-04-16 08:48:09 | razer.device0                  | DEBUG    | Adding razer.device.lighting.custom.setRipple method to DBus
huhti 16 08:48:09 JoonaBox razer-service[16279]: 2017-04-16 08:48:09 | razer.device0                  | DEBUG    | Adding razer.device.lighting.custom.setRippleRandomColour method to DBus
huhti 16 08:48:09 JoonaBox razer-service[16279]: 2017-04-16 08:48:09 | razer.device0.keymanager       | DEBUG    | Starting KeyWatcher
huhti 16 08:48:09 JoonaBox razer-service[16279]: 2017-04-16 08:48:09 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
huhti 16 08:48:09 JoonaBox razer-service[16279]: 2017-04-16 08:48:09 | razer                          | INFO     | Adding razer.devices.getDevices method to DBus
huhti 16 08:48:09 JoonaBox razer-service[16279]: 2017-04-16 08:48:09 | razer                          | INFO     | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
huhti 16 08:48:09 JoonaBox razer-service[16279]: 2017-04-16 08:48:09 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
huhti 16 08:48:09 JoonaBox razer-service[16279]: 2017-04-16 08:48:09 | razer                          | INFO     | Adding razer.daemon.version method to DBus
huhti 16 08:48:09 JoonaBox razer-service[16279]: 2017-04-16 08:48:09 | razer                          | INFO     | Adding razer.daemon.stop method to DBus
huhti 16 08:48:09 JoonaBox razer-service[16279]: 2017-04-16 08:48:09 | razer                          | INFO     | Serving DBus
~/s/s/doc >>> 
```

z3ntu, 2017-04-16:
I really don't understand wtf is going on there...

rudolfschmidt, 2017-05-05:
I have the same env i3 and using arch linux, my daemon is compiled by the official source code from the official razer website, starting razer-service is successful, no problems, buy using a client like polychromatic complains about not finding the device.

z3ntu, 2017-05-05:
[image]
I just set up a virtual machine with i3 and the driver and I got everything to work with the systemd unit found here: https://gist.github.com/z3ntu/da96a5799e951265aa8691f9e7fc557e (location is in the description of the gist).

And also my own frontend for the driver "RazerGenie" works without problems (even though it looks really bad with the default i3 Qt theme).
[image]

rudolfschmidt, 2017-05-05:
I even cant install openrazer-meta from aur, how did u install them?

z3ntu, 2017-05-05:
Can you give me logs?

rudolfschmidt, 2017-05-05:
Sry if it becomes too long
`
$ yaourt -S openrazer-meta

==> Downloading openrazer-meta PKGBUILD from AUR...
x .SRCINFO
x PKGBUILD
x razer-driver-arch.install
x razer-driver-dkms.install
GabMus commented on 2017-01-17 08:41			 
@Wolfiezn @m13253 added python-pyudev as a dependency

z3ntu commented on 2017-01-24 13:43			 
This package will be filled with razer-driver-dkms, razer-daemon  python-razer when the same-named packages are merged here.

GabMus commented on 2017-01-25 20:38			 
I approve of the merge request into the package razer-drivers.

GabMus commented on 2017-01-25 20:38			 
I approve of the merge request into the package razer-drivers.

GabMus commented on 2017-01-25 20:38			 
I approve of the merge request into the package razer-drivers.

openrazer-meta 1.1.12-1  (2017-05-05 17:25)
( Unsupported package: Potentially dangerous ! )
==> Edit PKGBUILD ? [Y/n] ("A" to abort)
==> ------------------------------------
==> n

==> WARNING: This PKGBUILD describes a split package.
==> Specific package options are unknown
==> razer-drivers dependencies:
 - make (already installed) [makedepend]
 - python (already installed) [makedepend]
 - python-setuptools (already installed) [makedepend]

==> Continue building openrazer-meta ? [Y/n]
==> ----------------------------------------
==> y

==> Building and installing package
==> Making package: razer-drivers 1.1.12-1 (Fri May  5 19:36:12 CEST 2017)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Downloading v1.1.12.tar.gz...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   132    0   132    0     0    218      0 --:--:-- --:--:-- --:--:--   218
100  155k  100  155k    0     0  97764      0  0:00:01  0:00:01 --:--:--  267k
==> Validating source files with sha256sums...
    v1.1.12.tar.gz ... Passed
==> Extracting sources...
  -> Extracting v1.1.12.tar.gz with bsdtar
==> Entering fakeroot environment...
==> Starting package_python-razer()...

:: Installing Razer python library
=====================================================
python3 setup.py install --prefix=/usr --root=/tmp/yaourt-tmp-rs/aur-openrazer-meta/pkg/python-razer
running install
running build
running build_py
creating build
creating build/lib
creating build/lib/razer
copying razer/__init__.py -> build/lib/razer
creating build/lib/razer/client
copying razer/client/macro.py -> build/lib/razer/client
[... 296 lines cut ...]
  -> Compressing package...
==> Leaving fakeroot environment.
==> Finished making: razer-drivers 1.1.12-1 (Fri May  5 19:36:19 CEST 2017)
==> Cleaning up...

==> Continue installing razer-drivers ? [Y/n]
==> [v]iew package contents [c]heck package with namcap
==> ---------------------------------------------------
==> y

loading packages...
resolving dependencies...
warning: cannot resolve "python-notify2", a dependency of "razer-daemon"
warning: cannot resolve "razer-daemon", a dependency of "openrazer-meta"
warning: cannot resolve "python-notify2", a dependency of "razer-daemon"
warning: cannot resolve "razer-daemon", a dependency of "python-razer"
warning: cannot resolve "python-razer", a dependency of "openrazer-meta"
warning: cannot resolve "python-notify2", a dependency of "razer-daemon"
warning: cannot resolve "razer-daemon", a dependency of "python-razer"
warning: cannot resolve "python-notify2", a dependency of "razer-daemon"
:: The following packages cannot be upgraded due to unresolvable dependencies:
      openrazer-meta  python-razer  razer-daemon

:: Do you want to skip the above packages for this upgrade? [y/N] 
`

z3ntu, 2017-05-05:
Yes that's because yaourt is too stupid to handle some dependencies apparently. Try some other AUR helper like pacaur 🙂

rudolfschmidt, 2017-05-05:
Thanks for the hint, pacaur is much better. The installation worked and everything was installed. Never had problems with yaourt before, but now I have a reason to not use it anymore.
I have installed openrazer-driver package and also polychromatic, both via pacaur.
Starting polychromatic-controller I get following: https://clbin.com/H1XSh0.png

[21 more replies]

closed 2017-05-05 by joonatoona