openrazer/openrazer #245: Question: Abyssus V2

borgdylan, 2017-04-02:
I am looking to buy the Blackwidow Chroma V2 from razer and they are giving away a free Abyssus V2 with it. I do know that the keyboard is supported but what about the Abyssus V2? If no what should I do once I get the mouse so as to provide the necessary technical data for it to get supported as well?

closed 2017-04-02 by borgdylan

z3ntu, 2017-04-02:
That's my template of what we need basically for new devices:

Please post the output of
`lsusb -d 1532:VID -v`
and
`ls -lah /dev/input/by-id/`
and read [this on how to get packet captures](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol). Also screenshots of all available options on Razer Synapse are needed.

borgdylan, 2017-04-22:
I ended up buying the deathadder elite instead as buying directly from razer turned out to be too complicated shipping wise. That means I did not get an Abyssus.

z3ntu, 2017-04-22:
Yep, that's already supported. :)

borgdylan, 2017-04-23:
Yes, it's working fine for me.