openrazer/openrazer #246: Korora (Fedora); "Daemon Not Installed"

cyanit, 2017-04-03:
So I set up my new desktop OS, Korora, and I read that I can install razer drivers, I read how to do it, I installed Polychromatic, noticed that I didn't had the daemon and before pressing Y I installed the daemon,  and it just isn't working. I Retried it a lot, i Restarted the deamon a lot and i restarted my whole system 5 times.
[image]

z3ntu, 2017-04-03:
Please post the output of `sudo dkms status`

cyanit, 2017-04-03:
`[16:23 leo ~]$ sudo dkms status
razer_chroma_driver, 1.0.0, 4.10.6-200.fc25.x86_64, x86_64: installed
`

z3ntu, 2017-04-03:
The output of `ls -al /sys/bus/hid/drivers/razer*/0*/` and `razer-service -v -F` please.

cyanit, 2017-04-03:
`[16:23 leo ~]$ ls -al /sys/bus/hid/drivers/razer*/0*/
'/sys/bus/hid/drivers/razerkbd/0003:1532:021E.0002/':
insgesamt 0
drwxr-xr-x. 5 root plugdev    0  3. Apr 08:17 .
drwxr-xr-x. 5 root root       0  3. Apr 08:17 ..
-r--r--r--. 1 root plugdev 4096  3. Apr 08:17 country
lrwxrwxrwx. 1 root plugdev    0  3. Apr 08:17 driver -> ../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x. 3 root plugdev    0  3. Apr 08:17 hidraw
drwxr-xr-x. 3 root plugdev    0  3. Apr 08:17 input
-rw-rw----. 1 root plugdev 4096  3. Apr 08:17 key_alt_f4
-rw-rw----. 1 root plugdev 4096  3. Apr 08:17 key_alt_tab
-rw-rw----. 1 root plugdev 4096  3. Apr 08:17 key_super
-r--r--r--. 1 root plugdev 4096  3. Apr 08:17 modalias
drwxr-xr-x. 2 root plugdev    0  3. Apr 08:17 power
-r--r--r--. 1 root plugdev 4096  3. Apr 08:17 report_descriptor
lrwxrwxrwx. 1 root plugdev    0  3. Apr 08:17 subsystem -> ../../../../../../../bus/hid
-rw-r--r--. 1 root plugdev 4096  3. Apr 08:17 uevent

'/sys/bus/hid/drivers/razerkbd/0003:1532:021E.0003/':
insgesamt 0
drwxr-xr-x. 5 root plugdev    0  3. Apr 08:17 .
drwxr-xr-x. 5 root root       0  3. Apr 08:17 ..
-r--r--r--. 1 root plugdev 4096  3. Apr 08:17 country
lrwxrwxrwx. 1 root plugdev    0  3. Apr 08:17 driver -> ../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x. 3 root plugdev    0  3. Apr 08:17 hidraw
drwxr-xr-x. 3 root plugdev    0  3. Apr 08:17 input
-rw-rw----. 1 root plugdev 4096  3. Apr 08:17 key_alt_f4
-rw-rw----. 1 root plugdev 4096  3. Apr 08:17 key_alt_tab
-rw-rw----. 1 root plugdev 4096  3. Apr 08:17 key_super
-r--r--r--. 1 root plugdev 4096  3. Apr 08:17 modalias
drwxr-xr-x. 2 root plugdev    0  3. Apr 08:17 power
-r--r--r--. 1 root plugdev 4096  3. Apr 08:17 report_descriptor
lrwxrwxrwx. 1 root plugdev    0  3. Apr 08:17 subsystem -> ../../../../../../../bus/hid
-rw-r--r--. 1 root plugdev 4096  3. Apr 08:17 uevent

'/sys/bus/hid/drivers/razerkbd/0003:1532:021E.0004/':
insgesamt 0
drwxr-xr-x. 5 root plugdev    0  3. Apr 08:17 .
drwxr-xr-x. 5 root root       0  3. Apr 08:17 ..
-r--r--r--. 1 root plugdev 4096  3. Apr 08:17 country
-rw-rw----. 1 root plugdev 4096  3. Apr 08:17 device_mode
-r--r-----. 1 root plugdev 4096  3. Apr 08:17 device_serial
-r--r-----. 1 root plugdev 4096  3. Apr 08:17 device_type
lrwxrwxrwx. 1 root plugdev    0  3. Apr 08:17 driver -> ../../../../../../../bus/hid/drivers/razerkbd
-r--r-----. 1 root plugdev 4096  3. Apr 08:17 firmware_version
-rw-rw----. 1 root plugdev 4096  3. Apr 08:17 game_led_state
drwxr-xr-x. 3 root plugdev    0  3. Apr 08:17 hidraw
drwxr-xr-x. 3 root plugdev    0  3. Apr 08:17 input
-rw-rw----. 1 root plugdev 4096  3. Apr 08:17 macro_led_effect
-rw-rw----. 1 root plugdev 4096  3. Apr 08:17 macro_led_state
-rw-rw----. 1 root plugdev 4096  3. Apr 08:17 matrix_brightness
--w--w----. 1 root plugdev 4096  3. Apr 08:17 matrix_custom_frame
--w--w----. 1 root plugdev 4096  3. Apr 08:17 matrix_effect_breath
--w--w----. 1 root plugdev 4096  3. Apr 08:17 matrix_effect_custom
--w--w----. 1 root plugdev 4096  3. Apr 08:17 matrix_effect_none
--w--w----. 1 root plugdev 4096  3. Apr 08:17 matrix_effect_reactive
--w--w----. 1 root plugdev 4096  3. Apr 08:17 matrix_effect_spectrum
--w--w----. 1 root plugdev 4096  3. Apr 08:17 matrix_effect_starlight
--w--w----. 1 root plugdev 4096  3. Apr 08:17 matrix_effect_static
--w--w----. 1 root plugdev 4096  3. Apr 08:17 matrix_effect_wave
-r--r--r--. 1 root plugdev 4096  3. Apr 08:17 modalias
drwxr-xr-x. 2 root plugdev    0  3. Apr 08:17 power
-r--r--r--. 1 root plugdev 4096  3. Apr 08:17 report_descriptor
lrwxrwxrwx. 1 root plugdev    0  3. Apr 08:17 subsystem -> ../../../../../../../bus/hid
-rw-rw----. 1 root plugdev 4096  3. Apr 08:17 test
-rw-r--r--. 1 root plugdev 4096  3. Apr 08:17 uevent
-r--r-----. 1 root plugdev 4096  3. Apr 08:17 version
[16:27 leo ~]$ razer-service -v -F
2017-04-03 16:27:38 | razer                          | INFO     | Initialising Daemon (v1.1.9). Pid: 16394
2017-04-03 16:27:38 | razer                          | CRITICAL | User is not a member of the plugdev group
`
here you go...

z3ntu, 2017-04-03:
Please read the messages 😉 
> 2017-04-03 16:27:38 | razer | CRITICAL | User is not a member of the plugdev group

You should run `sudo gpasswd -a <yourusername> plugdev`.

cyanit, 2017-04-03:
okay... thought that just matters for debugging or developer stuff....

z3ntu, 2017-04-03:
`CRITICAL` :)

closed 2017-04-03 by z3ntu

cyanit, 2017-04-03:
@z3ntu 
still not working.
[image]

`[16:36 leo ~]$ sudo gpasswd -a leo plugdev
[sudo] Passwort für leo: 
Benutzer leo wird zur Gruppe plugdev hinzugefügt.
[16:38 leo ~]$ razer-service -F
2017-04-03 16:40:45 | razer                          | INFO     | Initialising Daemon (v1.1.9). Pid: 17427
2017-04-03 16:40:45 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2012 --------------------- (1532:010D)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderChroma ------------------- (1532:0043)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderElite -------------------- (1532:005C)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerFireFly ---------------------------- (1532:0C00)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerImperiator ------------------------- (1532:002F)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerKrakenV2 --------------------------- (1532:0510)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaTE ---------------------- (1532:0046)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerNagaHex ---------------------------- (1532:0041)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexV2 -------------------------- (1532:0050)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaver -------------------------- (1532:0113)
2017-04-03 16:40:45 | razer                          | DEBUG    | Loaded device specification: RazerOrnata ----------------------------- (1532:021F)
[... cut ...]
PermissionError: [Errno 13] Permission denied: '/sys/devices/pci0000:00/0000:00:14.0/usb3/3-13/3-13:1.2/0003:1532:021E.0004/device_serial'
Exception ignored in: <object repr() failed>
Traceback (most recent call last):
  File "/usr/lib/python3.5/site-packages/razer_daemon/hardware/device_base.py", line 412, in __del__
    self.close()
  File "/usr/lib/python3.5/site-packages/razer_daemon/hardware/device_base.py", line 325, in close
    if not self._is_closed:
AttributeError: 'RazerOrnataChroma' object has no attribute '_is_closed'
`

z3ntu, 2017-04-03:
Did you re-login?

z3ntu, 2017-04-03:
If yes, reboot and try again.

cyanit, 2017-04-03:
okay, ill try...

cyanit, 2017-04-03:
nope. Still not working... 
(ich hab auch discord, also es wäre möglich das mal zu besprechen, ka ob das jemand jemals schonmal auf Korora/Fedora getestet hat...)

z3ntu, 2017-04-03:
If you have telegram it would be even better but I can also join you on discord in about an hour.

z3ntu, 2017-04-03:
@cyanit What's your discord tag?

cyanit, 2017-04-03:
https://discord.gg/ppthzpZ
This link can only be used **once**. It leads to my Discord server, Polygonnetzwerk. 
I'll be available around 18.30 (Berlin Time). Till then, feel free to have a bit of fun with our bot(chat) ;)

z3ntu, 2017-04-03:
1. I joined
2. It's already 19:15 in Europe...

z3ntu, 2017-04-03:
Everything alright now here :)

Martinusbe, 2017-04-05:
@cyanit Crazylex just let slip that pa crew get paid by oneplus, and then quicly deleted it

z3ntu, 2017-04-05:
@Martinusbe ?

Martinusbe, 2017-04-05:
@cyanit yea, deleted 10 seconds after. but i was just looking