openrazer/openrazer #248: daemon not responding CENTOS 7

editRT, 2017-04-08:
I'm trying to install the driver on Centos 7, 
my device is Tartarus Chroma, 

I got this error:
Importing device logic, waiting for daemon
ERROR: the daemon is not responding!
Try running `killall razer-service && razer-service` or rebooting. If this doesn't work, please fill an issue!

I'm new to linux, 
I've found no solution yet, 
can someone please help, 
thanks, 
Francesco

z3ntu, 2017-04-10:
This project needs Python 3 which is not available on CentOS 7. If you don't really need CentOS for anything I would suggest you using basically any other distro than CentOS 😉

terrycain, 2017-04-10:
Worst case you can use software collections to get python3 but they are major pita, to the point where I wont "support" the use of them (aka you'll be on your own). I know full well they'll run daemon fine, its just the hassle of running stuff in scl.

z3ntu, 2017-04-16:
Sorry, but the daemon doesn't support Python 2 and needs Python 3.

closed 2017-04-16 by z3ntu