openrazer/openrazer #249: No device found (Gentoo Linux - linux 4.10.8)

kripton, 2017-04-09:
Hi, I'm trying to get polychromatic to work under Gentoo Linux but it fails to recognize my Razer Diamondback Chroma. The module "razermouse" is automatically loaded but doesn't seem to bind correctly:

> kripton@ayanta ~ $ dmesg | grep -iC 2 razer
> [    3.673849] usb 1-13: New USB device found, idVendor=1532, idProduct=004c
> [    3.673852] usb 1-13: New USB device strings: Mfr=1, Product=2, SerialNumber=0
> [    3.673855] usb 1-13: Product: Razer Diamondback
> [    3.673857] usb 1-13: Manufacturer: Razer
> [    3.675248] input: Razer Razer Diamondback as /devices/pci0000:00/0000:00:14.0/usb1/1-13> /1-13:1.0/0003:1532:004C.0006/input/input19
> [    3.675483] hid-generic 0003:1532:004C.0006: input,hidraw5: USB HID v1.11 Mouse [Razer Razer Diamondback] on usb-0000:00:14.0-13/input0
> [    3.677268] input: Razer Razer Diamondback as /devices/pci0000:00/0000:00:14.0/usb1/1-13/1-13:1.1/0003:1532:004C.0007/input/input20
> [    3.728849] hid-generic 0003:1532:004C.0007: input,hidraw6: USB HID v1.11 Keyboard [Razer Razer Diamondback] on usb-0000:00:14.0-13/input1
> [    3.729612] input: Razer Razer Diamondback as /devices/pci0000:00/0000:00:14.0/usb1/1-13/1-13:1.2/0003:1532:004C.0008/input/input21
> [    3.781691] hid-generic 0003:1532:004C.0008: input,hidraw7: USB HID v1.11 Keyboard [Razer Razer Diamondback] on usb-0000:00:14.0-13/input2
> [    3.937610] usb 1-14: new low-speed USB device number 7 using xhci_hcd
> [    4.110988] usb 1-14: New USB device found, idVendor=046d, idProduct=c31c
> [    6.163245] EXT4-fs (dm-0): couldn't mount as ext2 due to feature incompatibilities
> [    6.165913] EXT4-fs (dm-0): mounted filesystem with ordered data mode. Opts: (null)
> [    8.125609] razermouse: loading out-of-tree module taints kernel.
> [    8.125610] razermouse: loading out-of-tree module taints kernel.
> [    8.346194] md/raid1:md126: active with 2 out of 2 mirrors
> [    8.393091] md126: detected capacity change from 0 to 499962503168
> kripton@ayanta ~ $ lsmod | grep razer
> razermouse             42947  0

> kripton@ayanta ~ $ find /sys/bus/hid/drivers/razer*
> /sys/bus/hid/drivers/razermouse
> /sys/bus/hid/drivers/razermouse/bind
> /sys/bus/hid/drivers/razermouse/unbind
> /sys/bus/hid/drivers/razermouse/module
> /sys/bus/hid/drivers/razermouse/uevent
> /sys/bus/hid/drivers/razermouse/new_id

Any idea what might be wrong or hint to get more debugging information? The modules don't seem to have parameters to make them more chatty ....

kripton, 2017-04-10:
Any positive reports on razer drivers working with linux 4.10.*? That way I could rule out the kernel version as source of the problem ...

z3ntu, 2017-04-10:
Yes, am running 4.10.8

terrycain, 2017-04-11:
Probably udev not playing ball, have you tried binding it manually?

kripton, 2017-04-11:
Yep, binding manually helps. The time I posted the issue, I was not aware of the razer_mount script and what it's doing. Now the daemon and polychromatic are working nicely. Thanks for your work!

closed 2017-04-11 by kripton

z3ntu, 2017-04-12:
Uhm does it also still work after rebooting?

wschlich, 2017-08-16:
The issue is due to the razer-drivers-9999.ebuild installing all udev stuff into /usr/lib/udev instead of /lib/udev.
After manually moving the files and re-plugging the device, everything works fine.

z3ntu, 2017-08-16:
CC @vifino

z3ntu, 2017-08-16:
@wschlich And thanks for finding that out.

vifino, 2017-08-16:
@z3ntu Fixed in https://github.com/vifino/vifino-overlay/commit/c6ae5b75f7516569a18acbab0f9703465d96743f

wschlich, 2017-08-17:
@vifino Thanks! :)