openrazer/openrazer #250: razer-daemon depends on python3-evdev but it is not available on debian testing "stretch"

cphuntington97, 2017-04-10:


eplgr, 2017-04-10:
I'm trying to make this work as well, same problem with `sid` (unstable version of debian).

eplgr, 2017-04-10:
Alright, I got it to work, @cphuntington97 . I am not sure this was the right approach, but it *does* work.
Download & install `python3-evdev` from:
http://packages.ubuntu.com/yakkety/python3-evdev
Then proceed with the rest, install `razer-daemon` and you are all set. I also was able to install Polychromatic and tweak all features successfully.

terrycain, 2017-04-10:
@eplgr meh, if it works it works :wink:, yeah in the ubuntu PPA, i've added the newer version of python-evdev. @z3ntu you think we can get newer evdev into the obs repo?

terrycain, 2017-04-11:
@eplgr Have spoken to the debian maintainer, we're gonna try and get our copy of the near-latest python-evdev into that repo like it is in my PPA

eplgr, 2017-04-11:
@terrycain That's really good, thank you.

cphuntington97, 2017-04-14:
Thanks everyone

cathedrall, 2017-05-03:
I know this is old, but I also got it work by using pip. You can install pip for python3 from the repo's, then use pip3 install evdev to get it installed.

z3ntu, 2017-05-04:
Can somebody test the instructions here: https://software.opensuse.org//download.html?project=hardware%3Arazer&package=openrazer-meta (click the debian tab and follow the instructions).

If everything works fine, then I can release the package officially.

z3ntu, 2017-05-04:
[Debian is now supported!](https://terrycain.github.io/razer-drivers/#debian) Please report any problems to us!

closed 2017-05-04 by z3ntu