openrazer/openrazer #251: Overwatch keyboard support -- 1532:0211

KayEss, 2017-04-13:
I guess you need the capture details for this to work on it, but I don't currently have access to a Windows machine.

lezed1, 2017-04-13:
The packet capturing for a device is done in Windows VM running on Linux. Windows is not run natively when capturing.

z3ntu, 2017-04-13:
Please post the output of
`lsusb -d 1532:0211 -v`
and
`ls -lah /dev/input/by-id/`
and read [this on how to get packet captures](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol
). Also screenshots of all available options on Razer Synapse are needed.
And last, please send the product and model number for the records.

KayEss, 2017-04-14:
$ sudo lsusb -d 1532:0211 -v

    Bus 002 Device 003: ID 1532:0211 Razer USA, Ltd 
    Device Descriptor:
    bLength                18
    bDescriptorType         1
    bcdUSB               2.00
    bDeviceClass            0 (Defined at Interface level)
    bDeviceSubClass         0 
    bDeviceProtocol         0 
    bMaxPacketSize0        64
    idVendor           0x1532 Razer USA, Ltd
    idProduct          0x0211 
    bcdDevice            2.00
    iManufacturer           1 Razer
    iProduct                2 Razer BlackWidow Chroma
    iSerial                 0 
    bNumConfigurations      1
    Configuration Descriptor:
        bLength                 9
        bDescriptorType         2
        wTotalLength           84
        bNumInterfaces          3
        bConfigurationValue     1
        iConfiguration          0 
        bmAttributes         0xa0
        (Bus Powered)
        Remote Wakeup
        MaxPower              500mA
        Interface Descriptor:
        bLength                 9
        bDescriptorType         4
        bInterfaceNumber        0
        bAlternateSetting       0
        bNumEndpoints           1
        bInterfaceClass         3 Human Interface Device
        bInterfaceSubClass      1 Boot Interface Subclass
        bInterfaceProtocol      1 Keyboard
        iInterface              0 
            HID Device Descriptor:
            bLength                 9
            bDescriptorType        33
            bcdHID               1.11
            bCountryCode            0 Not supported
            bNumDescriptors         1
            bDescriptorType        34 Report
            wDescriptorLength      61
            Report Descriptors: 
            ** UNAVAILABLE **
        Endpoint Descriptor:
            bLength                 7
            bDescriptorType         5
            bEndpointAddress     0x81  EP 1 IN
            bmAttributes            3
            Transfer Type            Interrupt
            Synch Type               None
            Usage Type               Data
            wMaxPacketSize     0x0008  1x 8 bytes
            bInterval               1
        Interface Descriptor:
        bLength                 9
        bDescriptorType         4
        bInterfaceNumber        1
        bAlternateSetting       0
        bNumEndpoints           1
        bInterfaceClass         3 Human Interface Device
        bInterfaceSubClass      0 No Subclass
        bInterfaceProtocol      1 Keyboard
        iInterface              0 
            HID Device Descriptor:
            bLength                 9
            bDescriptorType        33
            bcdHID               1.11
            bCountryCode            0 Not supported
            bNumDescriptors         1
            bDescriptorType        34 Report
            wDescriptorLength     159
            Report Descriptors: 
            ** UNAVAILABLE **
        Endpoint Descriptor:
            bLength                 7
            bDescriptorType         5
            bEndpointAddress     0x82  EP 2 IN
            bmAttributes            3
            Transfer Type            Interrupt
            Synch Type               None
            Usage Type               Data
            wMaxPacketSize     0x0010  1x 16 bytes
            bInterval               1
        Interface Descriptor:
        bLength                 9
        bDescriptorType         4
        bInterfaceNumber        2
        bAlternateSetting       0
        bNumEndpoints           1
        bInterfaceClass         3 Human Interface Device
        bInterfaceSubClass      0 No Subclass
        bInterfaceProtocol      2 Mouse
        iInterface              0 
            HID Device Descriptor:
            bLength                 9
            bDescriptorType        33
            bcdHID               1.11
            bCountryCode            0 Not supported
            bNumDescriptors         1
            bDescriptorType        34 Report
            wDescriptorLength      94
            Report Descriptors: 
            ** UNAVAILABLE **
        Endpoint Descriptor:
            bLength                 7
            bDescriptorType         5
            bEndpointAddress     0x83  EP 3 IN
            bmAttributes            3
            Transfer Type            Interrupt
            Synch Type               None
            Usage Type               Data
            wMaxPacketSize     0x0008  1x 8 bytes
            bInterval               1
    Device Status:     0x0000
    (Bus Powered)

    $ ls -lah /dev/input/by-id/
    total 0
    drwxr-xr-x 2 root root 160 Apr  6 17:07 .
    drwxr-xr-x 4 root root 460 Apr  6 17:07 ..
    lrwxrwxrwx 1 root root   9 Apr  6 17:07 usb-Razer_Razer_Abyssus-event-mouse -> ../event5
    lrwxrwxrwx 1 root root   9 Apr  6 17:07 usb-Razer_Razer_Abyssus-mouse -> ../mouse1
    lrwxrwxrwx 1 root root   9 Apr  6 17:07 usb-Razer_Razer_BlackWidow_Chroma-event-kbd -> ../event2
    lrwxrwxrwx 1 root root   9 Apr  6 17:07 usb-Razer_Razer_BlackWidow_Chroma-if01-event-kbd -> ../event3
    lrwxrwxrwx 1 root root   9 Apr  6 17:07 usb-Razer_Razer_BlackWidow_Chroma-if02-event-mouse -> ../event4
    lrwxrwxrwx 1 root root   9 Apr  6 17:07 usb-Razer_Razer_BlackWidow_Chroma-if02-mouse -> ../mouse0

z3ntu, 2017-04-15:
Just to clarify, the keyboard is that one, right? https://www.razerzone.com/licensed-and-team-peripherals/overwatch-razer-blackwidow-chroma/

KayEss, 2017-04-16:
Yes indeed. Is there a way to see the protocols followed by similar keyboards and try them out? I suppose I could write some Python code to try to send codes to the keyboard. I think this would be easier for me than buying a copy of Windows to run.

terrycain, 2017-04-16:
I have a theory that it's just a ultimate 2016 with a different colour. Can you grab telegram and add me @terrycain, would be easier to test it then

terrycain, 2017-04-16:
Should be able to see if it responds to a modified 2016 drivwe

terrycain, 2017-04-16:
For reference its a BlackWidow Chroma V1 clone

terrycain, 2017-04-17:
New release is out, is in launchpad, will be out soon. Non Ubuntu packages will be out soon after.

closed 2017-04-17 by terrycain