openrazer/openrazer #252: set keyboard brightness to 0% when display sleeps; restore brightness to previous level when display wakes

cphuntington97, 2017-04-14:
This functionality exists on Windows and I would like it to exist on Linux. If it is not currently possible, I would like to request it as a feature.

lah7, 2017-04-14:
This kind of already exists, but is based on the screensaver state (if it supports the Freedesktop spec, like GNOME, MATE and I think LXDE does too). This could just be the lock screen on some desktop environments, depends which one you're using.

So for me, under MATE - if I leave my computer idle long enough so the screensaver activates (and screen goes off), they'll go "off" until the system is unlocked again.

terrycain, 2017-04-14:
It works for all the standard desktop environments

z3ntu, 2017-04-14:
I can add that it also works with KDE perfectly.

cphuntington97, 2017-04-15:
Any advice for getting this to work in Enlightenment 21.5?

terrycain, 2017-04-15:
find out what does your screensaver would be a start

cphuntington97, 2017-04-20:
Can the brightness be controlled from the command line?

simotek, 2017-04-20:
Enlightenment doesn't use a screensaver, it controls the backlight itself. Do you have a link to the freedesktop spec you are using? we can atleast create a feature request bug for enlightenment to implement the right standard when it puts the display to sleep / wakes up.

terrycain, 2017-04-20:
Does it emit any dbus events when it changes backlight?

simotek, 2017-04-20:
Not currently, it just writes directly to the back light device. If all the common screensavers and some of the other DE's are using a standard to notify other applications / devices of this then it would be reasonable to get enlightenment to also do this.

terrycain, 2017-04-20:
@simotek org.freedesktop.ScreenSaver
Is the main one that has a getActive method i think

z3ntu, 2017-05-05:
Closing as it's not our fault. Feel free to file a bug report for Enlightenment or provide us another D-Bus path Enlightenment is using.

closed 2017-05-05 by z3ntu