openrazer/openrazer #254: Razer Ornata not being recognised on Arch!

fadedrobin, 2017-04-16:


z3ntu, 2017-04-16:
Please run the following commands and send the output here.
```
sudo dkms install razer_chroma_driver/1.0.0
groups
```

And please for the next issues you are reporting somewhere include at least some information without getting asked specifically for it 😉

fadedrobin, 2017-04-16:
This just appeared when I typed in the first one:
```
Your kernel headers for kernel 4.10.8-1-ARCH cannot be found at 
Error! echo
Your kernel headers for kernel 4.10.8-1-ARCH cannot be found at
/usr/lib/modules/4.10.8-1-ARCH/build or /usr/lib/modules/4.10.8-1-ARCH/source.

```

z3ntu, 2017-04-16:
Yep you have to install the kernel headers for dkms. Probably `linux-headers` for you.

closed 2017-04-16 by z3ntu

fadedrobin, 2017-04-16:
Thanks! This is my first time using Arch!

z3ntu, 2017-04-16:
Yes, dkms is a bit confusing as it has no hard dependencies on the kernel headers because you can install multiple kernels which need their own header package. The first line in https://wiki.archlinux.org/index.php/Dynamic_Kernel_Module_Support#Installation tells you *a bit* about it.

fadedrobin, 2017-04-16:
I'm still getting the same error after installing it and restarting my computer! Any idea why?

z3ntu, 2017-04-16:
`uname -a`
`pacman -Q linux linux-headers`

z3ntu, 2017-04-16:
So it's working now?

fadedrobin, 2017-04-16:
Yeah perfectly I had 4.8 for Linux and 4.10 for headers! lol

z3ntu, 2017-04-16:
That's kinda what is meant by https://wiki.archlinux.org/index.php/System_maintenance#Partial_upgrades_are_unsupported 😄

z3ntu, 2017-04-16:
But great to hear that it's working now!

fadedrobin, 2017-04-16:
Yeah thanks A TON!