openrazer/openrazer #256: Razer Naga 2014 lighting not working

kdb424, 2017-04-19:
https://github.com/lah7/polychromatic/issues/77
Continuing the issue raised here upstream to the driver. Lighting unable to change at all even with manual driver access. If you need any other details not listed in that issue, please let me know. I'll also test my mouse in a Windows VM to see if it is possibly my mouse itself.

z3ntu, 2017-04-19:
Anything in dmesg that comes from `razermouse`? Otherwise probably Synapse packet captures are needed (except it's a really stupid error - @terrycain ? 😉 )

kdb424, 2017-04-19:
dmesg | grep razermouse                                                                                                                                1 ↵
[99721.958304] razermouse 0003:1532:0040.001E: input,hidraw1: USB HID v1.11 Mouse [Razer Razer Naga 2014] on usb-0000:00:14.0-1/input0
[99722.014912] razermouse 0003:1532:0040.0020: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer Naga 2014] on usb-0000:00:14.0-1/input2
[99722.070880] razermouse 0003:1532:0040.001F: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer Naga 2014] on usb-0000:00:14.0-1/input1
 That's what I have from the mouse there. I'll look into packet captures as needed.

z3ntu, 2017-06-11:
Any update on the issue?

kdb424, 2017-06-11:
I haven't gotten to get the lights on even in a WIndows machine. It worked before I started playing with this driver, and no longer does. Don't know if it was bad timing, or corrupted something somehow, though I doubt that's easily possible. I guess my mouse won't be of any help unfortunately.

z3ntu, 2017-06-11:
I'll close this then as right now nothing is confirmed 😉

closed 2017-06-11 by z3ntu