openrazer/openrazer #257: Modifier keys ignored for M1-M5 keys on Blackwidow Ultimate 2014

felicitus, 2017-04-22:
Hi,

I got the blackwidow ultimate 2014 running, the razer driver works fine. However, there's one issue: It seems that modifier keys in conjunction with M1-M5 keys aren't working.

Output of xev with Mod4 (Windows Key) pressed:

```
KeyPress event, serial 33, synthetic NO, window 0x3800001,
    root 0xe4, subw 0x0, time 2351514, (863,55), root:(2787,1023),
    state 0x0, keycode 191 (keysym 0x1008ff81, XF86Tools), same_screen YES,
    XKeysymToKeycode returns keycode: 179
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False
```

Usually the state is showing "state 0x40" when using the Mod4 (Windows Key). Also, the M1-M5 keys aren't working with any modifier like Shift, Alt etc.

terrycain, 2017-04-22:
@felicitus Am assuming m1-5 emit keycodes though? Appearing as something like F13-F17

felicitus, 2017-04-22:
@terrycain yes, that's correct - they appear in xev as XF86Tools, XF86Launch5, XF86Launch6, XF86Launch7 and XF86Launch8 for M1-M5.

terrycain, 2017-04-22:
@felicitus Is the daemon running normally? As nowadays, the daemon will put the device into "driver mode" which makes it real dumb (aka no Fn+Keys work) but will emit keycodes for M1-5 and Fn (though driver hides Fn keypresses), and the daemon will handle in software the Fn+Key functionality (excluding the media keys as the driver converts those keycodes on the fly to the corect values, if the daemon is working).

Theres a log file in ~/.razer-service/logs/
If its filled with rubbish, remove it, reboot and then see if it runs, then put output on pastebin and link here

felicitus, 2017-04-22:
@terrycain I assume that the daemon is running normally. I tried digging around in the code, but I'm not very familiar with on how X11 translates a mod key plus another keypress to a composite keycode.

I'm not sure if the razer daemon should handle this, I will see if I can map M1-M5 to X symbols myself.

Anyways, attached is the log file of the razer daemon.
[razer.log.txt](https://github.com/terrycain/razer-drivers/files/949090/razer.log.txt)

terrycain, 2017-04-22:
@felicitus 2017-04-22 17:30:28 | razer                          | INFO     | Stopping daemon on signal 15

That doesnt look good.

As far as mapping to M1-5, the macro code is in the works, its not finished yet, and initially it wont care about modifier keys.

felicitus, 2017-04-22:
@terrycain the signal 15 was me killing it, so that's OK.

Well actually I'm not really interested in macros, having M1-M5 as a regular key would be perfectly fine.

However, there are a few oddities with the Linux keyboard handling I'm curious about.

First of all, I installed the udev rules and they seem to work, at least M1-M5 appears within xev as soon as the udev rules are in place and the daemon is running.

However, when I try to list the usb devices with `usb-devices`, it shows that the driver it's using is `usbhid` and not `razerkbd` as I would expect:

```
T:  Bus=02 Lev=02 Prnt=02 Port=05 Cnt=02 Dev#=  8 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
P:  Vendor=1532 ProdID=011a Rev=02.00
S:  Manufacturer=Razer
S:  Product=Razer BlackWidow Ultimate
C:  #Ifs= 3 Cfg#= 1 Atr=a0 MxPwr=100mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=01 Driver=usbhid
I:  If#= 1 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=01 Driver=usbhid
I:  If#= 2 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=02 Driver=usbhid
```

Also, I expected earlier that I could simply list the M1-M5 keycodes using `showkeys`, but they don't appear there unless I start the razer-service.

terrycain, 2017-04-22:
@felicitus Yeah theres a udev rule that will do and unbind and rebind to razerkbd (as hid-generic is evil) and then razer-service will upon recognising the keyboard put it into driver mode in which it will then emit keycodes

felicitus, 2017-04-22:
Well I found a driver which actually correctly enables the M1-M5 keys here: https://github.com/astrolince/razer-blackwidow-ultimate

Might it be helpful to others.

terrycain, 2017-04-22:
@felicitus yeah that script is essentially a tiny part of what happens in the daemon

closed 2017-06-15 by terrycain