openrazer/openrazer #260: Slow mouse speed after upgrading to Ubuntu 17.04

claell, 2017-04-24:
After I upgraded to Ubuntu 17.04 I experienced a pretty slow mouse speed. I also cannot change it at all in the system settings. I am unsure what causes this, but thought I'd post it here.

KayEss, 2017-04-24:
I had the exact same thing. I finally managed to get it fixed using xinput:

    xinput set-prop 11 154 2.400000, 0.000000, 0.000000, 0.000000, 2.400000, 0.000000, 0.000000, 0.000000, 1.000000

You need to use `xinput list` to find the device ID (11 above) and then `xinput list-props 11` to find the ID for the `Coordinate Transformation Matrix` property.

claell, 2017-04-24:
Thanks for your answer. Do you know what causes this? Is this a problem of Ubuntu?

KayEss, 2017-04-25:
> Is this a problem of Ubuntu?

That's unclear, but likely. I saw it on KDE using a Razor mouse. What UI shell are you using? Is your mouse also a Razer one?

terrycain, 2017-04-25:
I'd like to say its nothing to do with the driver so its gonna be closed until proven otherwise.

closed 2017-04-25 by terrycain

claell, 2017-04-25:
@KayEss I am using Unity and a Razer Abyssus 2014.
@terrycain Alright, thanks for clarification.

claell, 2017-04-28:
This is a bug of Ubuntu, see https://bugs.launchpad.net/ubuntu/+source/unity-control-center/+bug/1683145 and https://bugs.launchpad.net/ubuntu/+source/xorg/+bug/1682193

drnaseeb, 2018-02-24:
My touchpad ID 14, device ID 267
"xinput set-prop 14 267 2.400000, 0.000000, 0.000000, 0.000000, 2.400000, 0.000000, 0.000000, 0.000000, 1.000000" doesn't work for me in Ubuntu 17.10 on Asus X200MA laptop.
It says,
"X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  131 (XInputExtension)
  Minor opcode of failed request:  57 ()
  Value in failed request:  0x10b
  Serial number of failed request:  19
  Current serial number in output stream:  20"

I also tried:
"xinput --set-prop 14 267 0.1"
It gives same response.

Also tried
"xset m acceleration threshold"
 It is effective in reducing pointer speed but not much in increasing that.

Any other suggestion

simonhf, 2018-04-17:
I tried the suggestion of @KayEss on Ubuntu 17.10 but it made no difference at all :-( I'm also looking for other suggestions... :-)

KayEss, 2018-05-05:
I just had to redo this so let me give a bit more detail. The device ID and the property ID will change, and can change over time as drivers etc. get updated.

First of all you need to find the device with `xinput list`. On my machine I'm getting this:

```
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ Razer Razer BlackWidow Chroma             id=9    [slave  pointer  (2)]
⎜   ↳ Razer Razer BlackWidow Chroma             id=10   [slave  pointer  (2)]
⎜   ↳ Razer Razer Abyssus                       id=11   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Power Button                              id=7    [slave  keyboard (3)]
    ↳ Razer Razer BlackWidow Chroma             id=8    [slave  keyboard (3)]
    ↳ Eee PC WMI hotkeys                        id=12   [slave  keyboard (3)]
    ↳ Razer Razer BlackWidow Chroma             id=13   [slave  keyboard (3)]
```
You can see the mouse at ID 11. Now I need to list the properties using `xinput list-props 11`. I get the following

````
Device 'Razer Razer Abyssus':
        Device Enabled (153):   1
        Coordinate Transformation Matrix (155): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        libinput Natural Scrolling Enabled (290):       0
        libinput Natural Scrolling Enabled Default (291):       0
        libinput Scroll Methods Available (294):        0, 0, 1
        libinput Scroll Method Enabled (295):   0, 0, 0
        libinput Scroll Method Enabled Default (296):   0, 0, 0
        libinput Button Scrolling Button (297): 2
        libinput Button Scrolling Button Default (298): 2
        libinput Middle Emulation Enabled (299):        0
        libinput Middle Emulation Enabled Default (300):        0
        libinput Accel Speed (301):     0.000000
        libinput Accel Speed Default (302):     0.000000
        libinput Accel Profiles Available (303):        1, 1
        libinput Accel Profile Enabled (304):   1, 0
        libinput Accel Profile Enabled Default (305):   1, 0
        libinput Left Handed Enabled (306):     0
        libinput Left Handed Enabled Default (307):     0
        libinput Send Events Modes Available (275):     1, 0
        libinput Send Events Mode Enabled (276):        0, 0
        libinput Send Events Mode Enabled Default (277):        0, 0
        Device Node (278):      "/dev/input/event11"
        Device Product ID (279):        5426, 66
        libinput Drag Lock Buttons (292):       <no items>
        libinput Horizontal Scroll Enabled (293):       1
````
The property number you're after is the one for "Coordinate Transformation Matrix". The name might be slightly different as well so you may have to try a few that have similar looking configuration. The configuration is nine numbers because it's a 3x3 matrix used for affine transformations in a 2D space. Now that I have that I can set acceleration by giving suitable scaling values on `x` and `y` (with `h` being 1)

    xinput set-prop 11 155 2.400000, 0.000000, 0.000000, 0.000000, 2.400000, 0.000000, 0.000000, 0.000000, 1.000000

Hope that helps