openrazer/openrazer #263: Razer blade stealth, Daemon does not start

chazmead, 2017-04-27:
Hey, just got the late 2016 Kabylake RBS, running debian 9 ( stretch ) with an updated 4.10.12 kernel.

Attempting to install these drivers and all seems to make / install correctly, but installs to site-packages not dist-packages, but even running `PYTHONPATH='/usr/lib/python3.5/site-packages' razer-service -F` gives the following error message:
> 
> Traceback (most recent call last):
>   File "/usr/bin/razer-service", line 102, in <module>
>     run()
>   File "/usr/bin/razer-service", line 96, in run
>     daemonize(**daemon_args)
>   File "/usr/lib/python3.5/site-packages/razer_daemon/daemon.py", line 116, in daemonize
>     daemon = RazerDaemon(verbose, log_dir, console_log, run_dir, config_file, test_dir=test_dir)
>   File "/usr/lib/python3.5/site-packages/razer_daemon/daemon.py", line 176, in __init__
>     DBusService.__init__(self, self.BUS_PATH, '/org/razer')
>   File "/usr/lib/python3.5/site-packages/razer_daemon/dbus_services/service.py", line 81, in __init__
>     bus_object = dbus.service.BusName(bus_name, bus=dbus.SessionBus())
>   File "/usr/lib/python3/dist-packages/dbus/_dbus.py", line 211, in __new__
>     mainloop=mainloop)
>   File "/usr/lib/python3/dist-packages/dbus/_dbus.py", line 100, in __new__
>     bus = BusConnection.__new__(subclass, bus_type, mainloop=mainloop)
>   File "/usr/lib/python3/dist-packages/dbus/bus.py", line 122, in __new__
>     bus = cls._new_for_bus(address_or_type, mainloop=mainloop)
> dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

Hope this helps, and if you can provide some help on how to resolve that would be excellent, As I have heard your drivers will fix the horrid Caps lock crash that is happening!!

Thanks in advance.

lezed1, 2017-04-27:
Are you a member of the `plugdev` group?

chazmead, 2017-04-27:
Does that apply to the root user?  i'll see if that resolves it.

chazmead, 2017-04-30:
Hey guys, I managed to get the razer-service running, however it doesn't seem to be recognising the 1532:0220 USB Device, please see below for log output:

> 2017-04-30 14:25:16 | razer | INFO | Initialising Daemon (v1.1.11). Pid: 5895
> 2017-04-30 14:25:16 | razer.screensaver | INFO | Initialising DBus Screensaver Monitor
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerBlackWidow2012 --------------------- (1532:010D)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerCore ------------------------------- (1532:0215)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerDeathadderChroma ------------------- (1532:0043)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerDeathadderElite -------------------- (1532:005C)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerFireFly ---------------------------- (1532:0C00)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerImperiator ------------------------- (1532:002F)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerKrakenV2 --------------------------- (1532:0510)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerMamba2012Wired --------------------- (1532:0024)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerMamba2012Wireless ------------------ (1532:0025)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerMambaChromaTE ---------------------- (1532:0046)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerNaga2014 --------------------------- (1532:0040)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerNagaHex ---------------------------- (1532:0041)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerNagaHexV2 -------------------------- (1532:0050)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerNostromo --------------------------- (1532:0111)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerOrbweaver -------------------------- (1532:0113)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerOrnata ----------------------------- (1532:021F)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerOrnataChroma ----------------------- (1532:021E)
> 2017-04-30 14:25:16 | razer | DEBUG | Loaded device specification: RazerOrochiWired ------------------------ (1532:0048)
[... cut ...]
> 2017-04-30 14:25:16 | razer | INFO | Adding razer.devices.syncEffects method to DBus
> 2017-04-30 14:25:16 | razer | INFO | Adding razer.daemon.version method to DBus
> 2017-04-30 14:25:16 | razer | INFO | Adding razer.daemon.stop method to DBus
> 2017-04-30 14:25:16 | razer | INFO | Serving DBus
> 2017-04-30 14:25:17 | razer | DEBUG | DBus called get_serial_list
> 2017-04-30 14:25:17 | razer | DEBUG | DBus called get_serial_list

Any ideas why the daemon isn't recognising the device?
Thanks.

terrycain, 2017-05-01:
```
ls -lah /sys/bus/hid/drivers/razerkbd/
```

gravelg, 2017-05-01:
I had the same problem as you and have fixed it by following this guide https://github.com/terrycain/razer-drivers/wiki/Setting-up-the-keyboard-driver and rebooting.

chazmead, 2017-05-01:
`ls -lah /sys/bus/hid/drivers/razerkbd/`:

> drwxr-xr-x 2 root root    0 May  1 16:50 .
> drwxr-xr-x 6 root root    0 May  1 16:49 ..
> --w------- 1 root root 4.0K May  1 16:49 bind
> lrwxrwxrwx 1 root root    0 May  1 16:49 module -> ../../../../module/razerkbd
> --w------- 1 root root 4.0K May  1 16:49 new_id
> --w------- 1 root root 4.0K May  1 16:49 uevent
> --w------- 1 root root 4.0K May  1 16:49 unbind

I have followed the setting up the keyboard driver on multiple occasions, and as soon as I reboot the driver gets bound back to the hid_generic driver.. is there something else I need to do to make this persist through reboot?  also the guide says we only need to do the mouse interface type, is this true, and is the interface number for the mouse interface of this keyboard also 002?

Thanks in advance.

gravelg, 2017-05-01:
I only had to do the mouse interface and it was numbered 002 for me. I didn't have to do anything for that setting to be persisted through reboots.

terrycain, 2017-05-01:
The turd that is `udev` should be doing all that for you. Does `/lib/udev/rules.d/99-razer.rules` exist?

chazmead, 2017-05-01:
@terrycain - No this file does not exist, what should be in it? thanks.

terrycain, 2017-05-01:
https://github.com/terrycain/razer-drivers/blob/master/install_files/udev/99-razer.rules

https://github.com/terrycain/razer-drivers/blob/master/install_files/udev/razer_mount should also exist as `/lib/udev/razer_mount`

chazmead, 2017-05-01:
smashed it!  seems to load correctly now.  Any ideas why those files didn't install in the make / make install process?  might be worth adding some info around these files to the setting up keyboard guide.

Thanks guys, much appreciated :+1:   Ready to close.

chazmead, 2017-05-01:
ah great.. I couldn't find that repo before, I tried the ubuntu repo, but kept getting GPG issues and it wouldn't install so had to go down the manual approach :)  thanks again.

terrycain, 2017-05-01:
Np, no gaurentee it works :wink: but soon the link will appear on the website.

closed 2017-05-01 by terrycain