openrazer/openrazer #264: Add support for Razer Blade Pro (2017)

jbagel2, 2017-04-28:
Have had the new blade pro for about a week. Working on getting a daemon working for it. so far no luck. Willing to provide any info necessary. 

Running a non standard Debian fork (Kali) .

Will update with new info or submit pull request if find work around.

hpyl, 2017-04-28:
Hello, here also running Kali rolling 2017.1 no luck on detecting any devices. daemon runs but fails to show any devices to polychromatic.

hpyl, 2017-04-28:
Update: I managed to bind the driver and I am able to change colours etc. on keyboard manually through /sys

z3ntu, 2017-04-28:
* Please post the output of
`lsusb -d 1532:0225 -v`
and
`ls -lah /dev/input/by-id/`
* Read [this](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol) because we probably need packet captures.
* Screenshots of all available options on Razer Synapse are needed.
* And lastly, the product & model number of the device for documentation reasons please.

terrycain, 2017-04-30:
Yeah `0225` isnt supported yet. Screenshots of synapse before the packet captures would be more useful as I can make educated guesses from that (which are then confirmed by packet captures)

jbagel2, 2017-05-01:
First the requested command output. 

[razer_blade_pro_2017_lsusb.txt](https://github.com/terrycain/razer-drivers/files/968150/razer_blade_pro_2017_lsusb.txt)
[razer_blade_pro_2017_dev_by-id.txt](https://github.com/terrycain/razer-drivers/files/968155/razer_blade_pro_2017_dev_by-id.txt)

terrycain, 2017-05-01:
Awesome

jbagel2, 2017-05-03:
What synapse screenshots are you looking for? I have a Windows vm set up. But I only get one attempts per vm install.. as after the synapse install if I reboot the vm it the guest just bluescreens. But prior to reboot. Synapse works great.

z3ntu, 2017-05-03:
Just go through every tab in Synapse and make a screenshot of that.

jbagel2, 2017-09-19:
sorry for the long delay in response, how do you want me to get you the screenshots?

minecrawler, 2017-09-19:
doesn't matter. Just upload them somewhere. I used imgur for my screenshot scripts, so that's one option. You can also upload them to dropbox. Just make sure people can access and view them.

KazWolfe, 2017-10-10:
For what it's worth, I got (at least some) of the features working on the 2017 Blade Pro by just reusing the code from the older Blade Pro (0210). 

[Here is the diff](https://github.com/openrazer/openrazer/files/1371037/diff.txt) for my changes. Just a simple substitution.

I can control lighting effects directly through the `/sys/bus/hid/drivers/razerkbd` interface, so that's... something I guess.

KazWolfe, 2017-10-11:
Pushed a PR as seen above (sorry for the mess, Git is dumb).

I did test lighting locally, and it all appeared to work decently well, with the exception of ripples which currently appear to be broken.

If possible, I would like to get confirmation from others that this driver version actually works for them too.

PR #407 (2017-10-16): Add support for the Razer Blade Pro 2017
- Add support for Razer Blade Pro 2017 (0x0225) from late 2016 (0x0210)

Fixes #264, for now.

----

This commit effectively just adds the Razer Blade Pro as a valid device and makes it (mostly) work. So far, the only thing broken I have been able to notice is the Ripple effect, though everything else seems to work just fine.
commit 3f270c4c (2017-10-16): Add support for the Razer Blade Pro 2017 (#407)

* Add support for the Razer Blade Pro 2017

- Add support for Razer Blade Pro 2017 (0x0225)
  from late 2016 (0x210)

Fixes #264, for now.

* Add Razer Blade Pro support to readme
closed 2017-10-16 by terrycain