openrazer/openrazer #265: INstall from PPA to mint 18.1 fails

jeremyjjbrown, 2017-04-28:
Attempting to install from  ppa fails on what looks like some python dependencies.

```sh
jeremybr@lake:~ $ sudo add-apt-repository ppa:terrz/razerutils
You are about to add the following PPA:
 This PPA contains open source drivers that allow many Razer peripherals to utilise their full feature set and allow customisation via the daemon.

Supported devices include:

 - BlackWidow (Chroma, Classic, Ultimate)
 - BlackWidow Mamba
 - Blade Stealth
 - Firefly
 - Mamba
 - Abyssus
 - Tartarus

For more information, see here: https://github.com/terrycain/razer-drivers

Licensed under the GPL. These drivers are not affiliated with Razer, Inc.
 More info: https://launchpad.net/~terrz/+archive/ubuntu/razerutils
Press Enter to continue or Ctrl+C to cancel

Executing: /tmp/tmp.oYZ0PnD7lW/gpg.1.sh --keyserver
hkp://keyserver.ubuntu.com:80
--recv-keys
DC9CC596
gpg: requesting key DC9CC596 from hkp server keyserver.ubuntu.com
gpg: key DC9CC596: "Launchpad PPA for Terry Cain" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
jeremybr@lake:~ $ sudo apt-get update
Ign:1 http://packages.linuxmint.com sarah InRelease
Hit:2 http://packages.linuxmint.com sarah Release                                                                                          
Get:3 http://security.ubuntu.com/ubuntu xenial-security InRelease [102 kB]                                                                 
Hit:5 http://download.virtualbox.org/virtualbox/debian xenial InRelease                                                                    
Hit:6 http://ppa.launchpad.net/caffeine-developers/ppa/ubuntu xenial InRelease                                                             
Hit:7 http://archive.ubuntu.com/ubuntu xenial InRelease                                                                                    
Hit:8 http://archive.canonical.com/ubuntu xenial InRelease                                                                              
Get:9 http://ppa.launchpad.net/terrz/razerutils/ubuntu xenial InRelease [18.0 kB]                                                       
Get:10 http://archive.ubuntu.com/ubuntu xenial-updates InRelease [102 kB]                                         
Hit:11 https://apt.dockerproject.org/repo ubuntu-xenial InRelease                                                            
Hit:12 http://ppa.launchpad.net/webupd8team/java/ubuntu xenial InRelease                                                       
Get:13 http://ppa.launchpad.net/terrz/razerutils/ubuntu xenial/main Sources [1,268 B]                              
Get:14 http://archive.ubuntu.com/ubuntu xenial-backports InRelease [102 kB]                        
Get:15 http://ppa.launchpad.net/terrz/razerutils/ubuntu xenial/main amd64 Packages [1,392 B]
Get:16 http://ppa.launchpad.net/terrz/razerutils/ubuntu xenial/main i386 Packages [1,384 B]                     
Get:17 http://ppa.launchpad.net/terrz/razerutils/ubuntu xenial/main Translation-en [844 B]           
Ign:18 http://dl.google.com/linux/chrome-remote-desktop/deb stable InRelease                                                   
Ign:19 http://dl.google.com/linux/chrome/deb stable InRelease
Hit:20 http://dl.google.com/linux/chrome-remote-desktop/deb stable Release
Hit:21 http://dl.google.com/linux/chrome/deb stable Release
Fetched 329 kB in 5s (60.1 kB/s)
Reading package lists... Done
jeremybr@lake:~ $ sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  python3-evdev python3-numpy python3-pyudev python3-setproctitle xautomation
Suggested packages:
  gfortran python-numpy-doc python3-nose python3-numpy-dbg
The following NEW packages will be installed:
  python3-evdev python3-numpy python3-pyudev python3-razer python3-setproctitle razer-daemon razer-doc razer-kernel-modules-dkms
  xautomation
0 upgraded, 9 newly installed, 0 to remove and 4 not upgraded.
Need to get 1,982 kB of archives.
After this operation, 11.0 MB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://ppa.launchpad.net/terrz/razerutils/ubuntu xenial/main amd64 python3-evdev amd64 0.6.4-0ubuntu1 [30.2 kB]
Get:2 http://archive.ubuntu.com/ubuntu xenial/main amd64 python3-numpy amd64 1:1.11.0-1ubuntu1 [1,762 kB]
Get:3 http://ppa.launchpad.net/terrz/razerutils/ubuntu xenial/main amd64 python3-razer all 1.1.11-0ubuntu1 [17.8 kB]
Get:4 http://ppa.launchpad.net/terrz/razerutils/ubuntu xenial/main amd64 razer-kernel-modules-dkms amd64 1.1.11-0ubuntu1 [41.1 kB]
Get:5 http://ppa.launchpad.net/terrz/razerutils/ubuntu xenial/main amd64 razer-daemon amd64 1.1.11-0ubuntu1 [45.8 kB]
Get:6 http://ppa.launchpad.net/terrz/razerutils/ubuntu xenial/main amd64 razer-doc all 1.1.11-0ubuntu1 [5,424 B]
Get:7 http://archive.ubuntu.com/ubuntu xenial/main amd64 python3-pyudev all 0.16.1-3 [29.2 kB]
Get:8 http://archive.ubuntu.com/ubuntu xenial/universe amd64 python3-setproctitle amd64 1.1.8-1build2 [14.2 kB]
Get:9 http://archive.ubuntu.com/ubuntu xenial/universe amd64 xautomation amd64 1.09-1 [36.8 kB]
Fetched 1,982 kB in 3s (501 kB/s)     
Selecting previously unselected package python3-evdev.
(Reading database ... 353881 files and directories currently installed.)
Preparing to unpack .../python3-evdev_0.6.4-0ubuntu1_amd64.deb ...
Unpacking python3-evdev (0.6.4-0ubuntu1) ...
Selecting previously unselected package python3-numpy.
Preparing to unpack .../python3-numpy_1%3a1.11.0-1ubuntu1_amd64.deb ...
Unpacking python3-numpy (1:1.11.0-1ubuntu1) ...
Selecting previously unselected package python3-pyudev.
Preparing to unpack .../python3-pyudev_0.16.1-3_all.deb ...
Unpacking python3-pyudev (0.16.1-3) ...
Preparing to unpack .../python3-razer_1.1.11-0ubuntu1_all.deb ...
Unpacking python3-razer (1.1.11-0ubuntu1) ...
dpkg: error processing archive /var/cache/apt/archives/python3-razer_1.1.11-0ubuntu1_all.deb (--unpack):
 trying to overwrite '/usr/lib/python3/dist-packages/razer/__init__.py', which is also in package razer-chroma-driver 1.0.0
Selecting previously unselected package razer-kernel-modules-dkms.
Preparing to unpack .../razer-kernel-modules-dkms_1.1.11-0ubuntu1_amd64.deb ...
Unpacking razer-kernel-modules-dkms (1.1.11-0ubuntu1) ...
dpkg: error processing archive /var/cache/apt/archives/razer-kernel-modules-dkms_1.1.11-0ubuntu1_amd64.deb (--unpack):
 trying to overwrite '/usr/src/razer_chroma_driver-1.0.0/driver/razermouse_driver.c', which is also in package razer-chroma-driver 1.0.0
Selecting previously unselected package python3-setproctitle.
Preparing to unpack .../python3-setproctitle_1.1.8-1build2_amd64.deb ...
Unpacking python3-setproctitle (1.1.8-1build2) ...
Selecting previously unselected package xautomation.
Preparing to unpack .../xautomation_1.09-1_amd64.deb ...
Unpacking xautomation (1.09-1) ...
Selecting previously unselected package razer-daemon.
Preparing to unpack .../razer-daemon_1.1.11-0ubuntu1_amd64.deb ...
Unpacking razer-daemon (1.1.11-0ubuntu1) ...
Selecting previously unselected package razer-doc.
Preparing to unpack .../razer-doc_1.1.11-0ubuntu1_all.deb ...
Unpacking razer-doc (1.1.11-0ubuntu1) ...
Processing triggers for man-db (2.7.5-1) ...
Errors were encountered while processing:
 /var/cache/apt/archives/python3-razer_1.1.11-0ubuntu1_all.deb
 /var/cache/apt/archives/razer-kernel-modules-dkms_1.1.11-0ubuntu1_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
jeremybr@lake:~ $ sudo apt-get remove razer-chroma-driver
Reading package lists... Done
Building dependency tree       
Reading state information... Done
You might want to run 'apt-get -f install' to correct these:
The following packages have unmet dependencies:
 razer-daemon : Depends: razer-kernel-modules-dkms but it is not going to be installed
E: Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution).
```

lah7, 2017-04-28:
> `trying to overwrite '/usr/src/razer_chroma_driver-1.0.0/driver/razermouse_driver.c',` **`which is also in package razer-chroma-driver 1.0.0`**

`razer-chroma-driver` is an old (very old) package, which is causing this conflict, please uninstall that and try again.

jeremyjjbrown, 2017-04-28:
@Iah7

`sudo apt-get remove -f razer-chroma-driver`
`sudo apt-get purge razer-chroma-driver`

but it seems it's still there.

```sh
sudo apt install   python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  gir1.2-javascriptcoregtk-4.0 gir1.2-webkit2-4.0 libasound2-dev libdbus-1-dev libdrm-dev libegl1-mesa-dev libfftw3-3 libfftw3-long3
  libgl1-mesa-dev libgles2-mesa-dev libglib2.0-dev libglu1-mesa-dev libice-dev libmirclient-dev libmircommon-dev libmircookie-dev
  libmircookie2 libpcre3-dev libpcre32-3 libpcrecpp0v5 libprotobuf-dev libpthread-stubs0-dev libpulse-dev libsdl2-2.0-0 libsdl2-dev
  libsdl2-image-2.0-0 libsdl2-image-dev libsm-dev libsndio-dev libsndio6.1 libudev-dev libwayland-dev libx11-dev libx11-xcb-dev libxau-dev
  libxcb-dri2-0-dev libxcb-dri3-dev libxcb-glx0-dev libxcb-present-dev libxcb-randr0-dev libxcb-render0-dev libxcb-shape0-dev
  libxcb-sync-dev libxcb-xfixes0-dev libxcb1-dev libxcursor-dev libxdamage-dev libxdmcp-dev libxext-dev libxfixes-dev libxi-dev
  libxinerama-dev libxkbcommon-dev libxrandr-dev libxrender-dev libxshmfence-dev libxss-dev libxt-dev libxv-dev libxxf86vm-dev
  mesa-common-dev x11proto-core-dev x11proto-damage-dev x11proto-dri2-dev x11proto-fixes-dev x11proto-gl-dev x11proto-input-dev
  x11proto-kb-dev x11proto-randr-dev x11proto-render-dev x11proto-scrnsaver-dev x11proto-video-dev x11proto-xext-dev
  x11proto-xf86vidmode-dev x11proto-xinerama-dev xorg-sgml-doctools xtrans-dev
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  python3-razer razer-daemon razer-doc razer-kernel-modules-dkms
0 upgraded, 4 newly installed, 0 to remove and 8 not upgraded.
Need to get 0 B/110 kB of archives.
After this operation, 821 kB of additional disk space will be used.
Selecting previously unselected package python3-razer.
(Reading database ... 354305 files and directories currently installed.)
Preparing to unpack .../python3-razer_1.1.11-0ubuntu1_all.deb ...
Unpacking python3-razer (1.1.11-0ubuntu1) ...
Selecting previously unselected package razer-kernel-modules-dkms.
Preparing to unpack .../razer-kernel-modules-dkms_1.1.11-0ubuntu1_amd64.deb ...
Unpacking razer-kernel-modules-dkms (1.1.11-0ubuntu1) ...
Selecting previously unselected package razer-daemon.
Preparing to unpack .../razer-daemon_1.1.11-0ubuntu1_amd64.deb ...
Unpacking razer-daemon (1.1.11-0ubuntu1) ...
Selecting previously unselected package razer-doc.
Preparing to unpack .../razer-doc_1.1.11-0ubuntu1_all.deb ...
Unpacking razer-doc (1.1.11-0ubuntu1) ...
Processing triggers for man-db (2.7.5-1) ...
Setting up python3-razer (1.1.11-0ubuntu1) ...
Setting up razer-kernel-modules-dkms (1.1.11-0ubuntu1) ...
Removing old razer_chroma_driver-1.0.0 DKMS files...
Loading new razer_chroma_driver-1.0.0 DKMS files...
Error! DKMS tree already contains: razer_chroma_driver-1.0.0
You cannot add the same module/version combo more than once.
dpkg: error processing package razer-kernel-modules-dkms (--configure):
 subprocess installed post-installation script returned error exit status 3
dpkg: dependency problems prevent configuration of razer-daemon:
 razer-daemon depends on razer-kernel-modules-dkms; however:
  Package razer-kernel-modules-dkms is not configured yet.

dpkg: error processing package razer-daemon (--configure):
 dependency problems - leaving unconfigured
Setting up razer-doc (1.1.11-0ubuntu1) ...
Errors were encountered while processing:
 razer-kernel-modules-dkms
 razer-daemon
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

How do I totally remove it?

jeremyjjbrown, 2017-04-28:
WARNING: if you try below have another regular keyboard handy until you can reinstall the newer chroma driver.

 `sudo rm -r /var/lib/dkms/razer_chroma_driver`  did it.

closed 2017-04-28 by jeremyjjbrown

z3ntu, 2017-04-28:
Did you get it working?