openrazer/openrazer #266: Razer Ouroboros (1532:0033) support needed

freerom, 2017-05-01:
Hello, this is my first issue on GitHub. (Sorry but my English is very poor, I'm French)
I have a recent version of ouroboros mouse and it is not a supported device.
Linux mint 18.1
Driver and daemon are installed.

I understood you need capture packets made with wireshark, I will try to do it.

**Razer Ouroboros**
    ID 1532:0033
    Model No. RC30-007701
    Product No. RZ01-00770100

**lsusb -d 1532:0033 -v**
```
Bus 001 Device 002: ID 1532:0033 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0033 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```
**ls -lah /dev/input/by-id/**
```
total 0
drwxr-xr-x 2 root root 220 mai    1 10:15 .
drwxr-xr-x 4 root root 560 mai    1 10:15 ..
lrwxrwxrwx 1 root root   9 mai    1 10:15 usb-Razer_Razer_Ouroboros_Gaming_Dock-event-mouse -> ../event3
lrwxrwxrwx 1 root root   9 mai    1 10:15 usb-Razer_Razer_Ouroboros_Gaming_Dock-if01-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 mai    1 10:15 usb-Razer_Razer_Ouroboros_Gaming_Dock-if02-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 mai    1 10:15 usb-Razer_Razer_Ouroboros_Gaming_Dock-mouse -> ../mouse0
```

**Screenshots**

[http://img1.imagilive.com/0517/cali.jpg](url)
[http://img1.imagilive.com/0517/custo.jpg](url)
[http://img1.imagilive.com/0517/light.jpg](url)
[http://img1.imagilive.com/0517/perf.jpg](url)
[http://img1.imagilive.com/0517/power.jpg](url)

terrycain, 2017-05-01:
Need packet captures for the following (each item in its own file please)
* Wired Brightness from off -> bright -> off -> bright a few times
* Wireless Brightness from off -> bright -> off -> bright a few times
* DPI 100 -> 8200 a few times
* Polling rate 125 -> 1000 a few times
* Sleep mode from 1 -> 15 a few times
* Flash battery indicator 5->25 a few times
---
* Capture starting from before the VM starts lasting all the way until synapse is open
* Capture of mouse not moving for 10 mins

Also could do with the two ls outputs for when the mouse is wired also (as it should have a different usb ID)

freerom, 2017-05-02:
Sorry but i don't understand wireshark filter, i can't find the good filter.
i've read different issues and your reverse engineering USB protocol page more and more (and other) but it is not "friendly user" for me. [https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol](url)
[https://wiki.wireshark.org/CaptureSetup/CapturePrivileges](url)
[https://wiki.wireshark.org/CaptureSetup/USB#USB_attached_network_interfaces](url)

**lsusb** 
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 008: ID 8087:0a2b Intel Corp. 
Bus 001 Device 006: ID 045e:028e Microsoft Corp. Xbox360 Controller
Bus 001 Device 007: ID 413c:2110 Dell Computer Corp. 
Bus 001 Device 005: ID 413c:1010 Dell Computer Corp. 
Bus 001 Device 004: ID 1532:0033 Razer USA, Ltd 
Bus 001 Device 003: ID 1908:0226 GEMBIRD 
Bus 001 Device 009: ID 046d:c01d Logitech, Inc. MX510 Optical Mouse
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
(i don't want to use my old MX510.)
wireless mode for the ouroboros.

i try these filter: 
`usb.bus_id == 1 && usb.device_address == 4`

See the result on the link below (idon't know how insert pictures.)
[http://img1.imagilive.com/0517/Sans_titre.png](url)

And if i try on usbmon1 without filter, there are all USB devices captured. (MX510, keyboard, etc.)

This is new for me and this is not easy... But i want to go to the finish line, can you help me?

lezed1, 2017-05-02:
@freerom Try without the "Capture filter" and then put that same expression in the "Display filter" (the text box above "Welcome to Wireshark". That seems to work for me.

freerom, 2017-05-02:
@terrycain this is a problem for you if each files you demand are with all USB devices?

terrycain, 2017-05-02:
@freerom nah should be ok

freerom, 2017-05-02:
[https://github.com/freerom/RazerOuroborosPacketCapture](url)
This is not the same device id for the wired brightness packet.
I wish is it good for you.

freerom, 2017-05-05:
@z3ntu What does driver support label mean?

lezed1, 2017-05-05:
@freerom The kernel module needs to be modified to add support for this device.

terrycain, 2017-06-25:
@freerom there should be a different USB PID for wired, and wireless mode. Wireless being the PID of the dock. what are the 2 PIDs? I have 0033

terrycain, 2017-06-25:
What OS are you on also? I have a prototype driver to test

freerom, 2017-07-05:
@terrycain Linuxmint 18.1 64 bits

z3ntu, 2017-07-05:
[feature_ouroboros0x0033.zip](https://github.com/terrycain/razer-drivers/files/1125762/feature_ouroboros0x0033.zip)
Please install these debs and reboot.

freerom, 2017-08-13:
system said: _The same version is already installed_ for all the .deb

z3ntu, 2017-08-13:
uhm google how to force reinstall these (probably a --reinstall flag?)

z3ntu, 2017-10-21:
@freerom Are you still using Linux Mint 18?

iiikhsan, 2019-04-18:
Hi,
I'm wondering whether there's update for this?

arklese1zure, 2019-05-14:
Hey, I have one of these as well (1532:0033). Do you need any help with packet capture or testing?

I'm currently running Manjaro.

Demayl, 2019-06-07:
Same here - on wireless mode it reports 0033 as PID and in wired - 0032

z3ntu, 2019-06-16:
To test it on Arch Linux (+derivatives), please clone the openrazer-git repository from the AUR, then edit the PKGBUILD where it says `source=("git+https://github.com/openrazer/openrazer.git")` and replace it with `source=("git+https://github.com/openrazer/openrazer.git#branch=feature_ouroboros0x0033")` , build the package as you would normally and test it.

CobrianTinander, 2025-07-13:
Any chance anyone could update this to the latest build so I could test it, confirm it works, and finally mark this as complete?