openrazer/openrazer #267: Support for Razer Turret

gttalbot, 2017-05-02:
[razer_turret_lsusb.txt](https://github.com/terrycain/razer-drivers/files/970261/razer_turret_lsusb.txt)
[razer_turret_dev.txt](https://github.com/terrycain/razer-drivers/files/970264/razer_turret_dev.txt)

This doesn't have any of the fancy lighting...it's a keyboard and mouse combination that's wireless for sitting on the couch.

I don't currently have a Windows VM, but can create one at some point soon if that seems necessary.  (This is a pretty basic keyboard/mouse...)

gttalbot, 2017-05-02:
The product is the Razer Turret gaming lapboard, model number RZ03-0133, product number RZ84-01330100.  It can connect to the machine by either a USB dongle that they provide or via bluetooth.  It has a charging stand that's USB as well.

terrycain, 2017-05-02:
If its a standard keyboard and mouse and doesn't do anything fancy, it won't really need driver support right? Unless there's some Razer features your after

gttalbot, 2017-05-02:
That's possibly true.  I didn't know whether there were any adjustments, for example, to mouse sensitivity or anything like that.  If not, no big deal.

terrycain, 2017-05-02:
Ahh there may well be. Wether it be DPI settings for the mouse or just battery levels. That's what the screenshots from Razer Synapse on Windows tell us.:wink:

closed 2017-09-01 by terrycain