openrazer/openrazer #269: Question: Can I set DPI via sysfs?

borgdylan, 2017-05-03:
The last polychromatic update regressed big time, and so I have to change the DPI on my mouse without a GUI. I am not managing to write to the dpi file in sysfs.  It keeps stuck to the value "500:500".

borgdylan, 2017-05-03:
I installed d-feet and managed to set the DPI via the  DBUS interface. The maximum DPI returned is incorrect. The DA Elite has a max DPI of 16000 not 500. The latest polychromatic is depending on the maximum DPI numbers returned by the daemon to limit the DPI values settable by the user.

terrycain, 2017-05-03:
Ahh I see the issue, I forgot the entry stating the DPI and it falls back to 500.

34bfb88234c5f622bdd6323d05cb2c25424610a1 Fixes your issue, I'll try and release a new version soonish.

borgdylan, 2017-05-03:
Thank you! And takes for letting me set the DPI via DBUS regardless of the max DPI value! I almost had to go through the painful act of moving the pointer at snail speed.

terrycain, 2017-05-03:
Ha :smile: no problem, I set the low speed on purpose purely so it would be reported if I missed a device :wink:

borgdylan, 2017-05-03:
IC. The DPI buttons on the mouse are life savers in cases like these but they seem to skip the 2500,2500 value that i like to use.

terrycain, 2017-05-03:
They're stored in memory on the mouse I think. Changing those settings is on my todo list.

borgdylan, 2017-05-03:
Yes the devices have EEPROMS on them AFAIK. The keyboard remembers the exact lighting profile value across reboots, so does the mouse.

terrycain, 2017-05-03:
Yep, the keyboards used to store macro info too but that's long since gone

borgdylan, 2017-05-03:
However I'm noticing that on powering down the machine, the keyboard's lights go off but the mouse stays lit up (unplugging works to turn it off as it does not turn on unless I boot up).

terrycain, 2017-05-03:
That i dont think ive come across, then again my pc rarely sleeps

borgdylan, 2017-05-03:
Mine is shut down everyday, as it consumes a lot of power (i7 6900K with a GTX 1080)

terrycain, 2017-05-03:
Packages are in launchpad now, should be available on the PPA in the next 10 mins or so. Other repositories like arch, fedora will a while longer.

closed 2017-05-03 by terrycain

borgdylan, 2017-05-03:
The update fixes the issue.

remilia15, 2017-05-04:
@borgdylan Idk what is this seriously :3

remilia15, 2017-05-04:
@terrycain This looks like for RIL