openrazer/openrazer #270: Why this driver is not part of Kernel ?

ikus060, 2017-05-04:
Any reason why this driver is not part of the official Linux Kernel ?

z3ntu, 2017-05-04:
See #29 probably

terrycain, 2017-05-04:
Currently driver still changes a bit too much anyway so having it in mainline kernel would restrict that more than I would like.

closed 2017-05-04 by terrycain

z3ntu, 2017-05-04:
And also new device support would take forever to land in some distros (eg Debian/CentOS (yes, people tried to use the driver on CentOS) or Ubuntu LTS).

madscientist42, 2017-05-04:
Yep.  It REALLY should be a userspace beastie...  Under Windows...they
don't bolt it on that way and since there's loads of hook ability to make
it work where it needs.to...I'd chimed in ages ago with a bit on Insight
back when GKH spoke up.  I just got mired down with the thing called 'life'
and didn't get done on my end of things.  About to change.  Look in the
linked thread for more...

Headcrabed, 2026-10-15:
I think we need to get this mainlined as fast as possible....

https://onlineservices.epicgames.com/news/introducing-easy-anti-cheat-linux-v2

lah7, 2026-10-15:
That's good to know about anti-cheat. Most parts of OpenRazer would be better off moved to user space though, see also:

* https://github.com/openrazer/openrazer/issues/311