openrazer/openrazer #272: Applying settings on reconnecting

simonszu, 2017-05-05:
Currently i am sharing a keyboard and a razer mouse between two PCs with the help of a KVM switch. 
I have noticed that, after enabling a color pattern on the linux PC with the help of this driver suite, and switching to the other PC which is running Windows with the official drivers, the color pattern of the official drivers are applied. But after switching back to linux, the color patterns from your driver suite aren't re-applied. 
So i am kindly asking for a possibility to create a udev hook or something, so that i can make the mouse breathe in green when connected to linux and breathe in e.g. blue when connected to the windows host. 
Or maybe this is already possible but i don't know how? If so, please tell me how to achieve this. 

Thank you very much.

lah7, 2017-05-05:
It's certainly possible using the Python library to achieve this. There are signals emitted from DBUS to detect when a device is plugged in, and you can script it to find your mouse and do whatever you wish.

Unfortunately, I don't know what the code is for listening to DBUS signals in Python.

terrycain, 2017-06-15:
@lah7 https://dbus.freedesktop.org/doc/dbus-python/doc/tutorial.html#receiving-signals
Will make better example at some point.

@simonszu Its something I would like to do but am super busy at the moment. As @lah7 said you can use the python library to do this, there are examples [here](https://github.com/terrycain/razer-drivers/tree/master/examples)

closed 2017-06-15 by terrycain