openrazer/openrazer #274: Razer Hex not listed on device support (1532:0036)

CMDRHorn, 2017-05-07:
Arch linux

Been trying to get the device working for months now and finally decided to look at the PID list and saw that mine isn't even listed.

`[root@Rogue-9 hid-generic]# ls -l /sys/bus/hid/devices/
total 0
lrwxrwxrwx 1 root root 0 May  7 12:19 0003:1532:0036.0001 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6:1.0/0003:1532:0036.0001
lrwxrwxrwx 1 root root 0 May  7 12:19 0003:1532:0036.0002 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6:1.1/0003:1532:0036.0002
lrwxrwxrwx 1 root root 0 May  7 12:19 0003:1B1C:1B09.0004 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-10/3-10:1.0/0003:1B1C:1B09.0004
lrwxrwxrwx 1 root root 0 May  7 12:19 0003:1B1C:1B09.0005 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-10/3-10:1.1/0003:1B1C:1B09.0005
lrwxrwxrwx 1 root root 0 May  7 12:19 0003:1B1C:1B09.0006 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-10/3-10:1.2/0003:1B1C:1B09.0006
lrwxrwxrwx 1 root root 0 May  7 12:19 0003:1B1C:1C0D.0003 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7:1.0/0003:1B1C:1C0D.0003
[root@Rogue-9 hid-generic]# lsusb |grep '1532:'
Bus 003 Device 005: ID 1532:0036 Razer USA, Ltd RZ01-0075, Gaming Mouse [Naga Hex]
`

z3ntu, 2017-05-07:
* Please post the output of
`lsusb -d 1532:0036 -v`
and
`ls -lah /dev/input/by-id/`
* Read [this](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol) because we probably need packet captures.
* Screenshots of all available options on Razer Synapse are needed.
* And lastly, the product & model number of the device for documentation reasons please.

CMDRHorn, 2017-05-07:
[root@Rogue-9 hid-generic]# lsusb -d 1532:0036 -v

Bus 003 Device 005: ID 1532:0036 Razer USA, Ltd RZ01-0075, Gaming Mouse [Naga Hex]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0036 RZ01-0075, Gaming Mouse [Naga Hex]
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Naga Hex
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     103
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0009  1x 9 bytes
        bInterval               8
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

CMDRHorn, 2017-05-07:
[root@Rogue-9 hid-generic]# ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 160 May  7 12:20 .
drwxr-xr-x 4 root root 560 May  7 12:20 ..
lrwxrwxrwx 1 root root   9 May  7 12:20 usb-Corsair_Corsair_K70R_Gaming_Keyboard-event-if01 -> ../event3
lrwxrwxrwx 1 root root   9 May  7 12:20 usb-Corsair_Corsair_K70R_Gaming_Keyboard-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 May  7 12:20 usb-Corsair_Corsair_K70R_Gaming_Keyboard-if02-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 May  7 12:20 usb-Razer_Razer_Naga_Hex-event-mouse -> ../event0
lrwxrwxrwx 1 root root   9 May  7 12:20 usb-Razer_Razer_Naga_Hex-if01-event-kbd -> ../event1
lrwxrwxrwx 1 root root   9 May  7 12:19 usb-Razer_Razer_Naga_Hex-mouse -> ../mouse0
[root@Rogue-9 hid-generic]#

CMDRHorn, 2017-05-07:
PN: RZ01-00750200
MN: RZ01-0075

working on getting setup for packet captures

z3ntu, 2017-05-07:
Great! :) Thanks for the quick response btw (some people take ages for everything)

z3ntu, 2017-05-07:
Quick note: Please send us the screenshots first because there are some "Razer Naga Hex" that are `	1532:0041` but yours is `1532:0036`. If you/we are lucky, then the mouse is the same except the usb pid.

CMDRHorn, 2017-05-07:
If contributors are willing to respond quickly, it only makes sense that whomever opens the ticket do the same.

Gotta spin up a VM to get synapse screenshots. This will take a little while, but I'll get them in asap.

z3ntu, 2017-05-07:
Awesome!

terrycain, 2017-05-07:
@CMDRHorn 👍

CMDRHorn, 2017-05-07:
Any other screens needed?

[image]
[image]
[image]

CMDRHorn, 2017-05-07:
Any other screens?

[image]
[image]
[image]

terrycain, 2017-05-07:
Looks good, couple of packet captures should do it, capture changing the dpi from min to max, and then turning lights on and off

CMDRHorn, 2017-05-07:
[wireshark2.zip](https://github.com/terrycain/razer-drivers/files/981891/wireshark2.zip)
Hope that worked. Never had to use wireshark before.

terrycain, 2017-05-07:
Looks good. Will drum something up this week.

CMDRHorn, 2017-05-07:
Sounds good. Thank you.

terrycain, 2017-05-20:
@CMDRHorn what distro you running/

CMDRHorn, 2017-05-20:
Arch

terrycain, 2017-05-20:
@z3ntu instructions to get branch `feature_naga_hex` plz

z3ntu, 2017-05-20:
You probably will get some dependency issues because some packages needed are only available in the AUR -> install them first (AUR helper or whatever you like)
**Run the following commands:**
```
git clone https://aur.archlinux.org/razer-drivers-git.git
cd razer-drivers-git
curl https://gist.githubusercontent.com/z3ntu/28cb958ce9a778de0975e54c54b77318/raw/94d43d029f4746cfa62bdffd7064f1f6944d9ad9/PKGBUILD.naga_hex -o PKGBUILD
makepkg -sr
sudo pacman -U python-razer-git-*.tar* razer-daemon-git-*.tar* razer-driver-dkms-git-*.tar*
```
If you need additional help, you can reach me on Telegram @z3ntu.

CMDRHorn, 2017-05-21:
Looks like it's working, thank you!

When I turn off the lights my DPI changes for some reason, I'm assuming that isn't supposed to happen?
[image]
[image]

z3ntu, 2017-05-21:
Please run (and send us the output)
```
cd $(dirname /sys/bus/hid/drivers/razermouse/0*/device_serial)
cat device_mode 
cat device_serial 
cat device_type 
cat dpi 
cat firmware_version 
cat poll_rate 
```

CMDRHorn, 2017-05-21:
[corran@Rogue-9 razer-drivers-git]$ cd $(dirname /sys/bus/hid/drivers/razermouse/0*/device_serial)
[corran@Rogue-9 0003:1532:0036.0006]$ cat device_mode 
0:0
[corran@Rogue-9 0003:1532:0036.0006]$ cat device_serial 
PM1347075302576
[corran@Rogue-9 0003:1532:0036.0006]$ cat device_type 
Razer Naga Hex (Red)
[corran@Rogue-9 0003:1532:0036.0006]$ cat dpi 
117:117
[corran@Rogue-9 0003:1532:0036.0006]$ cat firmware_version 
v2.1
[corran@Rogue-9 0003:1532:0036.0006]$ cat poll_rate 
1000

terrycain, 2017-05-21:
Do the lights work? and does the mouse feel different when the "dpi changes"

CMDRHorn, 2017-05-21:
@terrycain yes to both. the DPI just adjusts down to a seemingly random number whenever I turn off the lights with the toggle.

terrycain, 2017-05-21:
cd to the mouse folder in /sys/
and run l`s -lah`
it should contain some files like `device_mode` and `device_serial`

CMDRHorn, 2017-05-21:
[corran@Rogue-9 0003:1532:0036.0006]$ ls -lah
total 0
drwxr-xr-x 5 root plugdev    0 May 21 08:05 .
drwxr-xr-x 5 root root       0 May 21 08:05 ..
-r--r--r-- 1 root plugdev 4.0K May 21 08:05 country
-rw-rw---- 1 root plugdev 4.0K May 21 08:05 device_mode
-r--r----- 1 root plugdev 4.0K May 21 08:05 device_serial
-r--r----- 1 root plugdev 4.0K May 21 08:05 device_type
-rw-rw---- 1 root plugdev 4.0K May 21 08:18 dpi
lrwxrwxrwx 1 root plugdev    0 May 21 08:05 driver -> ../../../../../../../bus/hid/drivers/razermouse
-r--r----- 1 root plugdev 4.0K May 21 08:05 firmware_version
drwxr-xr-x 3 root plugdev    0 May 21 08:05 hidraw
drwxr-xr-x 3 root plugdev    0 May 21 08:05 input
-rw-rw---- 1 root plugdev 4.0K May 21 08:17 logo_led_state
-r--r--r-- 1 root plugdev 4.0K May 21 08:05 modalias
-rw-rw---- 1 root plugdev 4.0K May 21 08:08 poll_rate
drwxr-xr-x 2 root plugdev    0 May 21 08:05 power
-r--r--r-- 1 root plugdev 4.0K May 21 08:05 report_descriptor
-rw-rw---- 1 root plugdev 4.0K May 21 08:17 scroll_led_state
lrwxrwxrwx 1 root plugdev    0 May 21 08:05 subsystem -> ../../../../../../../bus/hid
--w--w---- 1 root plugdev 4.0K May 21 08:05 test
-rw-r--r-- 1 root plugdev 4.0K May 21 08:05 uevent
-r--r----- 1 root plugdev 4.0K May 21 08:05 version

terrycain, 2017-05-21:
```
cat scroll_led_state
cat logo_led_state
cat dpi
```

z3ntu, 2017-05-21:
@terrycain https://github.com/terrycain/razer-drivers/issues/274#issuecomment-302936100 ;)

CMDRHorn, 2017-05-21:
[corran@Rogue-9 0003:1532:0036.0006]$ cat scroll_led_state
1
[corran@Rogue-9 0003:1532:0036.0006]$ cat logo_led_state
1
[corran@Rogue-9 0003:1532:0036.0006]$ cat dpi                                                                                                                                                                                                  
124:124

terrycain, 2017-05-21:
ok try 
echo -ne '\x3c\x3c' > ./dpi
cat dpi

CMDRHorn, 2017-05-21:
[corran@Rogue-9 0003:1532:0036.0006]$ echo -ne '\x3c\x3c' > ./dpi
[corran@Rogue-9 0003:1532:0036.0006]$  cat dpi                                                                                                                                                                                                                                 
60:60

terrycain, 2017-05-21:
ok now
```
echo -n '0' > scroll_led_state
echo -n '0' > logo_led_state
cat scroll_led_state
cat logo_led_state
cat dpi
```

CMDRHorn, 2017-05-21:
Interesting...no change from CLI  only when from the GUI

[corran@Rogue-9 0003:1532:0036.0006]$ echo -n '0' > scroll_led_state                                                                                                                                                                                                           
[corran@Rogue-9 0003:1532:0036.0006]$ echo -n '0' > logo_led_state                                                                                                                                                                                                             
[corran@Rogue-9 0003:1532:0036.0006]$ cat scroll_led_state                                                                                                                                                                                                                     
0                                                                                                                                                                                                                                                                              
[corran@Rogue-9 0003:1532:0036.0006]$ cat logo_led_state                                                                                                                                                                                                                       
0                                                                                                                                                                                                                                                                              
[corran@Rogue-9 0003:1532:0036.0006]$ cat dpi                                                                                                                                                                                                                                  
60:60                                                                                                                                                                                                                                                                          
[corran@Rogue-9 0003:1532:0036.0006]$

terrycain, 2017-05-21:
ok so that proves the physical dpi doesnt change when the lighting changes.
The issue is (pretty sure i fixed it with commit d668637) that for the Naga Hex's (earlyish models) the DPI is stored in 1 byte (with a max of 255 values) so converting 5600 into a number that fits into 1 bytes worth of space will have some rounding errors. That plus the math in the daemon to covert the byte back to 5600 range was wrong.

@z3ntu how do you re-install from the branch? Once you've reinstalled from the branch, restart and the ui should give you slightly nicer dpi results now (though it might jump around +-50 from what you set).

z3ntu, 2017-05-21:
Yeah just run `makepkg -sri` again in the folder from before

CMDRHorn, 2017-05-21:
Exactly as you said. it now no longer changes when I adjust the DPI, but when i set it, it moves -+50 or so. Nothing to write home about. Thank you both!

terrycain, 2017-05-21:
Np. Will merge this into main codebase. Will probably release it properly soon.

terrycain, 2017-06-11:
c13cc1a0225e3002114810a16f8bc9f212de315d v1.1.13 released. Should be available in next 10 mins or so

closed 2017-06-11 by terrycain