openrazer/openrazer #275: DeathStalker Chroma no lights

MarioManChroma, 2017-05-08:
Hi... 

yesterday i got my DeathStalker keyboard and i am very happy with your driver, because it's work great. I am using ubuntu 16.04 right now and a startech triple display 4 port kvm. I am switching between windows and linux systems. that works also great!

Over the night, i have full turned off my system. Now i have it reactivated, but the backlights on my keyboard is not on!

Following works fine:
echo -n "1" > game_led_state
echo -n "0" > game_led_state

This has no effect, but yesterday have works perfectly:
echo -n "1" > matrix_effect_wave
echo -n "255" > matrix_brightness
echo -n -e "\xFF\xFF\xFF" > matrix_effect_static

I hope someone have an any idea, what's happen...
THX

MarioManChroma, 2017-05-08:
Additional information: If i disconnect and connect again the usb connector on the kvm, it works again.
But i want to avoid, that i need to do this procedure every time.

z3ntu, 2017-05-08:
So you are doing this inside a virtual machine?

MarioManChroma, 2017-05-08:
No... that are all real physical machines.

z3ntu, 2017-05-08:
> on the kvm

?

MarioManChroma, 2017-05-08:
Sorry, if my english is bad... I am from austria... :-)

z3ntu, 2017-05-08:
no problem ;) I'm from Austria too. If you have Telegram you can write me @z3ntu and I can help you in German (otherwise you can write at least some parts in German too). (hopefully others don't mind too much)

MarioManChroma, 2017-05-08:
oh nice... yes i have telegram!

MarioManChroma, 2017-05-17:
I have it fixed now! :-)

My KVM switch has 4 usb ports. two for usb devices like cam or stick and two dedicated ports for mouse and keyboard. 

Until now i used only the USB for cam or sticks. Because, the USB port for keyboard is general not working. But now i use the USB port for the mouse an all works perfektly. Except the hotkey feature from KVM. This works only with the USB port for keyboard. But here, i think, we have the general compatibility problem with Razer and KVM's. But at this time, it is ok for me.

Nice day!

MarioManChroma, 2017-05-17:
Additional Info:
This is the KVM Switch, that i am using!
https://www.startech.com/Server-Management/KVM-Switches/4-Port-Triple-Monitor-DVI-USB-KVM-Switch~SV431TDVIUA

BR MMC

closed 2017-05-17 by MarioManChroma

z3ntu, 2017-05-17:
👍