openrazer/openrazer #278: Abyssus V2 (1532:005b) Support

iamshapeless, 2017-05-10:
I have the drivers installed on Manjaro Linux for my BlackWidow X TE Chroma, but my mouse doesn't work. Is there a way to make the mouse use normal drivers instead of the razer ones until support is added?

tildearrow, 2017-05-10:
GO
AWAY!1!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!11!!!!e1

tildearrow, 2017-05-10:
Sorry, that was an accidental comment.

lezed1, 2017-05-10:
lol

lezed1, 2017-05-10:
You can unbind the device using these commands:

```
cd /sys/bus/hid/drivers/razermouse
ls # This will show you IDs of devices
echo -n "DEVICE_ID" > unbind
echo -n "DEVICE_ID" > ../hid-generic/bind"
```

The device ID will look like `000X:1532:005B:000X` (there might be multiple devices, you can rebind all of them). Also, you will not be able to use your mouse between the unbind and bind steps.

terrycain, 2017-05-15:
Then again, udev shouldnt unbind it as its not supported. @z3ntu paste support instructions

z3ntu, 2017-05-16:
* Please post the output of
`lsusb -d 1532:005b -v`
and
`ls -lah /dev/input/by-id/`
* Read [this](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol) because we probably need packet captures.
* Screenshots of all available options on Razer Synapse are needed.
* And lastly, the product & model number of the device for documentation reasons please.

iamshapeless, 2017-05-19:
> Please post the output of lsusb -d 1532:005b -v

```
Bus 002 Device 004: ID 1532:005b Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x005b 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

> and ls -lah /dev/input/by-id/

```
total 0
drwxr-xr-x 2 root root 180 May 19 01:06 .
drwxr-xr-x 4 root root 500 May 19 01:06 ..
lrwxrwxrwx 1 root root   9 May 19 01:06 usb-0461_DELL_Laser_Mouse-event-mouse -> ../event1
lrwxrwxrwx 1 root root   9 May 19 01:06 usb-0461_DELL_Laser_Mouse-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 May 19 01:06 usb-04G6200086K1AN22200000P_ASUS_USB2.0_WebCam-event-if00 -> ../event4
lrwxrwxrwx 1 root root   9 May 19 01:06 usb-Razer_Razer_BlackWidow_X_Tournament_Edition_Chroma-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 May 19 01:06 usb-Razer_Razer_BlackWidow_X_Tournament_Edition_Chroma-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 May 19 01:06 usb-Razer_Razer_BlackWidow_X_Tournament_Edition_Chroma-if02-event-mouse -> ../event5
lrwxrwxrwx 1 root root   9 May 19 01:06 usb-Razer_Razer_BlackWidow_X_Tournament_Edition_Chroma-if02-mouse -> ../mouse1
```

> Screenshots of all available options on Razer Synapse are needed.

 [Here](http://imgur.com/a/UYPj3), let me know if you need any more specific than that.

> And lastly, the product & model number of the device for documentation reasons please.

```
MODEL NO: RZ01-0190 GAMING MOUSE
PRODUCT NUMBER: RZ01-01900100
```

terrycain, 2017-05-20:
We'll need packet captures of 
* static (a few colours), breathing (all options) and no effect (also unlink the zones and do it on each)
* changing brightness
* changing poll rate
* changing dpi

agurenko, 2017-07-30:
Is there anything I can do to add support for this mouse? Or the case is closed? I've got this mouse recently.

z3ntu, 2017-07-30:
@agurenko Read the last comment of @terrycain

agurenko, 2017-07-30:
@z3ntu yep, got it, will try to do it soon.

agurenko, 2017-07-30:
@z3ntu @terrycain is this what you guys are looking for? https://goo.gl/M1Dc4A should be pretty clean capture. If that works, I will do remaining tomorrow (split zone for lighting)

terrycain, 2017-07-30:
Looks good

agurenko, 2017-07-31:
@terrycain great, I'll do the rest in the evening then. Thank you.

agurenko, 2017-07-31:
@terrycain Here is a second part: https://goo.gl/YWCjnu Let me know if that's enough and remaining combinations can be extracted from first batch or you need all possible combinations.

terrycain, 2017-08-03:
@agurenko what distro you on?

agurenko, 2017-08-03:
@terrycain
 Fedora 26 KDE

terrycain, 2017-08-03:
I've just pushed some initial driver functionality to the `feature_abyssusv2_0x005b` branch. There is some rpm stuff under `./rpm` but I dont do rpms so I dont know off the top of my head how to get it to build off a branch. @z3ntu any ideas?

But once you've got it installed + all its dependencies, reboot and run the following as root
```
ls -l /sys/bus/hid/drivers
ls -l /sys/bus/hid/drivers/razermouse
ls -l /sys/bus/hid/drivers/razermouse/0*/
```

z3ntu, 2017-08-03:
Search for "rpmbuild" in the closed issues and you should find something. ^^

iamshapeless, 2017-08-04:
I completely forgot about this, my apologies. Is there anything else I can do to help testing at this point?

terrycain, 2017-08-04:
@iamshapeless what os you on

iamshapeless, 2017-08-05:
Manjaro Linux @terrycain

agurenko, 2017-08-05:
@z3ntu hm, I'm getting following error when running "rpmbuild -bb razer-drivers.spec":

`RPM build errors:
    File not found: /home/agurenko/rpmbuild/BUILDROOT/razer-drivers-1.1.15-1.fc26.x86_64/etc/xdg/autostart/razer-daemon.desktop
    Directory not found: /home/agurenko/rpmbuild/BUILDROOT/razer-drivers-1.1.15-1.fc26.x86_64/usr/share/razer-daemon`

do have any idea what am I missing?

z3ntu, 2017-08-05:
Ah. The spec file is outdated. Get the newest one from GitHub.com/openrazer/OBS-packaging please

agurenko, 2017-08-05:
@z3ntu yeah, that worked, but after installation and reboot, no razermouse under /sys/bus/hid/drivers :(

terrycain, 2017-08-05:
You sure it built againt the `feature_abyssusv2_0x005b` branch?

agurenko, 2017-08-05:
@terrycain yes, 100% sure :)

terrycain, 2017-08-05:
Copy /lib/udev/rules.d/99-razer.rules to pastebin and link me

agurenko, 2017-08-05:
@terrycain on the other hand... openrazer.spec actually uses master branch... how do I specify branch in a spec file?

terrycain, 2017-08-05:
Err try and get it to use this commit hash 59136f5179ab0eda489e4cb2c01e799eee707f70
You want the installed 99-razer.rules udev file to look like this https://github.com/terrycain/razer-drivers/blob/feature_abyssusv2_0x005b/install_files/udev/99-razer.rules

z3ntu, 2017-08-05:
Replace the # with a % in the line `#define gitcommit` and replace the hash afterwards with the newest one on the target branch.

agurenko, 2017-08-05:
@terrycain @z3ntu yeah, I did that, that seems to work, at least 99-razer.rules looks as expected. I also got /sys/bus/hid/drivers/razermouse...but the mouse isn't working, although powered...

terrycain, 2017-08-05:
whats the output of all these commands

ls -l /sys/bus/hid/drivers
ls -l /sys/bus/hid/drivers/razermouse
ls -l /sys/bus/hid/drivers/razermouse/0*/

agurenko, 2017-08-05:
@terrycain https://pastebin.com/KkcDxhi8

terrycain, 2017-08-05:
does that ` /lib/udev/rules.d/99-razer.rules` file contain `005b`?

agurenko, 2017-08-05:
@terrycain yes it does https://pastebin.com/G0Tmxtit

terrycain, 2017-08-05:
did you reboot? and output of `dmesg | grep -i razer`

agurenko, 2017-08-05:
@terrycain yes I did reboot, here is an output: https://pastebin.com/sKVNEtpK

terrycain, 2017-08-05:
Ok as root do

```
ls -lah /sys/bus/hid/drivers/hid-generic
ls -lah /sys/bus/hid/devices
```

agurenko, 2017-08-05:
@terrycain https://pastebin.com/f0zPwPc1

terrycain, 2017-08-05:
Run the following as root, does it give you any issues?

echo -ne '0003:1532:005B.0009' > /sys/bus/hid/drivers/razermouse/bind
echo -ne '0003:1532:005B.000A' > /sys/bus/hid/drivers/razermouse/bind
echo -ne '0003:1532:005B.000B' > /sys/bus/hid/drivers/razermouse/bind

agurenko, 2017-08-05:
well, it says -bash: echo: write error: No such device and changes nothing

terrycain, 2017-08-05:
ok that means the loaded driver is wrong/old

```
sudo rmmod razermouse

clone the repo, change to the abyssus branch
make
sudo insmod ./drivers/razermouse.ko

ls -l /sys/bus/hid/drivers/razermouse
# If it doesnt contain any 0003* entries run those echo commands again
```

agurenko, 2017-08-05:
@terrycain so okay, that restores the mouse functionality, at least until reboot, udev rule are correct, but polychromatic tool still says daemon does not recognize the device. How would I actually know it's working, can I set settings with CLI?

agurenko, 2017-08-05:
I also have this in journalctl:

```
sudo journalctl -b | grep -i razer_mount
Aug 05 21:10:43 wasp-r7 razer_mount[1006]: Driver razermouse
Aug 05 21:10:43 wasp-r7 razer_mount[1007]: Driver razermouse
Aug 05 21:10:43 wasp-r7 razer_mount[1010]: Driver razermouse
Aug 05 21:10:43 wasp-r7 razer_mount[1011]: Device_ID 0003:1532:005B.0005
Aug 05 21:10:43 wasp-r7 razer_mount[1012]: Device_ID 0003:1532:005B.0006
Aug 05 21:10:43 wasp-r7 razer_mount[1013]: Device_ID 0003:1532:005B.0004
Aug 05 21:10:43 wasp-r7 razer_mount[1014]: Modprobing razermouse
Aug 05 21:10:43 wasp-r7 razer_mount[1015]: Modprobing razermouse
Aug 05 21:10:43 wasp-r7 razer_mount[1016]: Modprobing razermouse
Aug 05 21:10:43 wasp-r7 razer_mount[1035]: Modprobed razermouse
Aug 05 21:10:43 wasp-r7 razer_mount[1037]: Unbinding 0003:1532:005B.0004 from hid-generic
Aug 05 21:10:43 wasp-r7 razer_mount[1038]: Modprobed razermouse
Aug 05 21:10:43 wasp-r7 razer_mount[1039]: Modprobed razermouse
Aug 05 21:10:43 wasp-r7 razer_mount[1040]: Unbinding 0003:1532:005B.0005 from hid-generic
Aug 05 21:10:43 wasp-r7 razer_mount[1041]: Unbinding 0003:1532:005B.0006 from hid-generic
Aug 05 21:10:43 wasp-r7 razer_mount[1048]: Binding 0003:1532:005B.0004 to razermouse
Aug 05 21:10:43 wasp-r7 systemd-udevd[845]: Process 'razer_mount razermouse 0003:1532:005B.0004' failed with exit code 1.
Aug 05 21:10:43 wasp-r7 razer_mount[1051]: Binding 0003:1532:005B.0005 to razermouse
Aug 05 21:10:43 wasp-r7 systemd-udevd[871]: Process 'razer_mount razermouse 0003:1532:005B.0005' failed with exit code 1.
Aug 05 21:10:43 wasp-r7 razer_mount[1054]: Binding 0003:1532:005B.0006 to razermouse
Aug 05 21:10:43 wasp-r7 systemd-udevd[851]: Process 'razer_mount razermouse 0003:1532:005B.0006' failed with exit code 1.
```

terrycain, 2017-08-05:
`ls -l /sys/bus/hid/drivers/razermouse`

agurenko, 2017-08-05:
https://pastebin.com/kf848CrL

terrycain, 2017-08-05:
Thats looking better
`ls -l /sys/bus/hid/drivers/razermouse/0*/`

agurenko, 2017-08-05:
https://pastebin.com/0qtT2Q7J

terrycain, 2017-08-05:
ok in theory polychromatic should recognise it now?

agurenko, 2017-08-05:
No, it says Device Not Recognized

terrycain, 2017-08-05:
run (without sudo)

```
pkill razer-daemon
razer-daemon -Fv
```
It will just sit in a terminal, copy me the output of that

agurenko, 2017-08-05:
https://pastebin.com/t2x23Yxy

terrycain, 2017-08-05:
AHH thats it, control c the daemon for now

As root
```
chown plugdev:plugdev -R /sys/bus/hid/drivers/razermouse/
chown plugdev:plugdev -R /sys/bus/hid/drivers/razermouse/0*/
ls -lah /sys/bus/hid/drivers/razermouse/0*/
```

Then as your user
```
pkill razer-daemon
razer-daemon -Fv
```

terrycain, 2017-08-05:
and then paste me output of ls -lah and the daemon

agurenko, 2017-08-05:
Hm, I don't have  plugdev user, although I have a plugdev group. Doing chown root:plugdev actually crashes the daemon on set_logo_brightness.

Should I create plugdev user manually?

terrycain, 2017-08-05:
nah i should of used chgrp, what error does it give?

agurenko, 2017-08-05:
@terrycain well, permissions denied https://pastebin.com/U6pUBKt2

terrycain, 2017-08-05:
ahh, is your user part of plugdev group?

agurenko, 2017-08-05:
Oh, nevermind, that worked fine:
```
chown root:plugdev -R /sys/bus/hid/drivers/razermouse/
chown root:plugdev -R /sys/bus/hid/drivers/razermouse/0*/
```
Clicked retry in Polychromatic and it worked :)
[image]

[22 more replies]

closed 2017-09-01 by terrycain